EnviroDIY/ModularSensors #2: Change "MaxBotixSonar_Depth" to "MaxBotixSonar_Distance"

aufdenkampe, 2017-02-24:
To better reflect the direction of the measurement (e.g. down to water) and that the values gets larger as water level goes down, which is the inverse of Depth.

aufdenkampe, 2017-03-02:
@SRGDamia1 , what do you think of this? I could do it myself, but though you would be best to implement it.

SRGDamia1, 2017-03-02:
I'm sorry.  That's fine.  I have more changes I want to make and just haven't gotten to this yet.

SRGDamia1, 2017-03-02:
To be more consistent with the MaxBotix materials, it should actually be called range.

aufdenkampe, 2017-03-02:
It makes sense to be consistent with MaxBotix terminology.
I had suggested Distance to be consistent with ODM2 vocabularies: http://vocabulary.odm2.org/variablename/distance/

I would lean toward consistence with ODM2 variable name terms, but to include in the code a comment regarding how we translate the manufacturers term to the ODM2 term.

What do you think?

SRGDamia1, 2017-03-02:
There are two different places it comes up:  In the name of the sensor 
class object that you create and in the "GetVarName" function.  I can 
very easily name the object "range" but have the return of GetVarName be 
"distance".  The GetVarName function already specifies that it is 
returning the variables name using the ODM2 CV, but with the class names 
it just needs to be well documented what the final name is.

aufdenkampe, 2017-03-02:
I like that approach, to name the sensor class object according to the terms used in the sensor documentation, and to have the return of GetVarName be the ODM2 CV variable term. That is a perfect solution, and explicitly maps the two sets of terms.

closed 2017-03-21 by SRGDamia1