EnviroDIY/ModularSensors #3: Add class for DS18B20 Waterproof Digital temperature sensor

aufdenkampe, 2017-03-02:
@s-hicks2 suggested that the [DS18B20 Waterproof Digital temperature sensor](https://www.adafruit.com/products/381) is an excellent choice for stand-alone water temperature. I've been thinking that they would make for an idea pairing with a Maxbotix sonar water level sensor, for a low cost system.

SRGDamia1, 2017-03-02:
Ok.  I have several other classes I want to add as well.  Right now, though, I'm working on getting the base where I want it.

aufdenkampe, 2017-03-02:
No worries. I understand your higher priorities. In the meanwhile, I'll lead the LimnoTech team to develop an example sketch that uses these DS18B20 as another example in: https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/tree/master/examples​

Once that is working, we'll try to implement that code into a new sensor class in the https://github.com/EnviroDIY/ModularSensors repo. We'll probably need your help, but it will be good practice for us to try on our own first.

BTW, the LimnoTech / TwinCities team consists of:
* Craig Taylor: @CtaylorLTI 
* Nick Grewe: @McGrupp10
* Ben Crary: @benjaminCrary
* Beth Fisher: @fisherba (she's also working with Prof. Tom Hickson and his crew from Univ. St. Thomas)

SRGDamia1, 2017-03-04:
@aufdenkampe and @s-hicks2 , I assume that you are thinking of using the DS18B20 to manually correct the MaxBotix data after-the-fact.  The assembled sensor is less expensive than the HR-MaxTemp that MaxBotix sells for automatic compensation, but I cannot find any details about how the data is output from the MaxTemp so I don't see how we could get the DS18B20 to emulate it.  I actually suspect that the built in temperature correction algorithms are going to be better than what we can come up with, and using them also allows us to get the corrected data live.  So, if the only reason to measure the temperature is to correct the sonar, I think it would be better to put the effort toward carefully mounting the MaxTemp according to the MaxBotix suggestions.  (Halfway between the sonar horn and the measurement surface, ventilated, sun-shielded, etc.)  In cases where the sonar is being mounted inside a stilling well, to correct the sonar well you'd also have to put the temperature probe inside the stilling well, which isn't going to give you useful environmental temperatures anyway.

aufdenkampe, 2017-03-04:
@SRGDamia1, thanks for bringing this up. I had read that the MaxSonar expects a very specific response from their MaxTemp sensor, so I have been expecting to always use a MaxTemp sensor with every MaxSonar.

My motivation for the DS18B20 to have a low-cost, stand-alone water temperature sensor, especially where we have a MaxSonar sensor. In MN, water temperature is commonly the primary water quality measurement being made, so many of the people we're talking to here want that plus water depth as a low-cost start to their sensor networks. We also want to develop mini-buoy systems in small lakes that measure water temperature at multiple depths to determine when the water column is stratified versus when it is not and becomes well-mixed.

SRGDamia1, 2017-03-04:
Ok.  That sounds great.  Sorry for the confusion, I thought you wanted to replace the MaxTemp.

aufdenkampe, 2017-04-07:
@SRGDamia1 commented offline that:
"The One-Wire protocol works like SDI-12 in that there's can be many sensors connected, each with their own address, etc.  Ideally, to include it, it should be written up taking all of that into account.  I can just throw it together assuming there's only one sensor, though."

My response:

Good point on the One-Wire. I'm happy with the easy path for now that just implements the single sensor approach shown here:
https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Simple/Simple.pde

Our plan, however, is to have these several of these temperature sensors at multiple depths, so we'll eventually want to figure out how to do this on a single wire using code similar to: https://github.com/milesburton/Arduino-Temperature-Control-Library/blob/master/examples/Multiple/Multiple.pde​

SRGDamia1, 2017-04-07:
Code here compiles:  https://github.com/EnviroDIY/ModularSensors/tree/DS18B20

Please make sure it works, then you can send it to develop

aufdenkampe, 2017-04-07:
Awesome! Thanks!

aufdenkampe, 2017-04-07:
Which sketch do I try?

SRGDamia1, 2017-04-07:
You can use any of them, just add a line for the DS18B20.

Or I just push a change to the single sensor to make it for the DS18B20.  I don't want that example kept that way in the master branch, though, because we know we need to change the way the one-wire sensors are polled so that instead of just assuming there's only one sensor connected and it's at index 0, the actual 64bit hex address for the sensor is used.

aufdenkampe, 2017-04-07:
I got your sketch to load. When I open the serial monitor, it says:

'''
Now running logging_to_EnviroDIY.ino on EnviroDIY Mayfly Mayfly_160073
Current RTC time is: 2000-01-02T00:26:05-05:00
There are 14 variables being recorded.
Setting up sensors.
--- Successfully set up DecagonCTD ---
--- Successfully set up Decagon5TM ---
--- Successfully set up DecagonES2 ---
--- Successfully set up CampbellOBS3+ ---

--- Successfully set up CampbellOBS3+ ---
--- Successfully set up MaxBotixMaxSonar ---
--- Successfully set up EnviroDIYMayflyRTC ---
--- Successfully set up EnviroDIYMayflyBatt ---
--- Successfully set up EnviroDIYMayflyHeap ---
Data will be saved as Mayfly_160073_2000-01-02.csv
Connecting to SD Card with card/slave select on pin 12
'''

Is this what I should expect?

aufdenkampe, 2017-04-07:
I got your sketch to compile and load on PlatformIO but couldn't get reasonable serial output. It didn't seem to be working.

I was not able to get the sketch to compile on the Arduino IDE. The SDFat library seems mixed up. When I "fixed" that, I got this error:
```
/Users/aaufdenkampe/Documents/Arduino/EnviroDIY_ModularSensors/examples/single_sensor/single_sensor.ino: In function 'void loop()':
single_sensor:95: error: 'class MaximDS18B20_Temp' has no member named 'getValueString'
     Serial.println(temp.getValueString());
                         ^
exit status 1
'class MaximDS18B20_Temp' has no member named 'getValueString'
```

SRGDamia1, 2017-04-07:
You got that from single_sensor.ino?  That's definitely not right.

Did you just hit upload and run whatever platformio.ini was set for?  Because it really looks like you uploaded and ran logging_to_EnviroDIY.ino.  That's what it says it's running, anyway.

SRGDamia1, 2017-04-07:
I just pushed a change so that you now do need to specify the device address.  That allows for using multiple sensors properly.  Underneath it's still a very inefficient way to access multiple devices, but it will work.  And now, even if it's made more efficient under than hood later, it's much less likely that we'll need to break reverse compatibility.

SRGDamia1, 2017-04-07:
This is the printout I'm getting:  (with no sensor attached)
```
Single Sensor Example - DS18B20
Data recieved from sonar: -127.0000
Data recieved from sonar: -127.0000
```

SRGDamia1, 2017-04-07:
WRT the Arduino IDE, you're libraries are out of date.  If you haven't modified anything in the libraries, and your using a clone from the EnviroDIY_Libraries repo, I'd completely delete everything in the whole arduino library directory and reclone from scratch.  I've done some forced pushes in that library which may have screwed you up.

SRGDamia1, 2017-04-29:
Closed by #37

closed 2017-04-29 by SRGDamia1