EnviroDIY/ModularSensors #11: Add SAMD Support for Sleeping/Logging

SRGDamia1, 2017-03-28:
Would enable support for Arduino Zero, Adafruit Feather, Sodaq One, Sodaq Autonomo, etc.

Need to change timing, sleeping, and RTC functions to support.

Would use RTCZero library for clock and sleep:  https://github.com/arduino-libraries/RTCZero
SAMD timer library for timing: https://github.com/avandalen/avdweb_SAMDtimer

Will not work at this time for SDI-12 devices b/c the SDI-12 library is avr only.

SRGDamia1, 2017-03-28:
I think SdFat already works for SAMD.

SRGDamia1, 2017-03-28:
On sleep modes on SAMD21:
The SAM D21 devices have two software-selectable sleep modes, idle and standby. In idle mode the CPU is stopped while all other functions can be kept running. In standby all clocks and functions are stopped expect those selected to continue running. The device supports SleepWalking. This feature allows the peripheral to wake up from sleep based on predefined conditions, and thus allows the CPU to wake up only when needed, e.g. when a threshold is crossed or a result is ready. The Event System supports synchronous and asynchronous events, allowing peripherals to receive, react to and send events even in standby mode.
https://forum.arduino.cc/index.php?topic=337289.0

aufdenkampe, 2017-08-09:
@SRGDamia1, I was meeting with @awickert last week about https://github.com/NorthernWidget and about teaming up to create an upper midwest open-source enviro data logging/sensing "makers" group, and he mentions that he found that firmware/bootloader for the SAMD21 was too unstable for them to use with some functions. They thus decided to focus future efforts using the AtMega1284p chip that we use. It might be worth connecting with him if you want to pursue the SAMD21.

awickert, 2017-08-09:
@SRGDamia1 Our main issue was with a radio communications library; @bschulz1701 might have some ideas about how they worked out sleep on their SAMD21 logger that they built as a design project with my group.

SRGDamia1, 2017-09-07:
#68 does most of the work to get this going, but there are still bugs to fix

SRGDamia1, 2017-10-27:
As of #77, this is done!  Just a head's up, because the USB support is integrated into the processor instead of into a separate chip, the USB port will close when the chip sleeps.  If you want to do debugging, I recommend using a serial port monitor like Tera Term which will immediately reopen a connection after a disconnected device reconnects.

Oh, and I fixed the SDI-12 library to work with SAMD boards a while ago, so everything should be supported.

closed 2017-10-27 by SRGDamia1

aufdenkampe, 2017-10-28:
Awesome!!!! Nice work!