EnviroDIY/ModularSensors #29: Add a diagnosis mode enterable on button push

SRGDamia1, 2017-04-07:
Activate a mode where data is sent every 10 seconds for faster diagnosis of problems.

aufdenkampe, 2017-04-07:
I was thinking about something similar, in which such a push-button diagnosis mode would also send those every-10-second measurement data to a SSD1306 OLED display (https://www.amazon.com/Grove-128x64-Board-Arduino-Raspberry/dp/B01D5GLDJ2). 

This would be similar to this: https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/blob/Mayfly_11_DT-LED/workshop/mayfly_10b_DHT_LED/mayfly_10b_DHT_LED.ino

SRGDamia1, 2017-04-07:
Or have the data go right to the serial port and use something like this to read it:  https://play.google.com/store/apps/details?id=com.felhr.droidterm&hl=es

Doing it that way means less work because there's no messing with another library or display board.

aufdenkampe, 2017-04-07:
Hmm. Yes. I can see why that would help.

My thinking was that if someone is doing maintenance on a sensor station in the field, they could quickly put it in instant measure mode to jot down readings. Alternately, someone could do a salt-injection experiment or something like that, by going to their station, putting it 10-sec mode and they could watch the salt pulse go by and also record the high-frequency data.  It's possible that these are separate functions, however. My thinking is that these come lower down the priority list relative to adding new sensor classes.

SRGDamia1, 2017-06-06:
#55 adds a debug mode, but it's currently only accessible when the "check for debug" function is running in the setup.  It would be much better if I could figure out how to run it as an interrupt.

closed 2018-02-26 by SRGDamia1