EnviroDIY/ModularSensors #30: Add a class for a DHT22/DHT11 digital humidity and temperature sensor

aufdenkampe, 2017-04-07:
This would also be a relative simple example to describe how to create new sensor sub-classes.

These sensors are used in Workshop Sketch 6.
* https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/tree/master/workshop/mayfly_06_humidity-temp​
and in this potential workshop sketch:
* https://github.com/EnviroDIY/EnviroDIY_Mayfly_Logger/blob/Mayfly_11_DT-LED/workshop/mayfly_10b_DHT_LED/mayfly_10b_DHT_LED.ino

Info on these sensors can be found at:
* http://wiki.seeed.cc/Grove-TemperatureAndHumidity_Sensor/
* http://wiki.seeed.cc/Grove-Temperature_and_Humidity_Sensor_Pro/

We've used the Adafruit DHT-sensor-library
https://github.com/adafruit/DHT-sensor-library/tree/c97897771807613d456b318236e18a04b013410b

SRGDamia1, 2017-05-03:
In progress in https://github.com/EnviroDIY/ModularSensors/tree/DHTSensor

SRGDamia1, 2017-05-03:
Closed by #40.

PR #40 (2017-05-03): Adds support for AOSong DHT sensors
Closes #30
closed 2017-05-03 by SRGDamia1