EnviroDIY/ModularSensors #39: Create first release?

aufdenkampe, 2017-05-03:
I think things are mature enough that it might be worth creating our first release in the next week or so, after a tiny bit of more testing.

How about v 0.1.0? Do we need to add "alpha" or "beta"? We could, but we're already using this code to run many Mayfly stations, so I'm thinking that we just give it a number. It works. The next bug fixes can be v 0.1.1. The next time we add sensors or any other important feature, it becomes v 0.2.0, etc.

SRGDamia1, 2017-05-03:
After getting everything merged up to master, it would be great to do a release.  The "version" I currently have written within my library is already "2.1.1-alpha" because I've made backwards-incompatible revisions several times.  I guess that doesn't really matter though and we can set the version back to 0.1.0 for the first official release.

aufdenkampe, 2017-05-03:
Awesome, it looks like we were both thinking along the same lines!

As for version number, I'm OK with either starting point, as long as we are generally following Semantic Versioning recommendations (http://semver.org).

If you think no one other than yourself has been using this repo for deployments, I think I might lean toward dialing back to zero at v 0.1.0, just to let people know that this is all relatively new. v 2.1.1 sounds like it is quite mature. BTW, in semantic versioning, any release that is v 0.x.x is allowed to make changes that are backwards incompatible. It's only after 1.x.x that you need to pay close attention to it.

If others have been deploying this code, then we might go with 2.1.1, but I would suggest we drop the "alpha".

One option might be for us to call this initial release v 0.2.1.1, then drop the forth digit after we go to v 0.2.2. That would allow you to somewhat preserve your old versioning system.

SRGDamia1, 2017-05-09:
Done!

closed 2017-05-09 by SRGDamia1