EnviroDIY/ModularSensors #54: Add a class for Apogee Sunlight Sensor

aufdenkampe, 2017-06-01:
SWRC has deployed several of these sensors, and LimnoTech is about to deploy one in the next weeks.
SQ-212: Amplified 0-2.5 Volt Sun Calibration Quantum Sensor
https://www.apogeeinstruments.com/sq-212-amplified-0-2-5-volt-sun-calibration-quantum-sensor/

@SRGDamia1 , can you work with @s-hicks2 to add a class for this sensor?

s-hicks2, 2017-06-01:
It's really simple.  Just do an aux analog measurement with the ADS1115 and
record the voltage.  1ma equals 1 PAR.  I think maybe I did some averaging
over a several second period, but the function is similar to the OBS
function but simpler because there's no calibration formula.

aufdenkampe, 2017-06-05:
@SRGDamia1 and @s-hicks2, I got an example sketch working, here: 
https://github.com/EnviroDIY/ModularSensors/tree/Apogee_SQ-212/sensor_tests/Apogee_SQ212_AnalogTest

It's based on https://github.com/EnviroDIY/ModularSensors/tree/master/sensor_tests/mayfly_turbidity_test1

It works, but the one thing I can't figure out is why divide the measured ADC registers by 17585?

In the typical examples for Arduino., I get why one divides by 1023 for the built-in 10-bit ADC (i2^10-1 = 1023), but we're using the 16-bit ADS1115, so shouldn't we be dividing by 2^16-1, or 65535?

One reason why I'm skeptical is that when I use 17585, I was able to max out the detector with the LED flashlight on my phone. When I used 65535 the values were about 300-600 Einsteins (μmol m-2 s-1), which seems more likely.

s-hicks2, 2017-06-05:
> 212/sensor_tests/Apogee_SQ212_AnalogTest
> sensor_tests/mayfly_turbidity_test1

The ADS1115 has a variable gain option.  We usually use it in the default
setting (which is 2/3 gain). If you look at the example code for the
ADS1x15 library, you can see in the comments that the resolution of the
default setting is 1 bit = 0.1875mV

On the Mayfly, the supply voltage of the converter is 3.3v, so a full-scale
signal would be 17585 bits.  I came up with that number on my own several
years ago through experimenting , but I've seen other people who derived it
using similar methods, though some use 17600, which might be mathematically
correct, but measurements with a precision voltmeter showed 17585 to be a
more accurate number to use.

You can test this by running the following sketch with a standard1.5v
alkaline battery connected to the AA2 pin of the Mayfly.  You should get
around 1600mV for a brand new battery:

#include <Wire.h>
#include <Adafruit_ADS1015.h>
Adafruit_ADS1115 ads;  /* Use this for the 16-bit version */

void setup(void)
{
  Serial.begin(9600);
  ads.begin();
}

void loop(void)
{
  int16_t adc2;
  adc2 = ads.readADC_SingleEnded(2);
  float voltage = (adc2 * 3300.0)/17585.0;
  Serial.print("AIN2 bits: "); Serial.print(adc2);
  Serial.print("   Voltage: ");  Serial.print(voltage);
 Serial.println("mV"); "
  delay(1000);
}

aufdenkampe, 2017-06-06:
@s-hicks2, thank you! This was very helpful. 
I'll record all of this in some documentation that I'm working on in the Apogee_SQ-212 branch of this repo.

aufdenkampe, 2017-06-06:
I got it all working as a new set of .h and .cpp files in the modular code! 
I created a pull request (#56) to merge into the Develop branch.

closed 2017-06-08 by SRGDamia1