EnviroDIY/ModularSensors #65: Reduce radio attempts if no response, for data & power savings

aufdenkampe, 2017-08-09:
@SRGDamia1, I just noticed that my data usage for my 160065 CROSSLANDSPOND - 10 VARIABLES station (http://data.envirodiy.org/sites/6/) shot up from 0.74 MB per day to 5.3 MB per day from July 26 to Aug. 3, exactly when the Data.EnviroDIY.org system went down. See:
[image]

I'm thinking that this jump might be due to repeated attempts by the logger to send the radio data. Can we limit the number of attempts? This 7x increase in data usage for 6 days added a good bit to my monthly cost (almost $30).

Also, I'm thinking that this may have also significantly chewed on power. @s-hicks2 mentioned that she's seen power issues. Are they related? Also, our six stations on the Kinni, deployed by @CTaylorLTI, all permanently lost connectivity around the 26th, and we're wondering if that might be due to the power getting drained. He'll go out soon to find out, and I'll try to download the voltage data for our data gap period to see if we had a similar problem, but I think we did (voltage on our first day back online was lower than usual).

s-hicks2, 2017-08-09:
> screen shot 2017-08-09 at 10 56 49 am

We had several of our other DRWI loggers use up a month's worth of data 
in less than a week at the end of July, so it sounds like the same 
issue.  Glad we weren't the only ones to have it.  So it's related to an 
outage on the data.envirodiy site?

SRGDamia1, 2017-08-09:
I can definitely see how the site being down would chew up power because the modem is waiting to connect, but I'm honestly completely baffled by the increased data usage.

SRGDamia1, 2017-08-10:
I will look into stopping the transmissions earlier so the logger doesn't bother to listen to a full response.  Also, on Jeff's end, some TCP changes might reduce usage:  http://www.techrepublic.com/article/take-advantage-of-tcp-ip-options-to-optimize-data-transmission/

aufdenkampe, 2017-08-10:
@horsburgh, can you look into above?

SRGDamia1, 2017-09-07:
Commit c85c2e8aaced9498ccddf0aa5c44f532cafce136 is untested, but should stop the logger from listening after the first 12 characters of the response have come in.  There never was more than one attempt to send data; if the first attempt fails, it fails.

SRGDamia1, 2017-09-08:
Fixed in #71, but untested as yet.

closed 2017-09-08 by SRGDamia1