EnviroDIY/ModularSensors #70: Complete Yosemitech modular libraries

aufdenkampe, 2017-09-07:
Let's renew our push to test and finalize our base library for YosemitechModbus sensors (https://github.com/EnviroDIY/YosemitechModbus), so that we can develop Yosemitech sensor sub-class files within this ModularSensors library. I just created a new Yosemitech branch for this work: https://github.com/EnviroDIY/ModularSensors/tree/yosemitech

@fisherba is ordering a pile of Yosemitech sensors to hopefully test in the field in October.

@SRGDamia1 said that by early July, she had everything working for her in YosemitechModbus.
Unfortunately, when I tested the code on my newer hardware versions of the sensors with brushes, I had issues. See https://github.com/EnviroDIY/YosemitechModbus/issues/3

SRGDamia1, 2017-09-08:
Classes now exist for all 5 Yosemitech sensors I have documentation for (#71).
It works for me, but I can't make any promises for you.  The classes are extremely simple; all the hard work is done by the YosemitechModbus library underneath.  I'm closing this issue because the future work to get your sensors going should go there.

SRGDamia1, 2017-09-08:
PS.  How are you attaching your sensors to your board?  Are you using SoftwareSerial?  SoftwareSerial is less than ideal, and I flat-out couldn't get it to work on the modbus sensors when I was working in the modular library.  I flipped over to using AltSoftSerial and suddenly everything worked.  I'd suggest trying AltSoftSerial  (or, better, Serial1) and see if that helps.

closed 2017-09-08 by SRGDamia1

aufdenkampe, 2017-09-11:
@SRGDamia1, that's a great tip. I hope to try it out in the next day or so. I'll let you know.