EnviroDIY/ModularSensors #75: Sphinx documentation?

aufdenkampe, 2017-10-10:
@SRGDamia1, are you inspired by http://odm2.github.io/ODM2PythonAPI?
It would be awesome to have this, to get others to start contributing.

I bet @lsetiawan would be happy to provide guidance and help.

SRGDamia1, 2017-10-10:
How does it work with C++?

lsetiawan, 2017-10-10:
Not sure, I have not explored that. @ocefpaf, Do you know if sphinx can do C++?

SRGDamia1, 2017-10-10:
I have a lot of comments and "doc-strings" in this repo, but everything is in C++

ocefpaf, 2017-10-10:
I never used it but there is a sphinx extension to `Doxygen` that can do that.

https://breathe.readthedocs.io/en/latest/

aufdenkampe, 2018-05-01:
I've learned a bit more about Doxygen, and I think it is the right choice for these Arduino libraries. https://en.wikipedia.org/wiki/Doxygen.
Changing the name of this issue. Thanks all for the feedback.

SRGDamia1, 2020-12-15:
Three years later, this is done.

closed 2020-12-15 by SRGDamia1

aufdenkampe, 2020-12-15:
@SRGDamia1, thank you for this very impressive effort (and outcome!) on your end!