EnviroDIY/ModularSensors #95: Fix loggingInterval. Presently 2 minutes regardless of value

aufdenkampe, 2018-02-07:
My code is set to loggingInterval = 5, but my timestamps are 2-4 minutes apart (they are not even always the same). What is going on there?

The sketch is here: https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-RPBCWD/RPB-3

Here's the output.
```
Now running logging_to_EnviroDIY.ino on Logger Mayfly_170010
Logger timezone is UTC-5
RTC timezone is UTC
Initializing logger Mayfly_170010 to record at 5.00 minute intervals ...    ... Success!
Current RTC time is: 2000-01-27T20:12:34-05:00
Attempting to synchronize RTC with NIST
Clock synced!
Successfully connected to SD Card with card/slave select on pin 12
Data will be saved as Mayfly_170010_2018-02-07.csv...   ... File created!
Logger setup finished!
------------------------------------------

Push button NOW to enter debug mode.
------------------------------------------

End of debug mode.
------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 12345678-abcd-1234-efgh-1234567890ab
Content-Length: 475
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:02:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.609, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 23.75, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 7389, "a58c3985-45cd-407c-92d6-18b89dde8c4d": 0.0, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": 21.22, "f05e563f-456e-459c-8dcb-aa87d06489c5": 0.00, "89baa175-b592-42b5-8322-de2df59e4bfd": 20.88 }

 -- Response Code -- 
201

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:02:00,4.609,10712,23.75,7389,0.0,21.22,0.00,20.88
------------------------------------------

------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 12345678-abcd-1234-efgh-1234567890ab
Content-Length: 474
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:04:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.624, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 25.00, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 792, "a58c3985-45cd-407c-92d6-18b89dde8c4d": 0.0, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": 21.25, "f05e563f-456e-459c-8dcb-aa87d06489c5": 0.00, "89baa175-b592-42b5-8322-de2df59e4bfd": 21.25 }

 -- Response Code -- 
2019

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:04:00,4.624,10712,25.00,792,0.0,21.25,0.00,21.25
------------------------------------------

------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 12345678-abcd-1234-efgh-1234567890ab
Content-Length: 475
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:05:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.624, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 25.25, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 792, "a58c3985-45cd-407c-92d6-18b89dde8c4d": 0.0, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": 21.26, "f05e563f-456e-459c-8dcb-aa87d06489c5": 50.77, "89baa175-b592-42b5-8322-de2df59e4bfd": 21.42 }

 -- Response Code -- 
2019

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:05:00,4.624,10712,25.25,792,0.0,21.26,50.77,21.42
------------------------------------------

------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 12345678-abcd-1234-efgh-1234567890ab
Content-Length: 475
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:06:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.639, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 25.75, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 792, "a58c3985-45cd-407c-92d6-18b89dde8c4d": 0.0, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": 21.26, "f05e563f-456e-459c-8dcb-aa87d06489c5": 38.19, "89baa175-b592-42b5-8322-de2df59e4bfd": 21.58 }

 -- Response Code -- 
201

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:06:00,4.639,10712,25.75,792,0.0,21.26,38.19,21.58
------------------------------------------

------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 12345678-abcd-1234-efgh-1234567890ab
Content-Length: 475
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:08:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.639, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 26.25, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 792, "a58c3985-45cd-407c-92d6-18b89dde8c4d": 0.0, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": 21.30, "f05e563f-456e-459c-8dcb-aa87d06489c5": 30.63, "89baa175-b592-42b5-8322-de2df59e4bfd": 21.75 }

 -- Response Code -- 
201

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:08:00,4.639,10712,26.25,792,0.0,21.30,30.63,21.75
------------------------------------------

------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 12345678-abcd-1234-efgh-1234567890ab
Content-Length: 475
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:10:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.639, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 26.50, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 792, "a58c3985-45cd-407c-92d6-18b89dde8c4d": 0.0, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": 21.31, "f05e563f-456e-459c-8dcb-aa87d06489c5": 25.64, "89baa175-b592-42b5-8322-de2df59e4bfd": 21.88 }

 -- Response Code -- 
201

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:10:00,4.639,10712,26.50,792,0.0,21.31,25.64,21.88
------------------------------------------

------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 
Content-Length: 481
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:12:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.655, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 26.50, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 792, "a58c3985-45cd-407c-92d6-18b89dde8c4d": -1999.8, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": -1982.74, "f05e563f-456e-459c-8dcb-aa87d06489c5": 1.59, "89baa175-b592-42b5-8322-de2df59e4bfd": 22.00 }

 -- Response Code -- 
201

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:12:00,4.655,10712,26.50,792,-1999.8,-1982.74,1.59,22.00
------------------------------------------

------------------------------------------

 \/---- Post Request to EnviroDIY ----\/ 
POST /api/data-stream/ HTTP/1.1
Host: data.envirodiy.org
TOKEN: 12345678-abcd-1234-efgh-1234567890ab
Content-Length: 474
Content-Type: application/json

{"sampling_feature": "12345678-abcd-1234-efgh-1234567890ab", "timestamp": "2018-02-07T15:14:00-05:00", "9e9f2c7b-0ab3-4dfe-b933-ef0a1920bb63": 4.670, "33218554-9cf6-48f3-b8d0-4d032dcb8578": 10712, "c34b0278-3267-4969-929e-62c45a1d8ff1": 26.75, "47a3117d-9c39-43c9-87c2-a05aa573e1b0": 793, "a58c3985-45cd-407c-92d6-18b89dde8c4d": 0.0, "908ad4f8-b6db-4b0c-85c6-b4f14c3229bf": 21.36, "f05e563f-456e-459c-8dcb-aa87d06489c5": 0.94, "89baa175-b592-42b5-8322-de2df59e4bfd": 22.13 }

 -- Response Code -- 
201

 \/---- Line Saved to SD Card ----\/ 
2018-02-07 15:14:00,4.670,10712,26.75,793,0.0,21.36,0.94,22.13
------------------------------------------

------------------------------------------
```

aufdenkampe, 2018-02-07:
@SRGDamia1, also note that 2nd to last Conductivity values were bad (large negative numbers). Not sure if its related, but nearly every other conductivity value looks like this when in debug mode.

SRGDamia1, 2018-02-07:
Within the first 20 minutes/10 measurements it takes readings at 2 minute intervals, per Shannon's request.

aufdenkampe, 2018-02-07:
Ah! That's a great idea, for trouble-shooting during deployment.

Let's close this issue once that's described in the ReadMe file.

aufdenkampe, 2018-02-07:
Done! See https://github.com/EnviroDIY/ModularSensors/commit/3d1dcf416787481594110ebf8149422085fa43eb

@SRGDamia1, please review that this is the right spot for this documentation.

closed 2018-02-07 by aufdenkampe