EnviroDIY/ModularSensors #96: Change Conductivity units from mS/cm to microS/cm for Yosemitech Y520

aufdenkampe, 2018-02-07:
The default output units for Yosemitech Y520 sensor is mS/cm (milli Siemens per centimeter), but nearly everyone looking at freshwater uses µS/cm (microS/cm, sometimes written as uS/cm). This includes the EnviroDIY Data Portal.

It seems we should provide the unit conversion in this library, rather than in https://github.com/EnviroDIY/YosemitechModbus.

@SRGDamia1, Where/how do we do that?

aufdenkampe, 2018-02-08:
@SRGDamia1, I'm happy to do code changes and test, if you point me to where it might be best to change. It would be nice to get this done before we issue the next release.

What do you think?

SRGDamia1, 2018-02-08:
It's in the YosemitechParent.cpp, in addSingleMeasurementResult.

closed 2018-02-08 by SRGDamia1

aufdenkampe, 2018-02-08:
Thank you! I appreciate you taking care of this. 

Seeing your commit helps me better understand ModularSensors in general, and also the fix to https://github.com/EnviroDIY/YosemitechModbus/issues/9 helped with understanding, especially after finding the code here: https://github.com/EnviroDIY/YosemitechModbus/blob/master/src/YosemitechModbus.cpp#L348-L357.

Thanks!

commit 5092d0a9 (2018-02-08): Convert mS/µS; closes #96