EnviroDIY/ModularSensors #98: Release 0.6.4

aufdenkampe, 2018-02-08:
@SRGDamia1, I just tested the Master branch, and its working great!
I think it's worth creating an official tagged release for this. Also, I'm thinking that we remove the "beta" tag. We're ready to make this public as a stable release.

SRGDamia1, 2018-02-12:
I'm still having some trouble with the modem with the code on develop.  I don't want to push to master until I figure that out.

aufdenkampe, 2018-02-12:
Sounds good. Thanks for working on that!
Do you have time this week to work on it?

aufdenkampe, 2018-02-12:
@SRGDamia1, does https://github.com/EnviroDIY/ModularSensors/commit/d8a49c17a7c722a3eb6b802deaa94cba2c2fdef6 fix the modem issue?

SRGDamia1, 2018-02-13:
Yes, it should fix the modem issue.  If tests go well for you, I think it's ready to release.

SRGDamia1, 2018-02-13:
Ok, there are still some problems with the OBS3 and nan's.  It's not ready yet.