EnviroDIY/ModularSensors #103: 5TM calculating VWC

fisherba, 2018-03-08:
Hi @SRGDamia1,

We have been running the multisensor_print sketch to test 5TM sensors. They are reading permittivity and temperature, but the calculation for volumetric water content is reporting as -9999. 

Example results.....
in air: 
```
Updated all sensors at 2000-05-27T11:57:43-06:00
EnviroDIY Mayfly at EnviroDIY Mayfly reports batteryVoltage is 4.852 Volt
EnviroDIY Mayfly at EnviroDIY Mayfly reports Free SRAM is 14401 Bit
MaximDS3231 at I2C_0x68 reports temperatureRTC is 22.75 degreeCelsius
Decagon5TM at SDI12-2_Pin7 reports permittivity is 1.2900 faradPerMeter
Decagon5TM at SDI12-2_Pin7 reports temperature is 20.7 degreeCelsius
Decagon5TM at SDI12-2_Pin7 reports volumetricWaterContent is -9999.00 percent
In CSV Format:  4.852,14401,22.75,1.2900,20.7,-9999.00
```
(My calculated VWC with this permittivity is -0.016.)

in water: 
```
Updated all sensors at 2000-05-27T12:16:30-06:00
EnviroDIY Mayfly at EnviroDIY Mayfly reports batteryVoltage is 4.852 Volt
EnviroDIY Mayfly at EnviroDIY Mayfly reports Free SRAM is 14401 Bit
MaximDS3231 at I2C_0x68 reports temperatureRTC is 22.75 degreeCelsius
Decagon5TM at SDI12-2_Pin7 reports permittivity is 53.1000 faradPerMeter
Decagon5TM at SDI12-2_Pin7 reports temperature is 21.3 degreeCelsius
Decagon5TM at SDI12-2_Pin7 reports volumetricWaterContent is -9999.00 percent
In CSV Format:  4.852,14401,22.75,53.1000,21.3,-9999.00
```
(My calculated VWC with this permittivity is 0.59.)

When I calculate VWC, the in-water one should return a non-error result. I'm wondering if the calculation turns VWC from a float? I tried a version commenting out the percent conversion in Decagon5TM.h and had the same result.

aufdenkampe, 2018-03-09:
@SRGDamia1, are you able to work on this? If so, many thanks in advance. I want to learn from watching your fixes, so if you make any commits to fix this issue, could you tag this issue in the commit comments?

Also, are your recent commits to https://github.com/EnviroDIY/Arduino-SDI-12 at all connected to this issue?

SRGDamia1, 2018-03-09:
I'll give it a quick look, anyway.

The changes to the SDI-12 library shouldn't make any difference in this, especially because the "fix" I'd thought I'd made for the "parse" functions didn't actually change the behavior at all.  I think I do have it properly fixed now, but haven't pushed that yet.

SRGDamia1, 2018-03-09:
Found the problem...

I'd forgotten that I'd overwritten the update function for the 5TM because VWC is actually calculated in the program using the TOPP equation.  The sensor doesn't return that value.  When I changed all the modules so that instead of a unique update function they have separate "sstartSingleMeasurement" and "addSingleMeasurementResult" functions, I'd missed the 5TM.

SRGDamia1, 2018-03-09:
Okay.  I think I fixed it on the 5TM branch.  Can you test it?  I don't have one of those sensors right now.

fisherba, 2018-03-10:
@SRGDamia1, thanks for working on this!

I think I set up all the libraries correctly to test that branch, including the latest work on the SDI12 library. Unfortunately, I now get this compile error, which looks related to the new code you committed in https://github.com/EnviroDIY/ModularSensors/commit/268bd73631bb7dd46f519745a03d072fe2748296.

```
In file included from /Users/fisher/Documents/Arduino/libraries/ModularSensors_5TMfix/src/Decagon5TM.h:32:0,
                 from /Users/fisher/Documents/Arduino/libraries/ModularSensors_5TMfix/src/Decagon5TM.cpp:29:
/Users/fisher/Documents/Arduino/libraries/ModularSensors_5TMfix/src/DecagonSDI12.h: In member function 'virtual bool Decagon5TM::addSingleMeasurementResult()':
/Users/fisher/Documents/Arduino/libraries/ModularSensors_5TMfix/src/DecagonSDI12.h:66:10: error: 'char DecagonSDI12::_SDI12address' is private
     char _SDI12address;
          ^
/Users/fisher/Documents/Arduino/libraries/ModularSensors_5TMfix/src/Decagon5TM.cpp:40:23: error: within this context
     getDataCommand += _SDI12address;
                       ^
exit status 1
Error compiling for board EnviroDIY Mayfly 1284p.
```

I appreciate your help with getting this finalized. Thanks.

SRGDamia1, 2018-03-13:
Um, can you try the dopto branch instead?  Other parts of that branch are a bit off, but I've already fixed the private/protected in it.

fisherba, 2018-03-14:
dopto branch is working to calculate VWC.

SRGDamia1, 2018-03-14:
Oh, good.

closed 2018-03-14 by SRGDamia1