EnviroDIY/ModularSensors #108: Add support to trigger ISCO when value from a sensor exceeds threshold

aufdenkampe, 2018-03-18:
My use case is when YosemiTech Turbidity exceeds 50 NTU.

SRGDamia1, 2018-03-21:
This is a pretty different functionality from what any of the other parts of this library do.  Not that it couldn't be done, but I'm not sure how the ISCO would fit as a sensor or logger in this library.  This might be the sort of functionality that you'd want to add as part of a specific unit's programming rather than as part of this library.

If you really want it to be part of this library, I suppose you could do something like add a function to the variable base object like "trigger other function" where the inputs for the function would be the value that would activate the trigger, a pointer to some other function from outside of the modular sensors library that would be triggered, and some sort of keyword or value to define in what way the trigger is applied (less than value, greater than, etc).

A function like that would actually be relatively simple to write, and could be part of the variable base, but I don't think it would actually give you the functionality that you want.  For example, if you set up the turbidity variable to trigger some "activate isco" function whenever the value exceeded 50 NTU, it would activate every single time the variable value was updated with any value over NTU.  I suspect that what you really want to happen is for the ISCO to activate and begin a sampling program a single time after the value exceeded 50 and then not activate ever again, or to not activate again until after the turbidity had fallen back below 50 and then risen above it again.  For that sort of functionality, you'd need additional internal variables within each variable object that track if any of their assigned triggers had run, when it last ran, whether it should be reset, etc.  So, suddenly, it's not very simple at all.  And turbidity is actually particularly tricky since it has a nasty habit of being noisy, spiky data.  So you'd probably want to have a trigger more like "each of the last 3 values have been over 50" to avoid triggering your ISCO and dirtying all your bottles because a leaf passed by at the wrong moment.

In short, I think it could be done, but it would be difficult to get the actual functionality that you want and I'm not sure this library is really the place to do the heavy lifting wrt the triggers.

aufdenkampe, 2018-04-06:
Moving this work to a new repo, given that it has no dependencies on Modular Sensors.
New repo: https://github.com/EnviroDIY/Trigger

SRGDamia1, 2018-04-06:
Since this is now part of another repo, I'm going to close this issue.

closed 2018-04-06 by SRGDamia1