EnviroDIY/ModularSensors #110: Add support for MS5803-14BA pressure sensor

aufdenkampe, 2018-03-18:
For the [SparkFun Pressure Sensor Breakout - MS5803-14BA](https://www.sparkfun.com/products/12909), which uses the [Measurement Specialties MS5803-14BA Miniature 14 bar Module](https://cdn.sparkfun.com/datasheets/Sensors/Weather/ms5803_14ba.pdf).

Start with https://github.com/sparkfun/SparkFun_MS5803-14BA_Breakout_Arduino_Library ?

With @fisherba. 

Our friends @bschulz1701, @superchap123 & @awickert at NorthernWidget are also working with this sensor. See https://github.com/NorthernWidget/MS5803

aufdenkampe, 2018-03-18:
Barebones implementation from @ehost11:
https://github.com/ehost11/EnviroDIY_Mayfly_Logger/blob/master/capstone/WaterPressureBarebones/WaterPressureBarebones.ino

bschulz1701, 2018-03-19:
@aufdenkampe, since you mentioned the library we have built for the MS5803, I just wanted to give you a heads up that I will probably be cleaning that library up in the next week or two, all the functions should stay the same, it will just be fixing any bugs that I find while working with our TP-Downhole sensors (which is what we made that library for)

SRGDamia1, 2018-04-18:
This is now in the master branch.

closed 2018-04-18 by SRGDamia1