EnviroDIY/ModularSensors #112: Yosemitech sensors all giving -9999

aufdenkampe, 2018-03-22:
@SRGDamia1, I finally tried out the new release v0.6.10, and all of my YosemiTech sensors are giving me -9999 for all variables. They all work fine with the [YosemitechModbus/examples/GetValues.ino](https://github.com/EnviroDIY/YosemitechModbus/tree/master/examples/GetValues).

Any chance this is related to #103?

SRGDamia1, 2018-03-22:
I'll try to look into this.  I doubt it's related to the 5TM issue.

aufdenkampe, 2018-04-02:
@SRGDamia1, has any of your numerous commits in the last two weeks addressed this issue? If not, any ideas how to to debug this?

SRGDamia1, 2018-04-02:
Nothing I've done is directly related to this, but that being said, I am getting numbers fine from my Yosemitech sensors with the current version in the timers section.  It's actually the SDI-12 sensors that are giving me headaches.

aufdenkampe, 2018-04-06:
Yes, it looks like the problem is no longer there!

I just got it to work with the following location in the "timing" branch

```ini
; Using ModularSensors timing branch as of Mar 28, 2018: head, 36 commits ahead of 0.6.10
    https://github.com/EnviroDIY/ModularSensors.git#92a4e2628789fd92c7ae52f4901b1a489d10be74
```

closed 2018-04-06 by aufdenkampe