EnviroDIY/ModularSensors #113: Add support for external battery voltage measurement via voltage divider

aufdenkampe, 2018-04-02:
The use case is to measure the voltage on the 12V battery powering the connected ISCO (see #108 for ISCO trigger support), using this external voltage divider:
http://wiki.seeedstudio.com/Grove-Voltage_Divider/

SRGDamia1, 2018-04-02:
This is something I may also be able to work on, though I'm more interested in monitoring S::CAN batteries than ISCO's.

aufdenkampe, 2018-04-02:
@SRGDamia1, that would be great to have you contribute to this effort.
I've been thinking that it would be very cool to also have %charge value provided for each type of battery, based on the Amp Hour ratings and discharge curves that come with the battery spec sheets.

bschulz1701, 2018-04-02:
@aufdenkampe As a bit of a battery junkie, I might be able to help guide things in the right direction in terms of SOC (State of Charge) monitoring, if you would be interested

aufdenkampe, 2018-04-03:
@bschulz1701, sounds good. Let's see what we can do with that, but also as a second priority to other tasks getting done.

I talked to Sara yesterday, and she suggests we work off the head of the "timing" branch. I just created that new branch. https://github.com/EnviroDIY/ModularSensors/tree/battery_and_bucket. Let's put all new code and documents everything in there.

SRGDamia1, 2018-04-19:
This is currently done and up on master.

closed 2018-04-19 by SRGDamia1