EnviroDIY/ModularSensors #116: Ensure that logger can tell the difference between a 0 and no response from Tipping bucket

SRGDamia1, 2018-04-12:
We want to make sure that we got a real 0 count and not just that the trinket didn't respond properly.

SRGDamia1, 2018-05-03:
Should be fixed by 12ba3d5ef801080596b43e79e86430455cfef063

aufdenkampe, 2019-06-04:
@SRGDamia1, this issue may have come back with #252 & #256.

In sensor testing mode, everything works great, but when logging, I get -9999 values for both `RainCounterI2C_Tips` and `RainCounterI2C_Depth`.

This is my sketch: https://github.com/LimnoTech/EnviroDIY_deployments/blob/f8b2b5737706aaa0541d724653f033b489c23f7e/deploy-HH/BassLakeV2/BassLakeV2.ino

One clue is that in sensor testing mode, I also get -9999 values for the first round, then it works after that. I'm thus thinking that this might be an issue either with powering up or with averaging.

Below is an exceptionally verbose debugging output, in which:
```.ini
build_flags =
    -DMS_VARIABLEARRAY_DEBUG
    -DMS_VARIABLEARRAY_DEBUG_DEEP
```
and in which I switch to sensor testing mode after the first round of normal logging:
[BassLakeV2_debug_1.txt](https://github.com/EnviroDIY/ModularSensors/files/3254666/BassLakeV2_debug_1.txt)

aufdenkampe, 2019-06-04:
I just got it to work using a Mayfly v0.5 board instead of a Mayfly v0.3 board!

I figured this out when I tested my hardware on a different sketch using older version of the library (v0.21.4) that I know worked for my colleagues on their hardware. https://github.com/LimnoTech/EnviroDIY_deployments/blob/4d0f78b134a9a49816b5c99191de0aa7e21724be/deploy-RVSMRTDC/DoEE_McF_RG/DoEE_McF_RG.ino

Because I got the same answer, somehow got the idea to try a different version of the hardware. 
That is weird!

I'm not sure if we want to keep this open, because it might still be valuable to figure out what is going on.

SRGDamia1, 2019-06-05:
The variable array debugging isn't as helpful here as the sensorBase/variableBase/TippingBucket debugging would be.

SRGDamia1, 2019-06-05:
I think I might need to add another level of debugging called "gory" for the variable array..

aufdenkampe, 2019-06-05:
Moving discussion from #256 back to this issue, where it belongs:

From me:
> I let the BassLakeV2.ino sketch run overnight on a Mayfly v0.5 and other hardware, and after several hours the RainCounterI2C started logging -9999 values after working just fine for several hours. See https://monitormywatershed.org/sites/BassLakeWI/.
> 
> @SRGDamia1, any ideas on this? I'll try debugging mode for the sensor.

From @fisherba:
> @aufdenkampe, for what it's worth, the -9999's seem to correlate with lower voltage.

From @SRGDamia1:
> To see the full text printout, you need to turn on DataPublisherBase's debugging, or comment out lines 106 and 110 in dataPublisherBase.cpp. That can be changed back to always print if you prefer.

aufdenkampe, 2019-06-05:
@fisherba, thanks for noticing that clear correlation. I wonder if I restarted when I unplugged last night. When I re-plugged in this morning, it didn't magically help.

aufdenkampe, 2019-06-06:
I just figured out that these last issues that I had with the Mayfly v0.5 were because the particular board I had grabbed was set to I2C *switched* power! Facepalm!

The issues with the Mayfly v0.3 were real however.
Closing.

closed 2019-06-06 by aufdenkampe