EnviroDIY/ModularSensors #120: Add 'Contribute' section to ReadMe

aufdenkampe, 2018-04-15:
To help users understand how they might add new sensors to the library.

One part of that is to encourage users to learn basic and advanced Arduino programing concepts, by following these tutorials: 
- https://learn.adafruit.com/series/learn-arduino, lessons 0-3, 5, 6 and perhaps 8-9, 11-12, 
- https://learn.adafruit.com/adafruit-all-about-arduino-libraries-install-use
- https://learn.adafruit.com/multi-tasking-the-arduino-part-1
- https://learn.adafruit.com/multi-tasking-the-arduino-part-2

Also, to learn concepts of Object-Oriented Programming (OOP), including class-subclass structures, because ModularSensors utilizes these concepts heavily:
- http://v4.software-carpentry.org/oop/index.html

aufdenkampe, 2018-04-17:
As of 81652b70b0d492412fcd380f4d3b3322767a6951, we now have a placeholder for such a guide.

SRGDamia1, 2018-04-27:
I do need to write up some better explanation of *how* everything works under the hood (time stamps, linkages, etc) instead of just how to use the library.

SRGDamia1, 2018-12-21:
I've done some of this and updated ReadMe and wiki.  @aufdenkampe can you close this if you think it's ok.

SRGDamia1, 2019-02-20:
Still need more of the how.

SRGDamia1, 2020-12-15:
I think this is done.  I've also just updated the developer faq.
7773b8e1750e93ab984356fd3814135fc4e6c102

closed 2020-12-15 by SRGDamia1

aufdenkampe, 2020-12-15:
Yes! Thanks!

I wonder if we might better leverage https://envirodiy.github.io/LearnEnviroDIY/, or at least provide more links to it from this repo, to help migrate users over to Platform.io, which is a heavy lift for most to do on their own.

SRGDamia1, 2020-12-15:
You can certainly add more links to the ReadMe and getting started page.

@ScottEnsign is firmly of the opinion that asking beginners to use VSCode with PlatformIO is too much.  All of our workshops are still taught exclusively using the Arduino IDE.

aufdenkampe, 2020-12-15:
I agree with @ScottEnsign that a complete beginner should start with the Arduino IDE.
For that reason the [LearnEnviroDIY](https://envirodiy.github.io/LearnEnviroDIY/ ) tutorial starts with several hours of intro lessons using Arduino IDE.

That said, by the time a person is starting to actually use ModularSensors, it is nearly impossible for them to see long-term success if they continue to use Arduino IDE. I'm sure that they can do it in a workshop, for that instance in time where all the library versions are perfectly curated to work with each other, but I bet if you check in on them 3-6 months later that they can't get it to work.

On the other hand, if you teach them PlatformIO, then they can continue to operate independently for a long time. I have MANY examples of this, including a few who have taken Stroud workshops.

@ScottEnsign, let me know if it might be interesting to talk about this. It would be great to unify our training approaches.

mbarneytu, 2020-12-15:
> I'm sure that they can do it in a workshop, for that instance in time where all the library versions are perfectly curated to work with each other, but I bet if you check in on them 3-6 months later [...]

I very much agree!

neilh10, 2020-12-15:
My experience is the PlatformIO toolchain install is unbelievable simple, and combining with an examples/blinkey ; new beginner users,  with some graphical tutorials would find it just works.