EnviroDIY/ModularSensors #127: Create generic calculated variables

SRGDamia1, 2018-04-19:
That is calculated variables where the value depends on more than one sensor's variable result, or on one variable result and any other not-permanently-fixed value.

See discussion in 5e2c6f05c6d145c199ec5cb13f6d61090da3d7d4 and #125.

I think this can be done, but it will have to be done carefully so as not to have timing errors in arrays.  Perhaps a new Boolean flag could be added to the variable base object alerting the array to skip this variable until the end.

Note that there is already support for sensors that have calculated results based only on other values obtained in one reading from that sensor.  ie, the VWC from the 5TM is a result calculated by this library, but that result is nearly always a desired return from the sensor and the inputs for that calculation come entirely from one reading on the 5TM.  It would be easy to add a variety of other calculations, like altitude from any pressure sensor, but I don't think that particular calculation is nearly universally desired enough to both making it standard in this library.

fisherba, 2018-04-21:
This is helpful because we are also going to be adding a density/temperature correction for water level logging. We will follow the model you have created. Though we are more inclined to use the onboard temp for MS5803 (even though it's less accurate than , which makes the issue more like VWC from the 5TM.  

One recent recognition is that in ModularSensors we are using some terms, such as "variable", based more on ODM2/environmental science usage than its typical use in programming. Something to add to documentation about how ModularSensors works, to make it more friendly for new users and contributors.

SRGDamia1, 2018-04-23:
We don't need this feature, so I'm not going to be working on this much anytime in the near future, except maybe in between other work.  I think it will take a fair amount of thinking and testing to get it right. 

By "following the model I've created" do you mean the barometric correction example? Doing something like that should work for any form of calculation as long as the time it takes to update all of your sensors is less than two minutes. You could definitely make the header printed to the SD card prettier, if you wanted to, but it didn't seem like it was worth the effort to me.  As long as the code for doing the calculations happens in the loop after the updateSensors() function happens, you'll have access to any sensor result in the variable getValue() function. You can pick whichever temperature you want. 

I don't think that a calculated variable should be listed as coming from a sensor by default unless it is something the sensor is actually designed and specified for, as the 5TM is for vwc.

There should be some notes about the way I'm using the term variable. I'm so used to using the term that way, I never thought it was confusing. It honestly hadn't even occurred to me.

fisherba, 2018-04-23:
@SRGDamia1, thanks for your reply and for all of the help you've given to these projects during your in-between times. I really appreciate it. 

I uploaded a baro_rho_correction in the develop branch that effectively calculates the density correction, but I'm absolutely certain that it isn't organized as tidy as it could or should be (but I admittedly don't completely understand where to put everything). 

Speaking of variable, I was stumped in my density calculation by all that is contained in a "variable" and @aufdenkampe recognized that we needed to getvalue from the temp variable of interest. So that's working. But despite my effort to add rhoDepth to the csvToGo, it's not printing, which means I'm missing something. If you have a chance to see what I'm missing for the csv string, I would be extremely grateful.

fisherba, 2018-04-24:
It's all working. I think we can close this. Again, thanks for putting this together.

SRGDamia1, 2018-04-27:
I'm going to reopen this because I do want to keep it in mind as something that I think could be handled fully in the library, even though I'm not going to be able to do it any time soon.

SRGDamia1, 2018-05-22:
Closed by #153.  This was actually much easier than I thought it might be.  Wahoo!

PR #153 (2018-09-06): Implemented real calculated variables
Wahoo!  Closes #127 

@aufdenkampe and @fisherba please test out the baro_rho_correction example now.  You no longer have to do any messing with the text strings and can use the basic "log()" function.
closed 2018-05-22 by SRGDamia1