EnviroDIY/ModularSensors #128: Logging to EnviroDIY... suspected timing issues

fisherba, 2018-04-20:
Hi @SRGDamia1, 

I have a set of deployments at https://github.com/LimnoTech/EnviroDIY_deployments/
(I believe you have access to this repo) that contain several sensors: 
3 Yosemitech DO
2 Yosemitech conductivity
1 MaxSonar
1 5TM

When I run the tests individually for MaxSonar and 5TM, they work fine from this directory:
https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-BAF/testsketches
with lib_deps = EnviroDIY_ModularSensors@>=0.9.0

When I run a sketch with the whole group together, such as:
https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-BAF/CMP01
also lib_deps = EnviroDIY_ModularSensors@>=0.9.0
I have two problems: 

1) the MaxSonar reports -9999 but everything else works. 
2) the RTC set at the beginning of the sketch. Modem signals are usually in the 20-40% range, so not awesome, but @aufdenkampe suggested that this might be a timing issue (maybe that's the case for both issues?). 

Thanks for any advice you can give!

SRGDamia1, 2018-04-23:
I can try to look into the issue with the Maxbotix. Could you please open the MaxbotixSonar.h file and uncomment the line for debugging and recompile and post the output?

Does the modem give a stronger signal when you're running some other program?  I don't think a timing issue is likely, except if maybe the signal strength is variable or the sim800 chip registers better signal the longer it's connected to the internet. Again, open the loggermodem.h file and uncomment the line for both debugging and deep debugging and post the output.  I have no 2G signal at all in my office and very poor signal at home, so I'd be very happy to see numbers that high.

SRGDamia1, 2018-05-17:
Is this now working?

fisherba, 2018-05-17:
http://data.envirodiy.org/sites/NLP01/ is my current logging instance of this. It is spottier than I expected, but probably not much different than the 5TM. (I hadn't checked for a while.)

Now that we have the logger timing settled, I need to go back to these stations and upload new sketches.

SRGDamia1, 2018-05-17:
Okay, let me know.  Please close the issue if you think it's fixed.

SRGDamia1, 2018-05-18:
I added the "reality-check" filters to the 5TM.

SRGDamia1, 2018-10-10:
@fisherba - do you know if you're still seeing problems like this?

SRGDamia1, 2019-05-31:
Is this still an issue?

closed 2019-05-31 by fisherba