EnviroDIY/ModularSensors #135: Decagon 5TM returns data unconsistently

SRGDamia1, 2018-04-30:
@fisherba reports this for site NLP01.

SRGDamia1, 2018-04-30:
@fisherba  - This is really hard for me to debug seeing nothing other than some finally results that went to the website.  I don't actually have a 5TM sensor available right now.  It's possible that the 5TM isn't responding fast enough, that you have an older 5TM that isn't truly compatible with concurrent measurements (#115), or that something completely different is going on.  I pushed a tweak to the 5TMspottiness branch increasing the allowed response time and being more careful to set values to -9999, but if you really want me to be able to help much more, I need you to post the debugging logs so I can look at what might be going wrong.

To do that, checkout the new 5TM branch.  Then open SensorBase.h and find and uncomment the line ```#define DEBUGGING_SERIAL_OUTPUT Serial```.   Do the same in VariableArray.h and SDI12Sensors.h.  Run a "clean" in PlatformIO (the icon on the tool bar that looks like a trashcan) and then re-upload your program onto your board with all the sensors attached.  Open the serial port monitor and you should see many, many more lines printed to the screen.  Let the program start and then hit the button to go into "sensor testing" mode.  Let it run through a few sensor updates and then copy and paste the entire contents of your serial port monitor here.

SRGDamia1, 2018-04-30:
@fisherba - also make sure that all of your libraries are up-to-date.  To do that in Atom/PlatformIO open up a command terminal (the button with the ">_" next to the serial monitor button).  In the terminal type "pio lib update" and it should update the local version of all of your libraries.  You can also run "pio lib -g update" if you've had PlatformIO install any libraries in another global directory instead of the .piolibdeps folder for your project.

SRGDamia1, 2018-04-30:
Also, if you're for any reason running NeoSWSerial or any of the "tone" functions, they will conflict with the SDI12 library.  (You won't see any compile errors, just unexpected results.)

fisherba, 2018-05-02:
Hi @SRGDamia1, sorry for the delay, this is the first I've been able to run the test. After running the test a few times, it was only running one test, so I let it run for about a half hour (while preparing a kid breakfast and lunch) and it didn't run any additional rounds of testing. Here's the output: 

```
pio device monitor --port /dev/cu.usbserial-A50659OO --baud 57600
Beths-MacBook-Pro-3:EnviroDIY_deployments fisher$ pio device monitor --port /dev/cu.usbserial-A50659OO --baud 57600
--- Miniterm on /dev/cu.usbserial-A50659OO  57600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Now running NLP01.ino on Logger NLP01_Mayfly_170286
Logger timezone is UTC-6
RTC timezone is UTC
Initializing variable array with 20 variables...
The largest number of measurements to average will be 1.
There are 9 unique sensors in the group.
   ... Success!
Initializing logger NLP01_Mayfly_170286 to record at 15 minute intervals ...    ... Success!
Current RTC time is: 2018-05-02T06:59:37-06:00
Attempting to synchronize RTC with NIST
Beginning setup for sensors and variables...
Powering up sensors for setup.
Powering up sensors...
   ... Powering up Decagon5TM at SDI12-2_Pin7
Powering Decagon5TM at SDI12-2_Pin7 with pin 22
   ... Powering up MaxBotixMaxSonar at sonarStream_trigger-1
Powering MaxBotixMaxSonar at sonarStream_trigger-1 with pin 22
   ... Powering up YosemitechY504 at modbus_0x03
Powering YosemitechY504 at modbus_0x03 with pin 22
   ... Powering up YosemitechY504 at modbus_0x13
Powering YosemitechY504 at modbus_0x13 with pin 22
   ... Powering up YosemitechY504 at modbus_0x23
Powering YosemitechY504 at modbus_0x23 with pin 22
   ... Powering up YosemitechY520 at modbus_0x01
Powering YosemitechY520 at modbus_0x01 with pin 22
   ... Powering up YosemitechY520 at modbus_0x11
Powering YosemitechY520 at modbus_0x11 with pin 22
   ... Powering up EnviroDIY Mayfly at EnviroDIY Mayfly
Power to EnviroDIY Mayfly at EnviroDIY Mayfly is not controlled by this library.
   ... Powering up MaximDS3231 at I2C_0x68
Power to MaximDS3231 at I2C_0x68 is not controlled by this library.
Running setup functions.
   ... Set up of Decagon5TM at SDI12-2_Pin7 ...
Setting up Decagon5TM attached at SDI12-2_Pin7 which can return up to 3 variable[s].
It warms up in 200ms, is stable after 0ms, and takes a single measurement in 7536840ms.
1 individual measurements will be averaged for each reading.
Enabling interrupts for SDI12 on pin 7
   Activating SDI-12 instance for Decagon5TM at SDI12-2_Pin7
   Asking for sensor acknowlegement
      >>> 2!
      <<< ␀␗
   Decagon5TM at SDI12-2_Pin7 did not reply!
      >>> 2!
      <<< 2␘
   Decagon5TM at SDI12-2_Pin7 replied, unexpectedly
   Getting sensor info
      >>> 2I!
      <<< 213DECAGO\ 5TM   400
   SDI12 Address:2, SDI12 Version:1.30, Sensor Vendor:DECAGO\, Sensor Model:5TM, Sensor Version:400, Sensor Serial Number:
   ... succeeded. ...
   ... Set up of MaxBotixMaxSonar at sonarStream_trigger-1 ...
Setting up MaxBotixMaxSonar attached at sonarStream_trigger-1 which can return up to 1 variable[s].
It warms up in 160ms, is stable after 0ms, and takes a single measurement in 166ms.
1 individual measurements will be averaged for each reading.
   ... succeeded. ...
   ... Set up of YosemitechY504 at modbus_0x03 ...
Setting up YosemitechY504 attached at modbus_0x03 which can return up to 3 variable[s].
It warms up in 375ms, is stable after 8000ms, and takes a single measurement in 1700ms.
1 individual measurements will be averaged for each reading.
   ... succeeded. ...
   ... Set up of YosemitechY504 at modbus_0x13 ...
Setting up YosemitechY504 attached at modbus_0x13 which can return up to 3 variable[s].
It warms up in 375ms, is stable after 8000ms, and takes a single measurement in 1700ms.
[... 181 lines cut ...]
-- Collected result of reading 1 from YosemitechY504 at modbus_0x23 --
Putting 74.29 in result array for variable 0 from YosemitechY504 at modbus_0x23.
Putting 19.43 in result array for variable 1 from YosemitechY504 at modbus_0x23.
Putting 6.81 in result array for variable 2 from YosemitechY504 at modbus_0x23.
-- Success --
--- Finished all readings from YosemitechY504 at modbus_0x23 ---
- Starting reading 1 on YosemitechY520 at modbus_0x01 -
Starting measurement on YosemitechY520 at modbus_0x01.
- Success -
- Starting reading 1 on YosemitechY520 at modbus_0x11 -
Starting measurement on YosemitechY520 at modbus_0x11.
- Success -
-- Collected result of reading 1 from YosemitechY520 at modbus_0x01 --
Putting 260.43 in result array for variable 0 from YosemitechY520 at modbus_0x01.
Putting 19.10 in result array for variable 1 from YosemitechY520 at modbus_0x01.
Ignoring bad result for variable 2 from YosemitechY520 at modbus_0x01; no good results yet.
-- Success --
--- Finished all readings from YosemitechY520 at modbus_0x01 ---
-- Collected result of reading 1 from YosemitechY520 at modbus_0x11 --
Putting 268.79 in result array for variable 0 from YosemitechY520 at modbus_0x11.
Putting 18.84 in result array for variable 1 from YosemitechY520 at modbus_0x11.
Ignoring bad result for variable 2 from YosemitechY520 at modbus_0x11; no good results yet.
-- Success --
--- Finished all readings from YosemitechY520 at modbus_0x11 ---
```

fisherba, 2018-05-03:
Uploading tonight for deployment tomorrow and on the first of my 5TMs when I used the EnviroDIY_ModularSensors@>=0.9.0, every other iteration of sensor test sent good values for 5TM. In the spottiness branch, every iteration of sensor test seems to send good values. What you've done is an improvement. We'll see how it does in the wild!

SRGDamia1, 2018-05-03:
Okay.  I'm glad it seems to work.  In the log you posted above it looks like it wasn't working.

SRGDamia1, 2018-05-17:
So, is this fixed?

fisherba, 2018-05-17:
It's way better than the initial attempt. http://data.envirodiy.org/sites/NLP01/

SRGDamia1, 2018-05-17:
Does that mean fixed.. or do you still want me to try and work on this?  If it's fixed, please close the issue.

aufdenkampe, 2018-10-24:
According to @fisherba, the data from this summer gave a lot of bad values.

@SRGDamia1, is the fix from this commit https://github.com/EnviroDIY/ModularSensors/pull/173/commits/d347d50b8d7538a8ef79e2cc844651b8a44246c6 likely to improve stability?

SRGDamia1, 2018-10-24:
I have no idea.

aufdenkampe, 2018-10-26:
I guess my question should have been rephrased: What was the motivation for that fix/commit?

SRGDamia1, 2018-10-26:
That "fix" to the 5TM was just because a few commits earlier that day (a72fe9bfccc83b10c84441f8a648bde4cbdabc9a) I'd accidentally over-written the 5TM values with the D-Opto values when I was adding a few other notes.  So it was just a fix to something I'd broken the same day and shouldn't make any difference at all to the actual working of the 5TM.

aufdenkampe, 2018-10-26:
Thanks. That's helpful to know.