EnviroDIY/ModularSensors #140: Add to Modbus an AltSoftSerial "flush" or "end" function, to set pins low

aufdenkampe, 2018-05-04:
The problem is that when AltSoftSerial pins 5 & 6 are left HIGH, and power is shut down to our RS485 converter, there seems to be a bleed of 2.28V into the switched power lines.

Because Modbus Stop bits are HIGH, these pins are always left HIGH when the sensors shut down.

Let's figure out a way to get them to set low, before the power shuts down.

Unfortunately, if I set the pins to low in the loop using the following statements, then the AltSoftSerial streams seem to get shut down and I don't get any Modbus data.
```c++
    digitalWrite(5, LOW);   // Reset AltSoftSerial Tx pin to low, because Modbus Stop bit is HIGH
    digitalWrite(6, LOW);   // Reset AltSoftSerial Rx pin to low, because Modbus Stop bit is HIGH
```

SRGDamia1, 2018-05-04:
We can't put the functions in the Yosemitech/Keller files because they take an input of a "stream" not an AltSoftSerial. The begin and end functions are from serial objects, not their parent streams. Those two flush functions are fairly unique to AltSoftSerial. I definitely do not want to force people to use AltSoftSerial; I want people to have the option of any type of stream they want (ie, HardwareSerial, SoftwareSerial, NeoSWSerial, etc).

I'm guessing that when you set the pins low, it also deactivates the capture/compare functions on the pins that AltSoftSerial depends on. That's also what the end function explicitly does, along with running both flushes. (Though I'm not sure if the pins go low after being deactivated.) Try moving the begin function for AltSoftSerial from you setup() to the start of your loop so it can regain control of the pins again at the start of each loop.

SRGDamia1, 2018-05-10:
Did adding the begin for AltSoftSerial to the start of the loop work?

aufdenkampe, 2018-05-16:
@SRGDamia1, your idea worked!

I expanded out the `EnviroDIYLogger.log()` function, as done in `baro_rho_correction.ino`, and added this line:
```
        // Restart the stream for the modbus sensors, because we set pins LOW, below
        modbusSerial.begin(9600);
```
just before:
```
        // Send power to all of the sensors
        EnviroDIYLogger.sensorsPowerUp();
```

Also, just after this:
```
        EnviroDIYLogger.sensorsPowerDown();
```
I added
```
    // Reset AltSoftSerial pins to LOW, to reduce power bleed on sleep, because Modbus Stop bit is HIGH
    digitalWrite(5, LOW);   // Reset AltSoftSerial Tx pin to LOW, because Modbus Stop bit is HIGH
    digitalWrite(6, LOW);   // Reset AltSoftSerial Rx pin to LOW, because Modbus Stop bit is HIGH
```

Power saving accomplished!
This should address one of the two issues in #148.

closed 2018-05-16 by aufdenkampe