EnviroDIY/ModularSensors #141: New feature request: separate what writes to SD card vs to internet data portal

fisherba, 2018-05-07:
To economize data usage and make the data portal easier to read for clients, I have a need to log more information to the SD card than we send to the data portal. 

For example, many of the sensors have temperature on-board, coupled with their intended parameter. If a sensor is giving us trouble, seeing the temperature is somewhat helpful for understanding the extent of sensor failure, but it really does not make sense to pay for data for every sensor, especially for some of my current work, where the client will benefit (visually and financially) from a simple list of the parameters of interest. However, I see a substantial benefit of logging all the variables on the SD card for data verification/validation. 

I understand that expanding the `EnviroDIYLogger.log()` function similar to how it's done in the "baro correction" sketch  is one way we could do it now, but there's probably a cleaner, more sustainable and user-friendly approach for the main modular code.

SRGDamia1, 2018-05-08:
I think you should actually be able to do this relatively easily, going between the "double logger" and baro examples. So you'd create two loggers and variable arrays, one with everything and the other with only what you want to send out to the internet. Make sure when you create the variable arrays that you are pointing to the exact same variables in the second array as you used in the first. (That is, create the variables with names like in the baro correction example (not with the "new" statement) and then call the same variables in both arrays where it applies.)

In the setup, you need to run the "init" for *both* loggers.  You only need to create/setup one modem.  Attach that modem to both loggers if you want the time sync to happen at the begin, otherwise you can just attach it to the logger with the short list. You can also attach the same token/feature id, if you want it in the header.  Then you can just run the "begin" on the logger than had the full list and that will set everything else up.  (If you run the begin, you don't have to run the other functions for setting up the log file.)

OK, so on to the loop function. Do the check interval and power up your modem. Only run the power-up/wake/update/sleep/power-down on the logger with the full list. Then run the "logToSd" on that logger. Immediately after than, connect to the internet and run only the "postDataEnviroDIY" function with no arguments on the second logger. Then disconnect and put the modem and processor to be.  As long add you ensure you have the exact same variables, called by name, in the "short list" array, those variables will all have their values updated when the long list runs the sensor updates.  Then the short list logger can automatically create and send the json with no ugly text manipulation involved at all.

I'm fairly certain that should work as long as you're saving everything. Actually, it should work as long as you have at least one variable from every sensor in the array that you run the update function on.  If you have completely separate sensors to log and send, that works too, just run the updates on both arrays. The only thing that won't work is if you want to do a weird split and where you have some sensors that are logging and not sending half their variables and other sensors that are sending and not logging half their variables. I think that's a bad plan for all kinds of reasons.

SRGDamia1, 2018-05-08:
The double logger example is a bit out of date, I'll try to update it.

SRGDamia1, 2018-05-08:
Let me know if my explanation doesn't make sense (or the code doesn't work). If it works, I don't want to try to create some system of flagging variables as to whether data is saved or sent somewhere, etc.

SRGDamia1, 2018-05-10:
Ok, check out the new "data_saving" example.  It shows doing what you want for a group of Yosemitech sensors.

closed 2018-05-10 by SRGDamia1