EnviroDIY/ModularSensors #142: loggingInterval not working, logging faster than designated

fisherba, 2018-05-08:
My initial deployment of the following stations had the logging interval in the sketch set at 
`const uint8_t loggingInterval = 30`, but the stations logged every 2 minutes. Below I list each station with its sketch and libdeps for that deployment: 

CMP01 logged at 2 minute intervals regardless of which version of Modular I used, and is still an issue (the SIM card for this station is currently paused). 
http://data.envirodiy.org/sites/CMP01/
https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-BAF/CMP01

First deployment of CMP01 used ModularSensors *5TMspottiness* branch as of May 2, 2018, but I didn't end up deploying a 5TM at this site (always saturated). When I tried this libdep with 15 minute intervals, it still logged every two minutes, so I tried three other branches, listed below. 
    
```
https://github.com/EnviroDIY/ModularSensors.git#3e4afedec8bc5da908e4c4e65280abb4a58e2cc0
      https://github.com/PaulStoffregen/AltSoftSerial.git
      https://github.com/EnviroDIY/SoftwaterSerial_ExternalInts.git
```

I also tried the following versions of Modular and still had the result of 2 minute logging with this sketch:
```
; Using ModularSensors *develop* branch as of Apr 27, 2018:
;    https://github.com/EnviroDIY/ModularSensors.git#efdb763a84d59d544f42aeaa76531a1d59344938
; Using ModularSensors *master* branch as of May 7, 2018: v11.3
      https://github.com/EnviroDIY/ModularSensors.git#e1667f4fcd2649c03c75adb7ed2f144cb9b29881
; Using ModularSensors *develop* branch as of Apr 25, 2018: (for CMP01?)
;    https://github.com/EnviroDIY/ModularSensors.git#cedb6274a6c782114c838032acb3e2e960aaba6b
```

NLP01 logged at 2 minute intervals during the first deployment when set at 30 minute interval. Now that it's set at 15 minutes, the station logs every 11 minutes. 
http://data.envirodiy.org/sites/NLP01/
https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-BAF/NLP01
This deployment used the following libdeps for both deployment attempts:
```
; Using ModularSensors *5TMspottiness* branch as of May 2, 2018:
    https://github.com/EnviroDIY/ModularSensors.git#3e4afedec8bc5da908e4c4e65280abb4a58e2cc0
      https://github.com/PaulStoffregen/AltSoftSerial.git
      https://github.com/EnviroDIY/SoftwaterSerial_ExternalInts.git
      https://github.com/todbot/SoftI2CMaster.git
```

NLP02 and CMP02 are the redox loggers at each site. Both logged at 2 minute intervals during the first deployment when set at 30 minute interval. Now that it's set at 15 minutes, the stations log every 11 minutes. Libdeps didn't change between deployments. 
http://data.envirodiy.org/sites/NLP02/
https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-BAF/NLP02
and
http://data.envirodiy.org/sites/CMP02/
https://github.com/LimnoTech/EnviroDIY_deployments/tree/master/deploy-BAF/CMP02

```
lib_deps =
; Using ModularSensors *PaleoTerraRedox* branch as of May 2, 2018: head, 7 commits ahead of 0.11.3
    https://github.com/EnviroDIY/ModularSensors.git#70c5b551559e733ff08e35eeaa53888070c05023
    https://github.com/todbot/SoftI2CMaster
```

SRGDamia1, 2018-05-08:
I think I've found the problem.

For the first 10 readings the logger is designed to go every 2 minutes (so for 20 minutes).  (#95)

BUT... on top of that I'd assigned the logging interval in seconds to be an 8-bit integer, which has a max value of 255.  So any logging interval over ~4.5 minutes would have given unexpected results.  Since I very rarely run things on my desk for more than the first 20 minutes, I didn't ever notice the problem. I changed that to a 16-bit integer, so hopefully it should now keep proper logging intervals above 5 minutes.

I'll leave one on my desk overnight and try to remember to actually look at the SD card in the morning to see if it worked.

SRGDamia1, 2018-05-08:
Thank you so much for the detailed report!

fisherba, 2018-05-08:
Thank you! I look forward to hearing your results. Is there a simple way that we can make the first 10 readings for 2 minutes a user-defined option? Either to turn on/off that feature (which is a good feature when a deployment is working) or select the number of readings to do every 2 minutes?

aufdenkampe, 2018-05-08:
@fisherba, thanks for detailing those issues.
@SRGDamia1, thanks for fixing the code. Let us know when/where you make that fix commit! 

I'm going to need to update all of my deployed code!\
Any idea how long that bug has been there?

aufdenkampe, 2018-05-09:
@SRGDamia1, how did your test go last night? Any progress on implementing this fix?

SRGDamia1, 2018-05-09:
It seemed to run correctly at half-hour intervals for me last night.  I'll run it again tonight on a different board just to check.

aufdenkampe, 2018-05-10:
@SRGDamia1, thanks for fixing and testing this. We'll let you know if that solves our problems.

commit c6f1956b (2018-05-11): Changed logging interval from uint8_t to uint16_t

Hopefully fixes #142
commit 63edc2b3 (2018-05-11): Changed logging interval from uint8_t to uint16_t

Hopefully fixes #142