EnviroDIY/ModularSensors #143: Fix `barometric_correction.ino` to allow for Sensor Testing Mode

aufdenkampe, 2018-05-08:
The [barometric_correction.ino](https://github.com/EnviroDIY/ModularSensors/blob/master/examples/barometric_correction/barometric_correction.ino) example was created by @SRGDamia1 to demonstrate how to calculate and record variable values (see #127 and discussion in 5e2c6f05c6d145c199ec5cb13f6d61090da3d7d4 and #125) by unpacking the `log()` function into the .ino file `loop()` function.

At the present, it is a bit funky and needs some debugging. For example, it doesn't allow for the "Press Button to enter Sensor Testing Mode" functionality. When the button is pressed, the system hangs up.

There are other bugs that @fisherba and I could list out with a little additional testing on our side.

SRGDamia1, 2018-05-09:
Right now that example is written with a "check for debugging mode" at the end of the set-up instead of the button interrupt functionality.  So if you hit the button in the five second window after set-up finishes, you should get to debugging mode.  That part of the set-up is actually unique to each sketch.  I think it's kind-of hap-hazard right now which sketch has which functionality.  If the setup uses the check instead of the interrupt, when you hit the button any other time, nothing should happen at all.  It should not lock-up.  The button press would wake it, but unless it's on the right logging interval, it should just go back to sleep

SRGDamia1, 2018-05-10:
I don't have the sensors to check if the calculation is really happening correctly, but in it's current state the sensor-testing mode is acting exactly as I'd have expected it to:  it only enters with a button press at the end of setup, it only displays the not-calculated variables, and pressing the button any other time does nothing.

SRGDamia1, 2018-05-10:
91b83c401452eaff49012f924403dd8bfaa20e6c add's the button-press functionality to the baro correction and rho correction examples.

aufdenkampe, 2018-05-10:
@SRGDamia1, thanks for that update and your tests.
@fisherba and I should test this before we close this issue, as we were having several issues with the sketch hanging up when we pressed the button at the expected time.

aufdenkampe, 2018-05-14:
I should have tested if this fix worked, when working on #147! 
Perhaps @fisherba can do this.

aufdenkampe, 2018-05-15:
Tested this morning, and it works!

closed 2018-05-15 by aufdenkampe