EnviroDIY/ModularSensors #149: Allow i2c sensors to be fed a two-wire instance rather than always using default

SRGDamia1, 2018-05-16:
Similar to the way all serial sensors require a stream as input, instead of building the stream internally.  This is not supported by all of the sensor libraries, but could be implemented where it would be supported.

This could also solve the problem of the incompatibility of SoftI2CMaster with the Zero.

SRGDamia1, 2018-05-17:
Reading around, it doesn't look like any software I2C library supports this kind-of functionality.  So this really would only be helpful for SAMD boards or with multiple possibilities for hardware I2C.  To allow software I2C, some specific Software I2C library would have to be built in (and thus required) for each I2C module.

SRGDamia1, 2019-09-19:
I put in a PR to the AVR core to make the functions virtual.

aufdenkampe, 2019-09-19:
@SRGDamia1, you rock!

SRGDamia1, 2020-12-15:
This is now done, where possible (which isn't that many places).

closed 2020-12-15 by SRGDamia1

aufdenkampe, 2020-12-15:
Did your PR to the AVR code get pulled into a release?

SRGDamia1, 2020-12-15:
No.  It's here: https://github.com/arduino/ArduinoCore-avr/pull/288

aufdenkampe, 2020-12-15:
Thanks for that update and the link.