EnviroDIY/ModularSensors #162: Support new Digi LTE-M Xbee module

aufdenkampe, 2018-09-06:
The new Digi XBee3 Cellular LTE-M radio module is finally available for purchase. 
https://www.digi.com/products/xbee-rf-solutions/embedded-cellular-modems/xbee3-cellular-lte-m

@gcutrell and I each have one and would like to prioritize getting it to work with the Mayfly.

@SRGDamia1 and @s-hicks2, any thoughts or guidance as we get started?

SRGDamia1, 2018-09-07:
There shouldn't be anything needed for this; it should work out of the box using #define TINY_GSM_MODEM_XBEE.  One came in for us while I was out, but I haven't tried it yet.  Let me know if it doesn't work.

gcutrell, 2018-09-07:
@SRGDamia1 When using #define TINY_GSM_MODEM_XBEE our program gets hung up and never progresses when it is trying to update with NIST.

aufdenkampe, 2018-09-07:
@SRGDamia1, I had thought that was the case, so it's great to have that confirmed. 
We'll use this issue to post progress.

SRGDamia1, 2018-09-17:
@gcutrell - Can you turn on the "debugging" for the modem and post a log?  I'm not having problems with the XBee causing hanging.

@aufdenkampe - Have you tried the LTE-M XBee yet?  I was playing with it this weekend and I cannot get it to connect to the network with a Hologram SIM.  It's not a library problem - I'm not able to get it to connect even putting it in bypass mode and directly querying the cellular chip.  I only see one cellular signal available on it (310-410 (Cingular Wireless)) and that network denies the sim registration.

calkillen, 2018-09-17:
I'm very interested in getting this working.  I have 4 Mayflys and 4 LTE-M Xbees with Hologram SIMs.  The code I'm using from EnviroDIY never connects to AT&T Mobility - responding with "22" to "AI".  Let me know if/how I can help make progress on this.

SRGDamia1, 2018-09-18:
I can't get past the "22" state either.  I posted a question on [Hologram's forum](https://community.hologram.io/t/using-digi-xbee3-with-hologram-sim/2082).  It looks like similar questions have been asked at least 3 times in the last few months, though, without any solutions.

calkillen, 2018-09-18:
I just heard from Tech Support at Digi.  He said:

Thank you for contacting Digi. No I don't have anyone that is supporting the Arduino platform or code. But what I can tell you is that I do have customers who are using Hologram SIM's.  However, you do need to set the proper APN and you do need to provide enough current for the module to power up properly.  Note that most Arduino hardware is designed to run a 2.4 GHz low power XBee module wanting about 400mA of current at 3V.  The XBee Cellular module is going to want about 2A at 3.3 to 5V to properly power up and connect to a tower.  To help narrow down the possible cause, what I would suggest is that you start by mounting the module in an XBIB-U-DEV Rev F board with an external power source to see if that is not your issue.

I don't know if the Mayfly is up to the power requirement.  Either way,I'm probably going to get the development board to see if I can make progress here in Michigan.

gcutrell, 2018-09-26:
@SRGDamia1 Below is an output of the TINY_GSM_DEBUG directly below 'Attempting to synchronize RTC with NIST' using the TINY_GSM_MODEM_XBEE. This continues infinitely which is why my program never gets past the syncing using the new modem and a hologram card. Program works fine using a TINY_GSM_MODEM_SIM800. 

[6561] +++
[6561] ### ..: OK , ERROR ,  ,  ,
[6668] <<<  OK
[6674] ### AT:
[6674] ### ..: OK , ERROR ,  ,  ,
[6678] <<<  OK
[6807] +++
[6807] ### ..: OK , ERROR ,  ,  ,
[6815] <<<  OK
[6821] ### AT: AI
[6823] <<<  FF
[6832] ### AT: CN
[6832] ### ..: OK , ERROR ,  ,  ,
[6838] <<<  OK

[6965] +++
[6965] ### ..: OK , ERROR ,  ,  ,
[7071] <<<  OK
[7079] ### AT: EE 2
[7081] ### ..: OK , ERROR ,  ,  ,
[7086] <<<  ERROR
[7094] ### AT: CN
[7094] ### ..: OK , ERROR ,  ,  ,
[7098] <<<  OK
[7227] +++
[7227] ### ..: OK , ERROR ,  ,  ,
[7333] <<<  OK
[7342] ### AT: EE 2
[7344] ### ..: OK , ERROR ,  ,  ,
[7350] <<<  ERROR
[7356] ### AT: CN
[7358] ### ..: OK , ERROR ,  ,  ,
[7362] <<<  OK
[7489] +++
[7489] ### ..: OK , ERROR ,  ,  ,
[7598] <<<  OK
[7606] ### AT: EE 2
[7606] ### ..: OK , ERROR ,  ,  ,
[7612] <<<  ERROR
[7618] ### AT: CN
[7620] ### ..: OK , ERROR ,  ,  ,
[7624] <<<  OK

SRGDamia1, 2018-09-26:
Look at your modem constructor.  There are both wifi and cellular xbees, you need to use the cellular constructor for the cellular version.  You must be using the wifi one.  Also, update TinyGSM.

gcutrell, 2018-09-28:
That was it. Thanks.

aufdenkampe, 2019-01-29:
I'm reviving this GitHub issue, given the significant interest in getting the Digi XBee3 LTE-M module to run on a Mayfly. The short story is that to our knowledge, no one has gotten it to work with this [ModularSensors](https://github.com/EnviroDIY/ModularSensors) library.

There has been, however, some successes with lower level libraries and a considerable amount of discussion in these two forums:
-  https://www.envirodiy.org/topic/gprsbee/
- https://www.envirodiy.org/topic/trouble-initializing-xbee3-lte-m/

SRGDamia1, 2019-01-29:
@aufdenkampe - it works.

aufdenkampe, 2019-02-01:
@SRGDamia1, thanks talking to me on the phone Wednesday afternoon, explaining to me all your recent advances and successes.

From what I understand, to get the Digi XBee3 LTE-M module to work on the Mayfly with Modular Sensors, it is necessary to:

1. **Use "Bypass" mode** to directly interact with the UBLOX modem chip, bypassing Digi's chip, rather than using the standard "Transparent" mode, which would cause full freezes (probably due to power instabilities because there isn't a capacitor chain on the Digi LTE-M module or the Mayfly).
2. **Turn off the NBIoT radio** to speed up the connection, by setting the network technology (“N#) to 2 (LTE-M only).
3. **Turn off Power Saving mode**, because this mode interacts with the network regarding when it will wake again, and if that doesn't match the logging interval that we give it, it becomes impossible to wake up the modem when we need it.

@SRGDamia1 has implemented these changes in the current develop branch:
https://github.com/EnviroDIY/ModularSensors/commits/develop

She's had it working since her Jan. 29 commit, https://github.com/EnviroDIY/ModularSensors/commit/be9e2eb47f1d919579703d7dfa3e416e2ab632d1, but she's been working on improvements since then.

This solution still isn't perfectly stable, because of the following hardware issues (which could all be addressed by an Xbee adaptor shield that I would like to build):
- Neither the Digi XBee3 LTE-M module or the Mayfly have the recommenced capacitor chain on the Xbee power line. This is the likely culprit for brownouts.
- Benefit of optionally switching Xbee-Mayfly pin connections for the "status", depending on radio module
- The Mayfly doesn't have a means to separately reboot the Xbee module
- The Mayfly doesn't set all the unused pins to Ground, which would be ideal for broader radio compatibility, with for example the Nimblelink Skywire.

The new SODAQ SARA UBEE R410M radio module, https://shop.sodaq.com/sodaq-sara-ubee-r410m.html, has the same UBlox SARA-R410M radio chip as the Digi XBee3 LTE-M module​, but it also has a built-in capacitor chain, which could substantially improve the stability when deployed with the Mayfly.

@SRGDamia1, what did I miss?

SRGDamia1, 2019-02-02:
You don't "need" to use bypass mode - I just found it was faster and somewhat more stable.  I spent more time with it and the lock ups weren't as bad as I remembered.  But I'd still probably go with bypass.

And it's actually been working much, much longer than this January.  I just hadn't gotten it optimized until now really.

closed 2019-02-08 by SRGDamia1