EnviroDIY/ModularSensors #163: Enable automatic restart after battery get solar recharged

aufdenkampe, 2018-09-09:
Many of our Mayfly stations have batteries drop below operating voltages, especially during winter storms, etc. We would like the Mayfly to automatically restart as soon as solar charging brings the battery voltage back into operating ranges. Note that solar charging doesn't require the Mayfly to be on.

Mayfly stations with more streamlined programs (i.e. those not dependent on ModularSensor codes), will automatically restart. In other words, the hardware allows it.

We believe that the reason we don't see see auto-restart with the ModularSensors is that the various automatic checks that run during startup use too much power for it to proceed, and drain the battery to below the threshold BEFORE it can record any measurements. 

Therefore, this fix will require us to implement a number of power-saving features during setup functions. 

Top on the list of power consuming setup functions is the elimination of the NIST sync feature, which although handy, has not been working well, even with the latest modifications. See #136 for discussion on why the feature isn't worth the time or battery drain that it seems to require.

There may be other power-draining functions that we could eliminate or reduce during startup.

This is a relatively high priority these next weeks, as we've already seen some remote stations die during some long thunderstorms. We'll be making a relatively big effort to upgrade solar panels and batteries, and it would be great to upload new sketches at that time.

SRGDamia1, 2018-09-15:
I don't anything about it, but the board does have a "brown out detector."  I'll try to read up if that's anything that we could use.

SRGDamia1, 2018-09-17:
Using the brown out detector seems to require the boards.txt to be modified, so I don't think we want to try to use it.

SRGDamia1, 2018-09-17:
There's some brain-storming below this, but, in short, I think the modem is the only thing really worth going through much effort to cut out of the set-up.

For that, you can use the battery voltage to switch between the begin/beginAndSync functions in the setup and make sure that whenever you run begin (w/o sync) you're running it on an array that doesn't include the modem.

I could also look into using the modem power functions to use AT commands to put the modem into reduced functionality modes.  I think this will be trickier, though.

Currently, the setup function for most internet loggers is:
- begin the serial ports [very quick, doesn't require any power except to the main processor]
- set some pin modes [very quick, doesn't require any power except to the main processor]
- set some settings [very quick, doesn't require any power except to the main processor]
- create a logger file [not sure how much of a power draw this is, but I don't think it can be skipped] 
- run the setup for the variable array [multi-step process]
    - run the power-up functions for all sensors
    - run the individual setup functions for all the sensors [this includes the modem]
    - run the power down functions for all sensors
- connect to the internet for a clock sync [power hungry!!!]
- set some interrupts [very quick, doesn't require any power except to the main processor]
- go to sleep [zzz...]

So, cutting the clock sync would be easy, but once we get into the individual sensors.. it gets hairy.

For a bunch of sensors, the set-up function is really just setting more pin modes on the main processor.  That must happen during the main setup function, but it's very fast and doesn't require the sensor to have power.

For another group of the sensors, the set-up function sets pin modes and then goes on and pings the sensor to make sure it's really there and talking.  The ping is really just a hello to verify connection.  When in a variable array if the sensor doesn't respond, the array function pings up to 5 times and then warns that set-up failed.  The only result of the sensor not responding is that warning.  If we cut out the "hello" ping for these sensors, they wouldn't need to be powered up.

For the last group of sensors, the set-up function actually asks the sensor to do something that only needs to be done once - like give information on internal coefficients or ensure it is in the correct mode.  If the sensor isn't powered to talk back at this point, chances are that it won't be able to make measurements later.  The MPL is in this group.  So, unfortunately, is the modem.

I could move the power-up to part of the individual sensors set-up function where it's needed instead of broadly powering up the array.  This would mean the power would never go on to the sensors if there wasn't one that needed it, but it would also mean the power would flicker on and off if there are several that want to say hello and the optimization in the set-up to look for the next sensor to have warmed up would be useless.  And even then, this only kicks the problem 5 minutes down the line until the next logging interval.  Since the logger isn't likely to have charged much in that time, I don't know if there's any savings in this approach and it would be a bunch of hassle.

The good thing is that, for the most part, just turning the sensors on doesn't suck that much power.  Older Yosemitech's spun their the cleaning brushes as soon as they got power, but I think I'm the only one with one old enough.  The modem is the only big offender because it starts looking for a network right away.

aufdenkampe, 2018-09-21:
@SRGDamia1, thanks for that in-depth analysis and thoughts on how to improve. I agree that turning off modem sync during startup is the first step. However, I'm confused on how an IF statement (related to battery voltage) would work in a startup/setup function, given that this would be needed after the system dies due to low battery.  I would like to have the sync only happen in the loop. Is that what you had in mind?

BTW, the latest version of the Yosemitech Y4000 sonde also has the brush spin immediately when it is powered up. I'm guessing that will change, but we have 7 sondes of the old kind.

SRGDamia1, 2018-09-21:
```cpp
float getBatteryVoltage(const char *MFVersion = "v0.5b")
{
    float rawBattery = analogRead(A6);

    if (strcmp(MFVersion, "v0.3") == 0 or strcmp(MFVersion, "v0.4") == 0)
    {
        return (3.3 / 1023.) * 1.47 * rawBattery;
    }
    if (strcmp(MFVersion, "v0.5") == 0 or strcmp(MFVersion, "v0.5b") == 0)
    {
        return (3.3 / 1023.) * 4.7 * rawBattery;
    }
}
```

In ```setup()```:
```cpp
    // Begin the logger
    if (getBatteryVoltage(MFVersion) > 3.6) EnviroDIYLogger.beginAndSync();
    else EnviroDIYLogger.begin();
```

In ```loop()```:
```cpp
void loop()
{
    // Log the data
    if (getBatteryVoltage(MFVersion) > 3.6) EnviroDIYLogger.logAndSend();
    else EnviroDIYLogger.log();
}
```

SRGDamia1, 2018-09-21:
You can't use ```Variable *mayflyBatt = new ProcessorStats_Batt(&mayfly)``` and then ```mayflyBatt ->getValue()``` as a condition for the setup because ```mayflyBatt ->getValue()``` won't return a valid value until the observer is linked, which happens during the setup.  So you need a different function.  I can think about a way around it.

You could use ```mayflyBatt ->getValue()``` as a condition in the loop and then always use ```begin()``` in the setup function instead of ```beginAndSync()```.

aufdenkampe, 2018-09-21:
Yes, I was thinking along those same lines:

> You could use ```mayflyBatt ->getValue()``` as a condition in the loop and then always use ```begin()``` in the setup function instead of ```beginAndSync()```.

Thanks!

SRGDamia1, 2018-09-21:
I'm not sure if it's possible to make the observers set in the constructors.  I'm guessing yes, but I'm not sure.  I'll have to read and experiment.  That would allow the get value call to work immediately.

aufdenkampe, 2018-09-21:
I've been looking at https://github.com/EnviroDIY/ModularSensors/commit/525b43aa8932d203bc0a77cb81fba2148cb144d0 and I'm confused. I see the renamed `LoggerEnviroDIY::begin(void)` to `LoggerEnviroDIY::beginAndSync(void)`, but I don't see a separate `begin` function. Am I looking in the wrong place?

SRGDamia1, 2018-09-21:
The "begin" is the function it inherits from a logger base object.

gcutrell, 2018-10-17:
@SRGDamia1 's getBatteryVoltage code works as advertised! Setting a voltage above the battery minimum allows the Mayfly (with non-power hungry devices) to store enough power to supply a modem connection. 

@aufdenkampe I can still see how this would not solve the issue if someone has a power hungry device that will continually draw down the battery as soon as the mayfly gets enough power to boot. 

Novice question... Could you do something like 
`bool startupHigh = false;
    do {
      Serial.print(getBatteryVoltage(MFVersion));
      if (getBatteryVoltage(MFVersion) > 3.3) {
        startupHigh == true;
        if (getBatteryVoltage(MFVersion) > 3.4) EnviroDIYLogger.beginAndSync();
        else EnviroDIYLogger.begin();
      }
    } while (startupHigh == false);`

to keep the mayfly from "starting" until the battery voltage is above a certain threshold?

SRGDamia1, 2018-10-17:
You could do that, but your charging rate would be faster if you actually put the logger back to sleep between checks.

SRGDamia1, 2018-10-17:
You could create a minimal begin-type function that will only set up the sleep mode and sleep interrupts on the main processor and run that in the setup and create another minimal "don't log" type function that just restarts sleep.  Then you could just keep running "don't log" until your battery level is high enough.

One of the problems with doing it that way is that when you go to sleep, you resume in the loop, not in the only-ever-run-once set-up.  So if you don't set up your sensors in the set-up and then resume logging at some point later, your sensors may never get set up at all.

hmm.. hmm.. OR, you could make your loop function just a tiny bit smarter and make use of the loggers (internal, protected)_numTimepointsLogged value and run the sensor setup any time that value is 0...

neilh10, 2018-10-17:
In a previous life, with a solar powered device, using a mega2560 I defined a state machine for power up. 
For the arduino concepts this is extending setup() in to different phases and I think that also means changing the way initialisors are run.  
There is a basic machine intialisation first, that is absolutely minimal in determining available power - and then there would be pre-sensor processing including determining if there is valid time, followed by sensor initialisaton.  This could be mean being more defined as to when the sensor class initializers are called.  
Battery voltage is also notoriously bad indicator of power availability.  If you take ***too much*** power off a discharged battery it just doesn't supply it and hopefully the processor is reset through brownout.  So that means keeping a history of warm resets and dumping them on startup to be able to figure out an algorithm that works.
Another algorithim was to delay heavy power users (one minute of cellular) until sufficient solar derived joules had been stored.
Practically speaking - is a logger useful if it hasn't got valid time? If it hasn't got valid time, should it get it before anything else is attempted. Thats why I used a state machine at startup.

neilh10, 2018-10-18:
Some data from a test Mayflyrev0.4 just running Xbee Wifi overnight with a charged LiIon 3.7V 500mA (Adafruit).  
All it shows is the voltage drop knee for when power is lost. Of course if it was connected to a solar panel, there would be a weak solar signal at sunrise which is usually seen as a little voltage increase, and the solar power slowly increases, before the full power of the solar panel would come on a couple of hours later.

For this data I removed the ext solar power at 10/17/2018  7:35:00 PM  - it ran out of power at  10/18/2018 2:10am
From this mornings measurements, it appears to be consuming ~100mA all the time, so for 500mAHr battery it lasted about 6.5hrs - a bit better than expected from the battery's rating.
https://data.envirodiy.org/sites/LCC45test1/
Here are some snaps 1) of the VbattProc from data.EnviroDiy of the last couple of days. The sawtooth is the charge/relax cycle
[image]
2) the same period as recorded by the more accurate ExtVoltage/AA0 with 100K+100K R for divider.
[image]
3) The same period with the applied solar voltage (USB 5V source) as recorded from ExtVoltage/AA1 with 100K+100K R divider
[image]

4) From the SD card a more focused view of the voltage droop before it looks like the LiIon Batt did a low-voltage cuttoff. When I measured the voltage with an external meter it showed 0V, indicating the LiIon for self protection did the low-voltage cutoff.
[image]

Here is the full data set from the SD card and also the downloaded version of the data from data.EnviroDIY - 
note1: I've implemented a seqNumber that increments up. In the sdCard its called seqNum 1-128
for the data.EnvDiy its "Digi_WiFi_SignalPercent" 1-128, then restart1-3

[nh07e_2018-10-17.xlsx](https://github.com/EnviroDIY/ModularSensors/files/2492821/nh07e_2018-10-17.xlsx)

note2 the downloaded data from data.EnviroDIY appears to have offset the new added sensors to date recording. ie all readings start at the first date:time row even though some sensors added later.
[181018mayfly1_LCC45test1_TimeSeriesResults.xlsx](https://github.com/EnviroDIY/ModularSensors/files/2492822/181018mayfly1_LCC45test1_TimeSeriesResults.xlsx)

--

neilh10, 2018-12-05:
I'm seeing my bench test system with a small LiIon 0.5Ahr run out of power pretty quickly, and then it doesn't cope with the low power availability. 
I have it set an external supply to simulate "low solar" with a max ~ 50mA, simulating what might happen when a battery is running low, and has some solar power.
What I would like it to do, is keep logging essential functions as long as possible, and then when power availability is UNUSEABLE, to have as little drain on the LiIon as possible, till there is sufficient power availability.

It seems to me that TimeSync is only valuable if there isn't a valid realtime. If seems there is a good time, then should let the standard method of updateing time be used.

I've separated  EnviroDIYLogger.beginAndSync() to .begin();  & .timeSync();

I'm looking to go to this startup algorithm. 
```
  batteryV=getBatteryVoltage(MFVersion);
/*LiIon BATTERY condition as measured by V - actual values may be optimized
  * UNUSEABLE_ - can't use any power, should do everything to turn power off, put processor to sleep.
  * LOW - do low power sensor and logging functions
  * MEDIUM - do any sensor that require medium power, and any medium power comms
  * GOOD - do all sensors and and high power comms 
*/
#define BATTERY_UNUSEABLE_V 3.6
#define BATTERY_LOW_V   3.7
#define BATTERY MEDIUM_V 3.8
#define BATTERY_GOOD_V 3.9
const char build_epochTime = BUILD_TIMESTAMP;
#define RTC_TIME_VALID() getNowEpoch() >(build_epochTime-2*3600)
  if (batteryV< BATTERY_UNSEABLE_V) {
     // Sleep wait for battery V to improve -tbd
   }
    // Begin the logger
    EnviroDIYLogger.begin();
    if (!RTC_TIME_VALID() && (BatteryV >  BATTERY_GOOD_V) ) { 
       EnviroDIYLogger.timeSync();  //if can't sync condition not handled
    }
//exit setup
```

neilh10, 2018-12-06:
I've prototyped a framework for both automatic solar restart and manage radio power #164 .
Its been testing for the last 24hrs with a 500mAhr battery and looking good.
I've prototyped it as being part of ProcessorStats. - is this a good place for it. Or should it be somewhere else

ProcessorsStats.h:
`    
 /* Battery Usage Level definitions
   * LiIon (and any battery) has a charge and internal resistance.
   * For the Logger there are different types of loads that need to be mapped to what the battery is capable of delivering.
   * In terms of priority the system needs to:
   * Determine battery charge status and what loads are possible
   * - UNUSEABLE charge so low, it should immediately sleep and wait till charge improves (solar?)
   * - USEABLE charge to run a basic logger, basic sensors and SD card logging, but no internet comms
   * - LOW Battery, that can run sensors that use more power, and SD card logging, but no internet comms
   * - MEDIUM Battery, that can support low power communications (eg WiFi/900Mz/ CAT1-NB?) and all sensors & SD card logging
   * - Heavy power demand, that can support cellular comms, and all sensors and SD card logging
   */

    
    ps_Lbatt_status_t  isBatteryStatusAbove(bool newBattReading=false,ps_pwr_req_t status_req=PS_PWR_USEABLE_REQ);
    /* Get Battery voltage
     * This measures a complex Voltage, which when there is no other external voltage, is the LiIon Battery.
     * What is really desired, is to measure the charge remaining in the LiIon battery - 
     *   and to do this its desirable to know both the float volt (with a small inconsequential load) and load voltage (voltage with system load) 
     * If the USB V is present, then it measures this voltage, but that typically means there is a good powersupply.
     * There are a number of ways of measuring the voltage with degrees of accuracy.
     * methdod1:Using the external 3.3V as reference, results in the most accurate absolute voltage providing the measured voltage range is 3.7V-15V.
     *   Its inacurrate beneath some threshold due to the cct.
     * method2: determing an accurate voltage drop from an applied load, the Mayfly's mega1284P internal 2.4V reference can be used.
    */
    float getBatteryVm1(bool newBattReading);
    float getBatteryVm1(float *BattV);
    //Future
    float getBatteryVm2a(bool newBattReading);//capture float voltage 
    float getBatteryVm2b(bool newBattReading);//capture load voltage
    float getBatteryVm2diff(bool newBattReading); //getDifference in measured batteryV using internal reference`

The reference values for the above are:
`    
/* Battery Thresholds - may need adjusting for specific battery sizes and Mayfly loads.
   A LiIon manufactuer typically has a protection IC that will do a 
   Low Voltage Disconnect (LVD) at some voltage ~ 3.0V to protect the chemistry of the battery.
   As the current drawn takes the battery to the LVD thresshold, it will then drop to 0V.
   After that it will needs some conditioning - so the following is a first pass.
   It may require some hysteresis thresholds to be added.
*/

#define PS_LBATT_USEABLE_V 3.6
#define PS_LBATT_LOW_V     3.7
#define PS_LBATT_MEDIUM_V  3.8
#define PS_LBATT_GOOD_V    3.9
#define PS_LBATT_HYSTERESIS 0.05 

typedef enum {
   PS_PWR_STATUS_REQ, //0 returns status
   //Order of following important and should map to ps_Lbatt_status_t
   PS_PWR_USEABLE_REQ,//1 returns status if 1 or above, or else 0
   PS_PWR_LOW_REQ,    //2 returns status if 2 or above, or else 0
   PS_PWR_MEDIUM_REQ, //3 returns status if 3 or above, or else 0
   PS_PWR_HEAVY_REQ,  //4 returns status if 4 or above, or else 0
   //End of regular STATUS 
} ps_pwr_req_t;

typedef enum {
   PS_LBATT_UNUSEABLE_STATUS, //0 PS_LBATT_REQUEST_STATUS,
   //Order of following important and should maps to  ps_pwr_req_t
   PS_LBATT_BARELYUSEABLE_STATUS,//1 returns status if above 1, or else 0
   PS_LBATT_LOW_STATUS,    //2 returns status if above 2, or else 0
   PS_LBATT_MEDIUM_STATUS, //3 returns status if above 3, or else 0
   PS_LBATT_HEAVY_STATUS,  //4 returns status if above 4, or else 0
   //End of regular STATUS 
} ps_Lbatt_status_t;`

So I've been testing and usage would be 
setup() very early
`    // Start the serial connection with the modem
    ModemSerial.begin(ModemBaud);
    bool LiBattPower_Unseable;
    do {
        LiBattPower_Unseable = ((PS_LBATT_UNUSEABLE_STATUS == mayflyPhy.isBatteryStatusAbove(true,PS_PWR_LOW_REQ))?true:false);
        if (LiBattPower_Unseable)
        {
            PRINTOUT(F("BatteryLow-ShouldSleep"));
            MS_DBG(F("BatV="),mayflyPhy.getBatteryVm1(false));
            delay(10); // OR do systemSleep - but needs setting up
            //EnviroDIYLogger.systemSleep();
        }
    } while (LiBattPower_Unseable); 

    MS_DBG(F("BatV="),mayflyPhy.getBatteryVm1(false));`

Future: timeSync - only if invalid time.

After wakeup 
`    if (EnviroDIYLogger.checkInterval())
    {
        if (PS_LBATT_UNUSEABLE_STATUS==mayflyPhy.isBatteryStatusAbove(true,PS_PWR_USEABLE_REQ)) {
            MS_DBG(F("---NewReading CANCELLED--Lbatt_V="));
            MS_DBG(mayflyPhy.getBatteryVm1(false));
            MS_DBG("\n");
            return;
        }
        PRINTOUT(F("---NewReading-----------------------------"));
        MS_DBG(F("Lbatt_V="),mayflyPhy.getBatteryVm1(false));
`
Then when looking to do internet comms
`        //if Modem  is Cellular then PS_PWR_HEAVY_REQ
        if (PS_LBATT_UNUSEABLE_STATUS==mayflyPhy.isBatteryStatusAbove(false,PS_PWR_MEDIUM_REQ)) {          
            MS_DBG(F("---Cloud Update CANCELLED--"));
            MS_DBG(F("Lbatt_V="),mayflyPhy.getBatteryVm1(false));
            MS_DBG("\n");
            return;
        }
        if (EnviroDIYLogger._logModem != NULL) modemPhy.modemPowerUp();`

Unvarnished code https://github.com/neilh10/ModularSensors/commit/02f930d5ab2ebc4cdf671b79f212626defd0ffe6

neilh10, 2018-12-07:
@SRGDamia1 @aufdenkampe @fisherba @bschulz1701   - I wonder if you have any thoughts on this - thumbs UP/DOWN.

bschulz1701, 2018-12-07:
So just coming across this thread this morning and have not gone through it in detail, but do have some thoughts based on what I have seen:

### Brown Out Detect (BOD)
Changing the brown out detection level does require setting of the fuses (which is done via the board text file). However, I happen to have a Mayfly lying around and I just pulled the fuses and it is already configured for a BOD level of 2.7v (which is what is desired for an operating voltage of 3.3v), so this appears to already be properly configured

### Power Sequencing 
I am generally not a fan of power sequencing (turning things on/off in a given sequence to avoid exceeding a power threshold), this is certainly a technique which is used, but usually in situations much more tightly constrained than ours, and it results in a delicate setup, which is not cohesive with a modular system.

### Neilh's suggestion
This is not far from what I would recommend, but I would approaching things in a more proactive manor (as I understand his recommendation), and to monitor battery capacity during all regions of operation, not just at startup, but this could be a more simple fix to the problem, to simply not bother to try to startup when below a threshold. 

However, for both this system or for any other battery monitoring options, a more sophisticated approach to battery estimation is likely required. Either adding current and/or temperature into the equation, or using an external battery monitoring system (BMS), because float voltage alone will not tell you everything you need to know. For example, if I am at 25&deg;C and my battery is at 3.8V (just as an example), I will probably have enough capacity to boot up and use the radio just fine. But if I am say at 5&deg;C and my battery is at 3.8V, even though my voltage is the same, my capacity is much less, and now when I boot up the radio, my voltage sags too low and browns out the system. One answer to this is to just make your margins larger, but of course this means on one end you're going to be sitting idle when really you are able to log just fine. The best answer to this is that better battery monitoring is likely required.

**See Lithium Battery discharge curve below to illustrate what I am getting at**
[image]

### Battery Monitoring
The best way to solve this problem (in my opinion) is to monitor the state of the battery and have a threshold below which the device should no longer function. The key here is that the logger should be making this decision, rather than the hardware. Something like the following:
``` C++
if(CheckBatteryStatus() < MinBatCapacity) {
    SendDeathbedMessage();
    GoIntoLowPowerLogging();
    while(CheckBatteryStatus() < MinBatCapacity + Hysteresis) {
        LogLowPower();
    }
    SendWaveupMessage();
}
```
In this case, `LogLowPower()` could log just to the SD card, with long sleep intervals, etc, taking a significantly smaller fraction of battery power, hopefully low enough to stay alive, or just go into sleep mode if battery capacity is too low, etc.
This way, you watch the battery, when it gets below a safe capacity (as you determine), you can send one final message (`SendDeathbedMessage()`), letting everyone know that the system is about to stop sending data, but it's "ok", it's just because of low battery power, not an equipment failure, etc.
Then, when the battery capacity is back to a safe range (MinCapacity + a Hysteresis value), the system can go back to doing its full thing.

Sorry is this is a whole pile of information, meant to make it short, but then my battery junkie side kicked in and ended up writing a lot. Hopefully this helps move towards a solution at least!

neilh10, 2018-12-07:
HI @bschulz1701 - thanks for the comment - and I appreciate the description, and I suggested the same (#196).   
As you say its a detailed challenge - and takes time to absorb the detail.  I'm trying to break out the can of worms - we just want it to work - to a simple model.  

I believe I'm suggesting an API that fits with what you  are talking about. :)
 
I'm framing it as a proactive "Power Demand Management" API, with out knowing what it takes to do it underneath the hood in terms of monitoring the battery charge/power available. Proactive because the user does need an understanding of what causes the power demand, and that can be managed.  

Battery capacity is the holy grail, however practically with this Mayfly's 0.5b architecture the available charge/coloumbs/nAHrs  is a fuzzy analysis for different  battery capacities, battery types and battery life (ugh!).  Looking just a battery voltage, is simplistic at best - but a starting point. (#196)

I'm trying to simplify it to  define what the requesting layer is asking for. This is an attempt to simplify having to understand the can of worms of battery curves and battery impedance's.

I'm suggesting it can be implemented as a PowerManagementSystem for the Mayfly,  and other Power Supply  Architectures that evolve.  (One architecture I  implemented, I used a 5F SuperCapacitor that with a days solar charge could store enough  coloumbs for overnight activity- thereby extending the LiIon life)

With a "Power Demand Management"  framework, the user/requestor  needs a model for the hardware supports, and that should define what power the optional next activities are going to consume 
eg from activating  a sensors to activating a peak power modem

I'm proposing an API that can ask 4 simple questions of the PowerManagementSystem (aka BatteryManagementSyste)   
mayflyPhy.isBatteryStatusAbove(,PS_PWR_USEABLE_REQ) - eg basic operation including SD card   
mayflyPhy.isBatteryStatusAbove(,PS_PWR_LOW_REQ)      - eg a heavy demand sensor   
mayflyPhy.isBatteryStatusAbove(,PS_PWR_MEDIUM_REQ)   - eg a low power comms   
mayflyPhy.isBatteryStatusAbove(,PS_PWR_HEAVY_REQ)       - eg a Cellular Modem   

Maybe there is a better label or way of phrasing this ... perhaps mayflyPhy.isPowerAvailable()

In the above API, if the answer is NO power for request, then zero value is returned.

If the answer is YES power is available for request, then a NONE-zero value, equivalent to the battery status is returned.

Hence I believe it fits with your distributed concept (and #164) and anytime a power demand activity could be scheduled a simple test can be made.

SRGDamia1, 2018-12-10:
I'm sorry, I haven't had time to look into this much yet.

I'm nervous that this is getting to be very detailed and specific to certain hardware.  I've been trying to make the whole library usable by a variety of boards.  The processor stat/battery voltage is almost the only chunk that is board specific.  

The battery voltage pin on the Mayfly is definitely not designed to be used for detailed power management or smart battery optimization.  It's just meant to give a ball-park number.  I doubt any general purpose Arduino boards would do much better.  And the cut-off numbers that I'd put in the examples as switch-overs between functions were more-or-less pulled out of a hat.

If we want a really good power management system that can be used on the Mayfly, it might be better to move all of that into its own new repository/library separate from this library.  Then you can use that library with this library to get the result you want.  You can already created multiple "loggers" with slightly different functionality in this library, similar to what is being done in the "data_saving" example.  But in this case, instead of setting up different groups based on what you want to be transferred remotely, you set it up based on the power draw and available power returned by the power management library.

SRGDamia1, 2018-12-10:
Just FYI, my current priority is actually reducing data use.

neilh10, 2018-12-11:
Hi Sara, thanks for the comments. Yes thankyou thankyou for covering multiple embedded boards and that definitely allows a future extensibility. I definitely hope that this moves towards an embedded Cortex M0 type processor, but I've got it on my list to start building for the Feather M0 Alogger/SAM21D.   Its the reality of embedded processors that they vary in functionality a lot, so your concerns are what makes embedded processors so hard. 

I suppose my understanding of boards here is that they run off a  LiIon battery and are solar powered, and low power to keep the total cost of solar panels/LiIon storage down. Perhaps I've got that wrong?

The API proposal is a PowerDemandManagement based on a LiIon being an imperfect power source  - and a priority at least on my part, to run the sensors - gather data- as long as possible. I think its an simple way of abstracting the imperfect LiIon. The LiIon curves should decreasing voltage, what is hard to follow is the increasing source impedance, which result in processor reset.

The API I've proposed is based on the Mayfly's mega1284 ADC Vbatt sensor, though I've duplicated  other better LiIon V  sensors as well to verify the the Vbatt readings. I've been running a prototype with an Xbee Wifi & LiIon 500mAH to learn how the battery behaves under different loading.
 
For any board that can't or just doesn't need to support  Power Demand Management/Lion BMS - the API would always return a non-zero - behaving just as it does now. It can be moved into its own class as well for clarity, and so its optional.

So I'm floating and sharing this discussion partly to see what direction modularSensors  might go in  - and partly to check out if the community has any simpler ideas.   Its extra work for me to propose and share it, and probably a pain for you to incorporate this type of thing - so its open source in my repository - so I'm not attached.  :)    A modular library could work, but actually what evolved for me was the reality of  "PowerDemand Management" as it needs to be threaded through the code - as @bschulz1701 was saying. Call it a teaching moment. :)

SRGDamia1, 2019-06-24:
There's good discussion here, but I'm going to close this because the main issue of the board continuing operation after it gets more juice has definitely been fixed.

closed 2019-06-24 by SRGDamia1