EnviroDIY/ModularSensors #164: Enable power-saving mode, to turn off radio when battery drops below 3.6V

aufdenkampe, 2018-09-09:
To further avoid remote stations from shutting down when solar charging is reduced during winter storms (see #163), I'll develop some example sketches that employ a power saving mode.

My plan is to do this by modifying the [data_saving](https://github.com/EnviroDIY/ModularSensors/tree/master/examples/data_saving) example with some `if` statements comparing measured battery voltage relative to the voltage threshold.

Once the voltage is below the threshold, we'll only log to the SD card (skipping the radio logging) and also double the length of the measurement interval.

cc: @gcutrell, @everhamme

SRGDamia1, 2018-09-15:
This would probably be made much easier by modifying the EnviroDIY logger class so that instead of over-writing the base logger class's 'log' function, it kept that as is and had an additional 'logAndSend' function.  That would be an easy change in the library.

SRGDamia1, 2018-09-17:
I'm starting on a lowPower branch for this and #163.

I've already split the log function into "log" and "logAndSend" and the begin into "begin" and "beginAndSync."

Because the logger object itself doesn't actually know if it can measure it's own battery voltage, I think the if's would have to go in the sketch.

You should be able to combine the examples for data savings along with the double logger example to get both the longer spacing and fewer sensors.  I think you'll want to use the "Logger::markTime()" function at the top of the loop and then check batter voltage and use "checkMarkedInterval()" after that.

neilh10, 2018-10-04:
It would be nice for a powerGood check call before heavy power users eg cellular, and a good/fail return.  There may be certain instruments that are power users that might also want to do this.
The powerGood() method can have the methods for checking power. 
BTW: as it stands for ModularSensors\examples\logging_to_EnviroDIY  if a sensor reading doesn't get delivered  over the internet access, there isn't ever a retry later. Of course its logged to SD card.

SRGDamia1, 2018-10-04:
@neilh10 have a look at the data_saving example in the low power branch.  You can create a check using you battery voltage variable and then check against it before either logging the data or logging and sending it.  If you want to, it's not hard to create multiple "loggers" running together with the sensors grouped by their power draw.

I don't intend to include a blanket "powerGood" function within a logger because it would be impossible to create such a function while maintaining the library's usability on multiple processors/boards.  While I think the only thing anyone is running the code on "in the wild" is a Mayfly, I really don't want the library to be tied to that one board.  Right now the library fully supports being used on any AtMega or AtSAMD based board with sufficient processing power.  (An Uno or Leonardo isn't powerful enough, but a Mega or Zero or any of their clones should work "out-of-the-box" with the library.)

neilh10, 2018-10-04:
Thanks for the response. Just an idea. Perhaps powerGood always returns TRUE, but can be over-ridden by any board that should have a need.  ? That way communications and internet access is standard. Any special power processing gets implemented in a per board function.  
I'm just remembering a solution I came up with 10yrs ago in dealing with low power on solar powered battery systems.

neilh10, 2018-12-06:
@SRGDamia1 - just wondered you mentioned a lowpower branch above - but I can't find that. Is it a private branch. I'm thinking out some issue in #163 and would be nice to check in with you.

SRGDamia1, 2018-12-07:
The lowpower branch has been merged and deleted.

neilh10, 2018-12-07:
Ok thanks. Just wanted to check if there was any other activity. I'm having some good results from the algorithm I've implemented in #163

neilh10, 2018-12-07:
To summarize from #163, I'm proposing an API that would allow this question to be asked as   

mayflyPhy.isBatteryStatusAbove(,PS_PWR_MEDIUM_REQ) - eg a low power comms   
mayflyPhy.isBatteryStatusAbove(,PS_PWR_HEAVY_REQ) - eg a Cellular Modem

SRGDamia1, 2018-12-22:
@aufdenkampe, please close this out let me know if you think more is needed.

neilh10, 2018-12-25:
just been looking at this in 0.19.2 before signing off here in California.
so as a 2cents on coding ~ 
It seems to me the multiple usage getBatteryVoltage(), which  originates a new internal ADC conversion could cause critical hard to track problems in the future.
Batteries with variable impedance losing power are challenging to debug.
It will be different value returned each time getBatteryVoltage() is called and takes up CPU time to do ADC on each call. Most of the time it won't really matter, and will be pretty similar, but occasionally it might be.
For an embedded system, referencing real world values (battery voltage)  its typical to do the conversion once, and then use the same value for all related decisions comparisons. 
I did this with my version.
  My version was equivalent to getBatteryVoltage(bool newConversion=false); - so first time its called its getBattery(true) and caches the reading, thereafter  its getBattery();
(Also I don't know why the Mayfly version is a runtime decision - its really a compile time option)

It would seem to me using a voltage number (3.4  ..... 3.7) directly in the comparison multiple times across many files is a maintenance challenge and not informative of the meaning attributed to the number.
It would be a benefit  to  define a meaningful label, that then has a number attributed to it.
Some labels for the lower limits could be 
BATTERY_UNUSEABLE  .. BATTERY_POWERGONE 3.4
for the upper end could be 
BATTERY_POWERGOOD  .. BATTER_POWERMINIMAL  3.7

so just my 2cents.

SRGDamia1, 2019-05-31:
Are the corny solutions for this ok?

aufdenkampe, 2019-05-31:
Yes! @gcutrell did a bunch of testing this winter, and it works great!

One thing we've decided is that 3.65V is a better yet still safe radio transmit threshold than 3.70V, so I'll probably eventually update all the examples.

closed 2019-05-31 by aufdenkampe