EnviroDIY/ModularSensors #176: I2C sensors on switched power cause startup to hang

aufdenkampe, 2018-10-11:
@gcutrell, @fisherba and I have all noticed this summer that when I2C sensors, when on "switched power" have been causing hangups with the startup, but when we put them on constant power thing start up fine. I'm wondering if this might have been introduced with the I2C work for #148.

@SRGDamia1, have you noticed anything similar? 
Is this issue fixed with all the updates with PR #173?

SRGDamia1, 2018-10-11:
I've not noticed anything.  Is it just at startup?  Which sensors?  Remember, I don't even have some of the sensors and heave different breakouts of the others, so for some of them I'm going by datasheets.

SRGDamia1, 2019-02-20:
I finally had this happen for the first time and looked into it a bit.  Apparently the wire library sometimes hangs on the endTransmission function if it doesn't get a clean end.  In my case, I'd accidentally connected a sensor sending continuous UART data to the I2C lines.  I'm guessing when you've had this happen it's probably a sloppy sensor sending noise over the line that's causing the hang.

aufdenkampe, 2019-02-20:
Thanks for that extra info. Is there a way to prevent this via the library?
I haven't had any new stations with I2C connections, so I don't know if it's been fixed or not.

SRGDamia1, 2019-02-20:
No, the hang is outside this library.  I don't think I can fix it.

SRGDamia1, 2019-02-23:
The hang is from one of the external I2C sensors having an pull up that connects their VCC to the I2C lines. When power to those sensors is cut, the SCL and SDA lines are now pulled low instead of high as they should be. The bus thinks this is an unending start or stop condition and everything hangs forever the next time any call to the I2C library is made (like taking to the clock).

The right solution is a hardware one, isolate the switched I2C peripherals from the umswitched ones.

SRGDamia1, 2019-02-23:
I'm going to close this because this isn't something I can fix via this library.

closed 2019-02-23 by SRGDamia1