EnviroDIY/ModularSensors #180: XBEE setup support LowPower options

neilh10, 2018-10-19:
The Xbee module pinSleep is managed - in the current master/develop branches it has no effect on power usage of the Digi WiFi S6 that I've found. 

My research shows the S6 modules pins sleep settings need to be activated by setting SM register to 1, and possibly the D8 register to 1.  Setting the SM=1 definitely saves power.
I'm measuring power from the SOLAR input as supplied by a bench power-supply, with no LiIon battery. 
With SM=0 default the power-supply shows 110mA continuous through sleep.
With SM=1 the current draw drops dramatically. The current drawn as shown by the power-supply peaks at 60mA with WiFi active - but this may be smoothed . The S6 module datasheet spec for Tx is 309mA and 100mA Rx.  
With the module asleep the current is about 10mA 
The S6 module supports two types of sleep with SO bit 6
0x40 Stay Associated
0x00 Deep sleep   (default)
If the device was associated when it went to sleep, it should be ready to transmit data as soon as the ON_SLEEP pin indicates that the device is awake. 
If the device was not associated when it went to sleep, the host must wait until the device is associated before a transmission can occur.

The  TinyGSM lib TinyGsmClinetXbee.h:382 has a  setupPinSleep(bool maintainAssociation = false) 
but it needs a prototype in class TinyGsmModem eg TinyGsmCommon.h:~246

It seems to me the place to call the setupPinSleep() is
LoggerModem.cpp:256
        if (_tinyModem->hasWifi()) 
        {
            _tinyModem->networkConnect(_ssid, _pwd);
            **_tinyModem->setupPinSleep(true)**;
        }
        else _tinyModem->gprsConnect(_apn, "", "");

I'm still learning the class structure - so this is a "straw poll" to see if I've missed something.
The TinyGSM also manages the Xbee cellular modules, and I haven't researched low power usage here.

SRGDamia1, 2018-10-19:
If you're looking at the most recent release of this library (and the current master branch), the setup pin sleep function **is** called in the modem setup.

I removed the pin sleep set up from the logger modem set-up when I restructured the modem to require input wake and sleep functions.  I added instructions in the [wiki ](https://github.com/EnviroDIY/ModularSensors/wiki/Digi-XBee) for how to set up the sleep options in XCTU.  You can always call the setupPinSleep() function on your modem yourself in your programs's set-up if you'd prefer to do it that way.

In general, the TinyGSM library makes no attempt at all to manage power use or to turn the modem on or off.  The genius of TinyGSM is converting the AT command structure and response into an Arduino client/stream structure.

neilh10, 2018-10-19:
Wonderful. Thanks for the pointer to the wiki page - I hadn't seen that. 
I'm doing Xbee Setup before through the XCTU. My XCTU hardware is from when the modules where produced by the original Maxstream - though they still work. 
I moved to working off the ModularSensors develop branch on the basis that it is likely to be where things are going .....  so appreciating all the updated classes. :)
Yes I've been digging through the TinyGSM and appreciating someone else having generated the classes :) and making them available - I've used Digi Xbee API mode on other projects, which provides different opportunities.

neilh10, 2018-10-19:
just curious with a related question - so if the base assumption is that this is a solar powered system - that is the installer needs to invoke XCTU to program the SM and D register, shouldn't the WiFi & Password be setup through XCTU as well.? 
Having the SID/Password buried in code seems like its potentially setting up a trip wire for the unaware who might then "publish" it on the github. 
I was going to investigate putting it in the EEPROM, though that  will require a simple cmd line interface.
Also I've modified TinyGsmClientXbee.h such that if there is no wifi password it sets  security to NONE.

SRGDamia1, 2018-10-23:
You can set up the passwords on XCTU if you like for an XBee, though I very much doubt XCTU would play nicely with an ESP8266....

SRGDamia1, 2018-10-23:
The SSID and password and APN are set in the program because that's the way TinyGSM was set up for all the other modules before I added the XBee to it.  Be careful what you push to public repos.  A very common way to call passwords is to create and include a header file that's excluded from git.

While this code is intended to be used on a solar powered logger outside somewhere, it's definitely not meant to be an all purpose sensor/modem manager.  The expectation is that for ever sensor and for every interaction beyond "send me a data point" the user connects their sensors to their computer or whatever needs to happen to make sure the sensor is set up correctly.  That applies just as much to an XBee as to a D-Opto DO sensor.  You could write all the code into the set-up of a program using this library to make sure your D-Opto is in SDI12 mode (not RS485) and it has the address you expect and the calibration is good, etc.  It's do-able.  But that's not what this library is for.  I'm expecting that you've connected done all that somewhere else and I can just ask for values.

Unrelated to this issue, may I ask why you're using this library and EnviroDIY general?  When you talk about XBee's in API mode and tens of thousands of units, it sounds like you have dramatically more resources available than we have for the entire EnviroDIY project.  I'd really recommend you talk directly to @aufdenkampe and @darscott if you're interested in collaborating.

SRGDamia1, 2018-12-21:
I've added the example coding to the wiki and several examples.  I'm going to close this.

closed 2018-12-21 by SRGDamia1

neilh10, 2018-12-21:
Thanks for the implementation direction. 

In 3months of ModularSensors development I've only used the Digi WiFi for comms in the ModularSensors, focusing on characterizing one comms channel in the ModularSensors context.   

I have been running one field device with a solar powered 2Ahr battery for a month and it has worked flawlessly. 

However, on the development bench, I personally found some unreliability in the Digi Xbee WiFi device, and made modifications to get round them. see my mods in https://github.com/neilh10/TinyGSM
 I have seen the Digi Xbee WiFI device gointo a mode where it starts taking 100mA - and not responding - this is difficult to debug, and happened about 4 times. The only option has been to power down the whole board and restart. The mayfly 0.5B doesn't support reseting or power off of the Xbee socket.   
I personally like the facility on the  Autonomo for turning off the Digi Xbee socket - which I think would overcome this issue.

SRGDamia1, 2018-12-21:
I like the look of the Autonomo, but I've never actually held one.  If you get one, this library should work flawlessly on it.

neilh10, 2018-12-22:
Great - I have two on the way from the Nederlands. It has 6 serial channels + USB, Just hope it isn't really obsolete the way that Digikey said.