EnviroDIY/ModularSensors #181: XBEE Module Keep Alive

neilh10, 2018-10-19:
This is a straw poll for what experience anybody has for keeping the various Xbee modules alive on the Mayfly or other boards.
By alive I meaning responding to serial commands.
In using the Digi Xbee Wifi S6, and periodically the module appears to be pushed into a state of non-response. I've verified this by using the a Saleae logic analyzer on the Tx/Rx pins. The module has Tx requests going in and there is no response from the Xbee.

Every time I've used an external modem module such as the Xbee (or Mutlitech modules) I've designed in the facility for software to cut the power, or apply a hardware RESET to restart the module.

To prototype it, I propose to add a wire to the Xbee reset (pin5)  and activate it from software. 
Initially I'm thinking of overloading it on D21 which is also used as the pushbutton.    
That is when D21 Button is pressed it would reset the Xbee module in hardware, as well as software functions.
In software I would activate this line in setup() for 1mS, guaranteeing that the module is reset whenever setup() is performed. 
At a later stage a "Keep Alive" algorithm can be implemented from a non response to an already active   Xbee.

I wonder if anybody else has experienced xbee soft failures or found a better Keep Alive strategy.

SRGDamia1, 2018-10-19:
I'm not sure this is the right place to ask a question like that.

SRGDamia1, 2018-10-23:
I'm confused by what you mean by 

> I have found by disabling the Xbee S6 sensor percentage signal its better 

You mean that your module hangs less if you don't create a variables or update values for signal percentage and/or RSSI?  It's possible it's hanging when it pings NIST to get signal strength.  One of the quirks of the WiFi XBee is that it can only get signal strength after opening a TCP connection.  NIST's daytime server is a convenient one since it will immediately close the connection after responding.  That's particularly nice because none of the Digi XBee's have a good way to close a TCP connection themselves.  I can add an extra timeout to that request, to see if that helps.

SRGDamia1, 2018-10-23:
If you want the module to be reset every time it's used, just add a reset to your wake function.  If you want it to be reset every time the setup function is run, add it to your set-up function.

neilh10, 2018-12-21:
I'm closing this as from the "straw poll" I don't think it is of concern.

The problem is that its the Xbee WiFi S6 that isn't responding on the serial channel, using a logic analyzer to monitor characters on the physical pins.  So yes I found polling for the RSSI caused more unreliability. 
IMHO it needs to be hardware reset or power off function. The Sodaq Autonomo does implement this - so I want to test it. For any new hardware designs it makes $$ sense to use a SAMD21 rather than a mega1284.

IMHO,  For low power systems that need guaranteed reliability turning off the power guarantees no power is take and that the module will reset on power up - that is crosses off two potential bugs with one processor pin (restated the "Two birds with one stone" metaphor).

closed 2018-12-21 by neilh10