EnviroDIY/ModularSensors #182: XBEE S6 WiFi testing

neilh10, 2018-10-25:
I've running a 3 Xbee S6 - and reporting one of them #3 as one testing  point.  
I am digging into it, and learning as I go,  and if its useful for anybody good - but otherwise it helps me to put it down in a linear fashion - and also to figure out what I expect it to do. 
I do this professional, and though this open source, and its sharing the data on the way to getting it working.
Overview: over two days the sensor data recorded to SD drive was rocksolid. The WiFi comms worked for ~20hrs. 

I've a "develop" branch with debug enabled, running three Xbee S6 WiFi reporting to data.envirodiy.org
Two mayflys are inside and "Mayfly#3" (rev0.5b) is in a protected outdoors location with 4.4AHr battery +~4Wsolar +Keller Depth sensor in 0.07m of water with a good solar aspect to drive the water temperature through a diurnal cycle. 
Its reporting here http://data.envirodiy.org/sites/LCC45aa/
The mayfly was started on  Fri Oct/19 @17:22(UTC-8) and then I left town and monitored on data.envirodiy. 
 It ran successfully till  Sat Oct/20/2018 13:00(UTC-8:00) when it stopped reporting over wifi
On returning Sunday night ~17:40(UTC-8) it was still signalling with LEDS that it was running, and I copied off the log file. It was still collecting data.
Observation: From the LEDs it was good to see it was collecting data, but it would have been nice to have an indicator that it wasn't delivering to the cloud.  
 For Mayfly#3 I then  powered off/removed battery all night.
On Monday at ~7am I quickly applied the battery and then powered on. The RED led came on and stayed on, I didn't have time for any further observations.
Due to other commitments I left it running till,
Thursday ~10am , I plugged in a debug FTDI cable that doesn't reset the Mayfly and on the terminal (TeramTerm) it was showing
+++OK
ATCN
OK
+++OK ......
continuously 
The logfile had headers, but no data.

Sampling Feature: 79f702a9-368f-4940-9669-8978ffa3254b |   |  
-- | -- | --
Data Logger: nh07g | KellerNanolevel | KellerNanolevel | EnviroDIY Mayfly | MaximDS3231 | ExternalVoltage | ExternalVoltage | EnviroDIY Mayfly
Data Logger: nh07g | gaugeHeight | temperature | batteryVoltage | temperatureDatalogger | voltage | voltage | Seq Num
Data Logger: nh07g | meter | degreeCelsius | volt | degreeCelsius | volt | volt | Num
Data Logger: nh07g | 67c22f5d-e5d8-4a26-a82b-7f59132e5c81 | b03d4384-4623-4df5-b705-f50710d5e4e9 | 8c796edd-7863-4fe7-9e54-0cbe0d694d59 | e6159fe0-e30d-4a9d-bebc-1dc5c2435a22 | d3b78c2e-312b-4e2a-b804-8230c963f912 | c7da692b-6661-4545-bd3d-04938faa285b | 5fbb799d-630d-486f-a0ff-015f0195d393
Date and Time in UTC-8 | kellerHeight | kellerTemp | Battery | BoardTemp | extVoltage | extVoltage | SeqNum

I powered off for 2minutes - removing both LiIon and Solar panel - powered backon.
From the Tty it showed connection over the WiFi and success reporting to data.envirodiy.
It is left running.

So: With my test systems I've been seeing a lot of it not starting the  xbee S6B Wifi  and also once its running for the Xbee to stop  responding, so my next steps are to implement the Xbee  Reset, and try and detect the conditions causing the Xbee  to stop responding.

fisherba, 2018-10-25:
Super helpful, Neil! Thanks for reporting your observations. Voltage was at 4.1 when you lost power (and continued to drop after that), but shouldn't that be sufficient power for the radio? (If it isn't, my stations are in trouble.)

neilh10, 2018-10-25:
Hi Beth -  the powering to me looks  PURRfect and the sensor data capture on the SD card was excellent/reliable. It looked to me like the issue was software/XBEE S6B  
I put graphs in the .xlsx of the voltages - only two days but it charged nicely.  
This is the Vbat measured by internal ADC and recorded on SD card.
The peaks are fully charge with afternoon sun
[image]
This is the LiIon measured by Ext
[image]
This is the SolarCells "USB" Voltage output. as when power starts being delivered by the sun. The Solar panel was a west aspect, only getting fully afternoon sun. The 2nd day sunday shows a more distinct upwards slope as the LiIon is charged. 
[image]

I've set the system up again with logging from the debug connector to monitor what happens.

neilh10, 2018-12-21:
I'm closing this, as the reliability data I have from one field implementation has been pretty good. 

There are some powering issues - but I'm taking them into  https://github.com/neilh10/ModularSensors/issues/4

I have been running one field device with a solar powered 2Ahr battery for a month and it has worked flawlessly - no reboots. But has had the odd day of no solar power.

In 3months of ModularSensors development I've only used the Digi WiFi for comms in the ModularSensors, focusing on characterizing one comms channel in the ModularSensors context.

However, on the development bench, I personally found some unreliability in the Digi Xbee WiFi device, and made modifications to get round them. see my mods in https://github.com/neilh10/TinyGSM
I have seen the Digi Xbee WiFI device gointo a mode where it starts taking 100mA - and not responding - this is difficult to debug, and happened about 4 times. The only option has been to power down the whole board and restart. The mayfly 0.5B doesn't support reseting or power off of the Xbee socket.

I personally like the facility on the Autonomo for turning off the Digi Xbee socket - which I think would overcome this issue, and is probably where I am going to next try testing,

Repeat of info added to https://github.com/EnviroDIY/ModularSensors/issues/180

closed 2018-12-21 by neilh10