EnviroDIY/ModularSensors #185: Variable logging  rate

neilh10, 2018-10-30:
There is a variable update rate - as seen with an increasing sequenceNumber visualization
[image]
Build 'develop' at EnviroDIY/ModularSensors#bfeb4ca5b1b37b4343f239df2611dad29c6d2deb
It appears to be a result of loggerBase.h : int8_t _numIntervals; it increments every sensor update logging interval (~ target 5minutes)  and wraps from 127 to -128.  When _numIntervals is -ve it operates in a more complex manner
    // The number of logging intervals that have passed.
    // NOTE:  This is not intended to be a perfect count, and it will zero
    // after every time the logger restarts (such as running out of battery).
    // A value of -1 indicates that sensor set-up hasn't been run!
    int8_t _numIntervals;

I would think that when the number reaches a maximum it should be wrapped back to a minimium. 
The range of max to min is longest time interval that the device can sleep between samples, but there are some implications for aliasing for certain logger intervals.

With separately generated 16bit sequenceNumber this is what the https://data.envirodiy.org/sites/LCC45a2/ (Percent full scale) datatable shows
Logging set at 5minutes 
2018-10-28T13:40:00	-8	127
2018-10-28T13:45:00	-8	128   <<< correct so far
2018-10-28T13:46:00	-8	129   <<then logs after 1min
2018-10-28T13:48:00	-8	130  << logs after 2min 
2018-10-28T13:50:00	-8	131  << logs after 2min 
2018-10-28T13:52:00	-8	132
2018-10-28T13:54:00	-8	133
2018-10-28T13:55:00	-8	134 <<then logs after 1min
2018-10-28T13:56:00	-8	135 <<then logs after 1min
2018-10-28T13:58:00	-8	136
2018-10-28T14:00:00	-8	137
…….
2018-10-28T17:30:00	-8	263
2018-10-28T17:32:00	-8	264
2018-10-28T17:34:00	-8	265
2018-10-28T17:35:00	-8	266
2018-10-28T17:40:00	-8	267  << logging back to every 5minutes

With debug on in LoggerBase.h
</start Debug
Current Unix Timestamp: 1540825260
Logging interval in seconds: 300
Mod of Logging Interval: 60
Number of Intervals so far: 127
Mod of 120: 60
Not time yet.
Current Unix Timestamp: 1540825320
Logging interval in seconds: 300
Mod of Logging Interval: 120
Number of Intervals so far: 127
Mod of 120: 0
Not time yet.
Current Unix Timestamp: 1540825380
Logging interval in seconds: 300
Mod of Logging Interval: 180
Number of Intervals so far: 127
Mod of 120: 60
Not time yet.
Current Unix Timestamp: 1540825440
Logging interval in seconds: 300
Mod of Logging Interval: 240
Number of Intervals so far: 127
Mod of 120: 0
Not time yet.
Current Unix Timestamp: 1540825500
Logging interval in seconds: 300
Mod of Logging Interval: 0
Number of Intervals so far: 127
Mod of 120: 60
Time marked at (unix): 1540825500
Time to log!
------------------------------------------

    Running a complete sensor update...
.......

------------------------------------------

Current Unix Timestamp: 1540825560
Logging interval in seconds: 300
Mod of Logging Interval: 60
Number of Intervals so far: -128              <<<<< this is where it wraps, and starts udate
Mod of 120: 0
Time marked at (unix): 1540825560
Time to log!
------------------------------------------

    Running a complete sensor update...

------------------------------------------

Current Unix Timestamp: 1540825620
Logging interval in seconds: 300
Mod of Logging Interval: 120
Number of Intervals so far: -127    <<<< increments by 1
Mod of 120: 60
Not time yet.
Current Unix Timestamp: 1540825680
Logging interval in seconds: 300
Mod of Logging Interval: 180
Number of Intervals so far: -127
Mod of 120: 0
Time marked at (unix): 1540825680
Time to log!
------------------------------------------

    Running a complete sensor update...
....
-----
<! Debug

SRGDamia1, 2018-10-30:
Thanks for the catch.  The _numIntervals value only exists as a convenience feature - to log more frequently at start up.  It's rolling over way to quickly, causing the problem.  Since I've created an interrupt-driven "testing mode" to show frequent sampling after a button press, I don't think this feature is even needed any longer and should just be removed.

neilh10, 2018-10-30:
Sounds good. I don't have any perspective one way or another. The press button testing mode works nicely as well. Great to be playing with it.

SRGDamia1, 2018-10-30:
Removed extra logging with d6dec1e184005dd2732f8d14071d98d0cd9146d4

closed 2018-10-30 by SRGDamia1