EnviroDIY/ModularSensors #186: Scope of testing for develop?

neilh10, 2018-10-30:
I'm just wondering what the scope of testing is  for the develop branch? 
I'm getting some aspects of testing, and building up an array of Mayfly's but its just on Xbee wifi, and i hope soon Xbee CAT-m1.
The develop branch has some great changes, however I'm curious as to what everyone's thoughts would be  to branch develop to a 'test_snapshot' - and have that as the testing branch. Then for bugs that are found they would be added to 'test_snapshot'  to verify them.

SRGDamia1, 2018-10-30:
I'm sorry, I'm not a trained programmer, so I know nearly nothing about setting up test suites.

You can always take a "snapshot" but using a specific commit SHA.

And it's not just tested on a Mayfly with XBee wifi.  I've also tested with 3G XBee, 3G uBee (SARA u201), GPRSbee (sim800), ESP8266, and an Adafruit feather.

neilh10, 2018-10-31:
Hi Sara - apologies I was floating it around for everyone else involved in testing, so not sure who is planning what.  
From the submissions I seeing you are doing the coding (which makes you the programmer - on the job training, great work :)  and then there are a few people doing some testing. Which I'm trying to figure out how to help (I hope). So thanks for the all the work in the coding and also the build environment .... some amazing stuff (travis? my first introduction on this project) and the core of being able to dance with a mayfly.  *&  **&   &+*

Traditionally, and my experience when I'm coding, coding/programming is so intense, coding specific module/feature evolution that its difficult and **time** consuming to be disciplined about a system level testing. 
So thankyou for listing the targets that you are testing with - and that works very well to make sure they run. 
Groups like [purdue](https://www.cs.purdue.edu/undergraduate/curriculum/track_softengr.html) - have a matrix approach for software testing per release and the bug bear of embedded unattended uP is  stability test. Does it keep working foreever. 
The basics of more real world "`[scrum]( https://github.com/jvandemo/github-scrum-workflow)`" agile development is that the coding is snapshoted regularly (periodically) for coverage test - till it meets a good enough standard.
For open source - quasi volunteer driven - it becomes who volunteers for it.  
So just trying to suggest a conversation on the scope.
Testing is also pretty squiggy per project, so its a case of sharing what works, and what makes sense for this specific release. 
So I'm hoping to set up some stability projects running with Xbee WiFi and Cat-M and run through a combination of real world issues.  Thats what I want for the system - and my experience is its pretty challenging to catch the last few issue.
I'll be pulling from my background on what happens in the real world, which sometimes is pretty hard to achieve. Maybe some of the issues will be a discussion. The embedded world of a remote software running on the mayfly is pretty challenging for reliability. 
My devices have run in the field for years - though I did have some unexpected hardware problems with the mega2560 when the power died slowly.
So I'm hoping for Xbee WiFi (and Xbee Cat-m) do a few standard sensors to test the core.
- Long term stability (2weeks?) and then let it run
- Emulating a solar charging environment.

But the reality its challenging to get to a stability position - for some reason most of today 'develop' hasn't worked for me. Thankyou for adding the low power mode for the Xbee - but for some reason on my system it has pushed the Xbee into a sulking silence after it trys to contact NIST
Haven't figured out why, so still working on it. 
Also the latest release somehow is crashing my [platformIo](https://github.com/platformio/platformio-vscode-ide/issues/394).  I submitted them an issue after a rebuilt for the 3rd time   I;ve backed up to the previous release but doesn't make much sense to me why platformIo isn't happy.

neilh10, 2018-12-21:
I opened so I'll close as I hasn't stimulated any wider discussion. I really value the high level of stability that team members have built into the software.

closed 2018-12-21 by neilh10