EnviroDIY/ModularSensors #188: Support logging to ThingSpeak channels

SRGDamia1, 2018-11-06:
ThingSpeak looks like it has some interesting free data analysis tools.  It might be nice to set up a module to use their RESTful API's.

neilh10, 2018-11-06:
It would be  good feature, to have another publishing source. They also have an MQTT interface - which I haven't tried. I've used it for it about 2yrs and never had any issues.
One example I've done is a rain gauge - http://azonde.info/pm2/LiveRainGauge3.htm that has used an IMP type WiFi device. To get more history press the button on the bottom left "Load More Historical Data".
The display code is at https://github.com/neilh10/thingspeakmultichannel-highcharts - 

For the free account, they do have a limitation on the rate that you can publish to once every 15seconds.

aufdenkampe, 2018-11-06:
Seems cool, except that [ThingSpeak IoT](https://thingspeak.com/pages/commercial_learn_more) is built around Matlab. Could the data also be accessed with R or Python? I wonder if anyone has done that.

SRGDamia1, 2018-11-06:
https://pypi.org/project/thingspeak/

neilh10, 2018-11-06:
They have  a REST API, so definitely can be access in python
https://www.mathworks.com/help/thingspeak/channels-and-charts-api.html

aufdenkampe, 2018-11-06:
That's all great to hear that this is open to all users. Having been a Matlab user in grad school, I know how strict Mathworks is with only giving access to those with valid licenses. I wasn't sure if that might be the case here.

SRGDamia1, 2018-12-14:
This was *way* easier than I thought it would be!

SRGDamia1, 2018-12-14:
Ok.. maybe not perfect.  It worked on the ESP but not on a digi bee.

SRGDamia1, 2018-12-18:
Really done in #208 .  Tested with everything but a SIM800

closed 2018-12-18 by SRGDamia1