EnviroDIY/ModularSensors #189: Support sending data to Blynk

SRGDamia1, 2018-11-06:
https://www.blynk.cc/

Would need to test the compatibility of the Blynk library with this library and write and test an example.

This is very low priority, but it might be nice for people who want to see their data on a mobile app like what Blynk provides.

SRGDamia1, 2018-11-06:
REST API documentation: http://docs.blynkapi.apiary.io

aufdenkampe, 2018-11-06:
I think this is worth exploring too.

SRGDamia1, 2018-12-14:
The actual method an Arduino using the Blynk library publishes with is not the REST API, but it's not documented either.

Apparently if you have a private Blynk server there is also a beta grade MQTT endpoint, but it's also not documented.

SRGDamia1, 2019-03-06:
https://community.blynk.cc/t/new-mqtt-5-spec-is-coming/28876