EnviroDIY/ModularSensors #193: ATLAdata.envirodiy.org  DNS lookup

neilh10, 2018-11-09:
With a Mayfly0.5b/XBEE S6B wifi module on a stability test, I'm periodically getting it failing on looking up the IP data.envirodiy.org

Sometimes these failures become unrecoverable. I've got one case where it failed, then next time succeeded, before failing completely until a Mayfly reset.
This happened when the data.envirodiy.org was upgraded over the weekedn - and then I found its just happened yesterday.
When I press Mayfly reset and it does a softrestart - it connects successfully.

For this test the Mayfly sesssion started at 2018-11-05 18:25:00 then started failing 50hrs later at 2018-11-07 20:40:00. 
The log of the failed section is attached.
[ATLAdata.envirodiy.org-Attempts.txt](https://github.com/EnviroDIY/ModularSensors/files/2564303/ATLAdata.envirodiy.org-Attempts.txt)

It looks like its a XBEE S6B issue - but the solution seems likely to be in the Mayfly or TinyGSM.
When it is doing a lookup ATLAdata.envirodiy.org, it retries for up to 45seconds, sometimes retrying as much as 200+ times. The signal strength appears good
I've asked the question on digi forum - http://cms.digi.com/support/forum/69017/xbee-s6b-dns-lookup-unsuccessful.

One result of an ERROR is that ends up wasting a lot of power while it retrys each time. 

One solution might be to cache the IP address, and then using this until there is an error, and then attempt a DNS lookup. 
Alternatively if the number of retrys is fast it could do a TinyGsmClientXbee.restart() the Wifi.
This would also save power.

The code is
TinyGsmClientXbee.h:719: modemConnect()

So just sharing this to see if anybody has seen it, or got any thoughts on a solution.

The "LA command"
Performs a DNS lookup of the given fully qualified domain name (FQDN) and outputs its IP address.
When you issue the command in API mode, the IP address is formatted in binary four byte big-endian
numeric value. In all other cases (for example, Command mode) the format is dotted decimal notation

SRGDamia1, 2018-11-09:
It's an Xbee problem; sometimes the look up just fails.  I've let it keep retrying for up to 45 seconds because often it takes a long time to fail the first time, then will fail very quickly several more times and then after failing half a dozen or more times, suddenly succeed.  It is much more stable if you know the IP address in advance.

neilh10, 2018-11-09:
Gosh interesting,  thanks for the history. Great code to work with.
 
I prototyped the above:
https://github.com/neilh10/TinyGSM/commit/da5d52b92850fbbd6b670126fe66429f7b108019
cache the IP address (static bool gotIP),  
      and use it (ignore DNS lookup with !gotIP) till there is a POST error,  
      then force a a DNS lookup next time round (gotIP=false)
if the number of retrys on DNS Lookup ( int8_t dnsLookUpRetrys=15) exceeds 15 in 45seconds I do TinyGsmClientXbee.restart(). Using restart() forces a lot of things, so I haven't studied its impact. Not sure I've got everywhere that I need to for a DNS Lookup (gotIP = false);

To test, I turned off the dedicated WiFi test network  - a bit crude, but forced the algorithm.

I ran it over overnight on a 2minute update, 420+updates.
The previous failure was after two days on a 15minute update or about 200updates, so hopefully comparable.
At sampleNum=36 and 42 I got back an autonomous 504 error, which seems to indicate it did get a tcp/ip link but some intermediate process timed out, it did record in data.enviroDIY.org

SRGDamia1, 2018-11-09:
If you look, I did hard code in the IP for NIST for the XBee because it nearly never resolved correctly.

neilh10, 2018-12-21:
I'm closing this - I've implemented it as above. It does improve the lookup on the bench and I've had no problem with it. It should save power with devices in the field.

closed 2018-12-21 by neilh10