EnviroDIY/ModularSensors #197: OLED support in Modular

fisherba, 2018-11-19:
It's a low priority from the standpoint of required sensors, but I'm wondering if we can support OLED displays in the library... for demos, testing, training, etc. Not sure how to handle which values to display.print... of if that's complicated. If it is complicated, this request may not be worth the energy.

https://github.com/switchdoclabs/SDL_Arduino_SSD1306

aufdenkampe, 2018-11-19:
I think there could be some value to this, even also for some level of trouble-shooting/confirmation when deploying stations.

If we do this, it might be worth considering migrating library dependencies.

All our EnviroDIY code presently uses https://github.com/switchdoclabs/SDL_Arduino_SSD1306, because we often purchase the SwitchDocLabs [Grove 128x64 I2C OLED Board for Arduino](https://www.amazon.com/gp/product/B01D5GLDJ2) from Amazon ($14.45). However, that library hasn't been updated since it was launched in 2016, is not well documented, and was a poorly done fork and merger of two Adafruit libraries.

We might thus want to consider modifying our code to use the two well-documented Adafruit libraries (which are also highlighted in this tutorial: https://learn.adafruit.com/monochrome-oled-breakouts/):
* https://github.com/adafruit/Adafruit_SSD1306
* https://github.com/adafruit/Adafruit-GFX-Library
These were the foundations for the library from SwitchDocLabs, but they are are continually being updated.

SRGDamia1, 2018-11-19:
So, I've thought about adding display support a number of times in the past, but I don't quite think it's worth it.  Even after figuring out what to print and how to deal with the power draw or disconnection, you'd still only be gaining a tiny screen that you can't even scroll on.

Using an "OTG" cable and a serial port emulator on a cell phone works really nicely for watching sensor output after pushing the "test" button or any time you're debugging.  The only place that hasn't worked well for me is down in our wet lab where I have a Mayfly set up without a battery and power being supplied via a micro-usb low-amp cell phone charger.  It just means that if I want to watch what the sensors are doing and don't want to restart the board before and after I have to carry a battery with me so I can unplug it.  And even there, when the OTG cable is plugged into the phone, the mayfly will draw power from the phone and work fine, it just turns off while the cables are switched.

I have  a Samsung GS7 with this cable: https://smile.amazon.com/gp/product/B01M5GZ3N0 and this emulator app:  https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal

neilh10, 2018-11-19:
@SRGDamia1 thanks for the the cell phone monitor reference - I had been thinking of trying to do that. 

One of the thoughts I've had - and been documenting it for myself - is to have a "user interface" that for installation would provide positive feed on success (happy steady green) or an error indication (nervous red flashes) -  and also be visible for a  maintenance visit. 
In this case the basic "User interface" is a glorified words for two output LEDs and two input switches. 
Its basically identifying the stages an "installer" goes through, and also when visiting the site how to easily get reasonable confidence that everything is working. 
If you have to walk some distance to get to a site, its so nice to be able to get some easy confidence its working.
So it could be the two LEDs as basic, ( a screen for more advanced - with say a scrolling list of measurements from installed sensors)
For an input the BUtton1 can be used, or in the past I've renamed the reset button and called it an "Attention" button - so the user presses it to get the ATTENTION of the system - LEDs defining the system state.

It was a real hit with the hydrologists and biologists I was working with as they didn't have to get the computer out to see what it was doing.

For power saving, after say 30minutes of no user activity , the LED (OLEDs Screen) would be turned off.

so just a thought as to perhaps a bigger picture.

neilh10, 2019-08-11:
I've been experimenting with an adafruit eInk - that is screen stays visible when power is removed.
It recommended to use a slower update rate ... so periodic system status   ... but very workable. The Tri-state RED takes about 20seconds to change from black pixels to RED pixels  https://www.adafruit.com/?q=eInk  
The libs where very easy to use -https://github.com/adafruit/Adafruit_EPD      https://github.com/adafruit/Adafruit-GFX-Library,
From my own experience, a UserInterface benefits from a structured view of what the user needs to see. That is its valuable to think of the boards LEDs and a Textual display together. (part of a USER STATE HAL https://github.com/EnviroDIY/ModularSensors/issues/244)   
For a project the first stage is design the mechanical placing of the Textual display, and one specific solution is for a Feather format such that it can be stacked easily and connect electrically - https://www.adafruit.com/product/4128   I built a trial in a few hours on top of Feather SAMD processor.