EnviroDIY/ModularSensors #198: mayfly.ini reader

neilh10, 2018-11-23:
I've created an ini reader - it reads from the SD card, and can then be used to customize each specific cards environment. 
Would this be of interest?
https://github.com/neilh10/ModularSensors/commit/e76b4ba6042af8e95b88abc09fbe80562c071d1d
An example is on the sd card the following file is created (Windows10 in my case)
mayfly.ini  

>  [COMMON]
> REGISTRATION_TOKEN=7b19191b-b3d6-416d-9e07-4fbe549f2493
> SAMPLING_FEATURE=b918f150-4fff-424d-813a-a69a808a93b6
> ;apn
> WiFiId=testNetSsid
> ;WiFiPwd
> LOGGING_INTERVAL=2
> 
> [UUIDs]
> f40a9dff-2c54-43ae-a016-1c5095c188eb=1 ; Gage height (Keller_Acculevel_gageHeight)
> 7bc30855-ae08-4865-ab89-a8e17df93bfc=2 ; Temperature (Keller_Acculevel_Temp)
> 3907922a-56fe-46f3-a56e-9de6b77d3679=3 ; Temperature (EnviroDIY_Mayfly_Temp)
> 2c58e64d-6b66-4d9e-b893-bfdb10b65426=4 ; Battery voltage (EnviroDIY_Mayfly_Batt)
> 0cf94fc8-a5d2-4fbe-82f2-2a81650575a8=5 ; Percent full scale (Digi_WiFi_SignalPercent)
> 9e47e1aa-2ef6-4283-9a25-537cfd78b17b=6 ; Volt0 BatV All_ExternalVoltage_Battery
> ebddd4d7-562d-481b-8356-d1f464fc5685=7 ; Volt1 Solar All_ExternalVoltage_Battery
> 08b2e561-21fd-44cb-b429-c2be536c7bd9=8 ; Ina219_V All_ExternalVoltage_Battery
> 804be0c8-3dc0-4be6-9537-63fe8240e98f=9 ; Ina219_mA All_ExternalVoltage_Battery
> 

The result is a repeated call from an iniHandler_fn() with the following pairs, and comes out on debug as 

> ***parseIni
> Successfully connected to SD Card with card/slave select on pin 12
> [COMMON]
> REGISTRATION_TOKEN=7b19191b-b3d6-416d-9e07-4fbe549f2493
> SAMPLING_FEATURE=b918f150-4fff-424d-813a-a69a808a93b6
> WiFiId=testNetSsid
> LOGGING_INTERVAL=2
> 
> [UUIDs]
> f40a9dff-2c54-43ae-a016-1c5095c188eb=1
> 7bc30855-ae08-4865-ab89-a8e17df93bfc=2
> 3907922a-56fe-46f3-a56e-9de6b77d3679=3
> 2c58e64d-6b66-4d9e-b893-bfdb10b65426=4
> 0cf94fc8-a5d2-4fbe-82f2-2a81650575a8=5
> 9e47e1aa-2ef6-4283-9a25-537cfd78b17b=6
> ebddd4d7-562d-481b-8356-d1f464fc5685=7
> 08b2e561-21fd-44cb-b429-c2be536c7bd9=8
> 804be0c8-3dc0-4be6-9537-63fe8240e98f=9
> 
> [comment_test]
> test1=1;2;3
> Parse ini; mayfly.ini
> Completed.

SRGDamia1, 2018-12-19:
I'm puzzled; what functionality would this add?

neilh10, 2018-12-19:
Well ... its all about scaleability.  Defined parts of the per system configuration reside on the SD card in Mayfly.ini - not in the compiled software.

Currently ModularSensors requires a trained programmer/scientist to be attached for each physical node deployment.  Its going to be very inefficient in peoples time in the field.

For a project that has a defined ModularSensors platform - ie the instrument configuration is defined, then one tested image can be deployed across a number of physical platforms, and all the customization happens in the .ini file. 
That is could have 10 Mayfly physical nodes, all with the same known working image - downloaded through a standard AVR utility (for Mayfly). All the keys and passwords are then managed and configured in a uSD Mayfly.ini   - this I believe simplifies the field management, and would allow more team members to manage the physical nodes, not just the one person who has ModularSensors on their computer.   
Alternatively, could have a popular myEssentialModularSensors.HEX  build published on a web site, and all a user would have to do is download it to a physical Mayfly, configure a Mayfly.ini on a uSD and they would be operational.   

I currently have it tested and working, but I haven't decomposed it into Classes .

SRGDamia1, 2018-12-21:
I think there are more unique settings that need to be programmed than are practical to read from an ini file.

When we've needed to update or change, we walk out with a new programmed board and exchange, which isn't really much more hassle than carrying an SD card.  Or we've carried out a laptop.  We often have a laptop in the field anyway at station visits to download other sensors.  I think there are tutorials online on how to use one Arduino to program another, to avoid the laptop.

I think the functionality you are really looking for is an ability to reprogram from an SD card or, better yet, over the air.  That is simply not possible with the bootloader on the Mayfly or most other typical Arduinos.  You need a different bootloader.  I've looked into it.  If you could write a working bootloader that would have that capability, that would be awesome.  I know @aufdenkampe would really love it.  But that's way beyond this library.

SRGDamia1, 2019-01-20:
https://github.com/zevero/avr_boot - to read the full program from the SD card

aufdenkampe, 2019-01-21:
@neilh10, thanks for experimenting with different options for scalability. These are really very important to explore and I really like your ideas.

@SRGDamia1, it is true that the ideal solution to scalability would be over-the-air updates, similar to how Particle.IO boards (and others) function. This is something that I've discussed separately with colleagues at LimnoTech (@gcutrell) and also with @fisherba.

If over-the-air updates are not possible, then updating from an SD card could be a great option! The idea of mailing a partner an SD card to swap in and then hit the resent button -- that could be very, very appealing. 

@SRGDamia1, do you think we might be able to do that using https://github.com/zevero/avr_boot? Then we could merge in the work that @neilh10 has prototyped in this issue.

SRGDamia1, 2019-01-21:
The bootloader I linked and Neil's reader do different things.

Neil's work (I think) looks for the ini to tell it what the UUID's and logging interval should be, but it doesn't change anything else.  So, the sensors attached, the order of the array, and the underlying functions all stay the same.  It would be useful for moving a programmed mayfly to a new site with identical set up except for the site itself.  So a user could yank their sensor station, write the new uuid's to the sd card and put it back out without the board itself touching a computer.  The advantage of this is that the user has a plain text file to look at to see their settings which could (possibly) be less scary than seeing the Arduino code.  The board also continues to act just like a regular Arduino any time you want to do something completely different with it.  The disadvantage is that you wouldn't ever be updating any of the underlying programming, so you can't get any new functionality.  And, although you might possibly make the ini seem less scary than the ino and editing the text in notepad and saving might feel a little less scary than editing and hitting the upload in an ide, almost every line in the logger program really is a setting for the user to change, so your ini file would end up being just like the ino pinned at a (possibly) out of date version.

The avr_boot I linked would actually read a completely new pre-compiled program hex file from the SD card and then begin running that new program.  The advantage would be that you'd get to change every setting, including adding sensors, and you'd could update everything under the surface as well.  The disadvantage is that it's no longer a text file on the SD card, it's a binary file so you can't see what it is.  And, probably more crucially, after you put on this bootloader, your board would, in some sense, no longer be an Arduino.  You would have to re-burn the Arduino bootlader on it to be able to upload programs in the normal way again.  Programming the bootloader is not the same as programming the firmware.  You need the right cables and set up to do it.

To upload over the air, you would again need a different bootloader that could look for a new firmware online and then use that, just like avr_boot does from the SD card.  I have seen a few people mentioning bootloaders that might work that way over UDP or TCP when the board is connected over ethernet to a secured network, but I haven't seen anything "Arduino" style that's more over-the-air than that.

neilh10, 2019-01-22:
Some great thoughts here - good find @SRGDamia1.

  I've hope this is OK to track it I've opened https://github.com/EnviroDIY/ModularSensors/issues/219

The intent that I had for "Mayfly.ini" which I'm now calling "ms_cfg.ini"  is what I've implemented with a release I just made -  https://github.com/neilh10/ModularSensors/wiki/Feature-INI-file 

Its turned out to be quite interrelated to a number of classes - as the persistent customer configurations needed to be pushed down into those classes.  https://github.com/neilh10/ModularSensors/wiki/Release-Notes

The end objective is that I have one firmware.hex (taken .pioenvs\mayfly\firmware.hex) that is given a release name eg mayfly_v0.17.3.release1a.hex that I (or any person visiting the location) can load into any mayfly rev0.5B with avrdude and then customize with an ms_cfg.ini on the SD card.

Customizations that are mandatory are the UUIDs, but its also nice to have the .log file that represents the core data be a unique file name for that geographical site.

When/if new functionality is released, then the Mayfly firmware can be upgraded (.. avrdude, SD card, OTA), and the core site functions/identity are in the "ms_cfg.ini"

SRGDamia1, 2019-01-22:
I'll leave this open because I think it's reasonable to have this library look for some configuration information from an SD card.