EnviroDIY/ModularSensors #203: Make sure we're printing to client as few times as possible with as many characters as possible each time

SRGDamia1, 2018-12-03:
I *think* we might be getting "charged" TCP overhead (~40 bytes per packet) with every new print.  I'd set the post up as many different print statements to reduce the size of single strings, but this might be hurting us.

SRGDamia1, 2018-12-03:
This won't make any difference for the "XBees" but might for all the other modems

neilh10, 2018-12-04:
I was wondering about this. 
I was thinking about about making a processorStats - commsBytes -  a count of actual tx/rx (as far as possible to count) sent to the modem.

SRGDamia1, 2018-12-06:
The number of bytes sent to the modem is different than the number of bytes of data consumed on the cellular plan.  From what I've read, every data packet sent on the TCP connection requires a minimum of 40 bytes of TCP overhead,  The modem tacks that on automatically (invisible to both ModularSensors and TinyGSM) so the Tx/Rx sent to the modem may not be a very good reflection of how much data you'll comsume.

I'm assuming most of the modems just put on the overhead and send data every time a tcp send command is called rather than attempting to maximize packet sizes and minimize overhead.  So to reduce data usage, we want to maximize the packet sizes ourselves, but not at the risk of creating a String to large to be held in the RAM of the mcu.

neilh10, 2018-12-06:
The  TCP/IP stack packetizes data to transmit it over the network, so that means for a particular sending stream you could characterize what the overhead is. For very slowly transmitted data (1byte every second?) it could be 40bytes per payload, but for a fast stream of user data (~90bytes per second@9600baud) I would expect more data per network packet.   At least by collecting the tx/rx data counts it would be possible to get an idea of how much baseline data is being pushed, which really is the only part that can be controlled. For a wireless signal that is very weak, its likely that there would be a lot of re-transmission - pushing up the network counted data.

SRGDamia1, 2018-12-14:
Completed on develop.

closed 2018-12-14 by SRGDamia1