EnviroDIY/ModularSensors #205: Whats the best way of testing a PR

neilh10, 2018-12-06:
Just wondering if you have a preferred way code in a PRs, is submitted and tested.
I'm currently based on examples/logging_to_EnviroDIY/logging_to_EnviroDIY.ino

As far as I can see, in the recent PR/204 I just submitted, I wasn't able to figure out a way to easily test adding the new sensor src/sensors/TiIna219.*

As it is being submitted to src/sensors ... it wont' copy into the new environment on build.
So I copied it to examples/logging_to_EnviroDIY/lib/src/sensors/TiIna219.*
and then tested that it built there wrt to 
examples/logging_to_EnviroDIY/logging_to_EnviroDIY.ino

Is there any other method you would recommend?

SRGDamia1, 2018-12-06:
I'm sorry; I messed up Travis so it won't work for pull requests.  I'll add fixing it to my list.

neilh10, 2018-12-06:
Oh no problem.. there are a lot of moving parts with all the versions being compiled for all the examples I imagine. I'm very happy to fit in with the way you have it setup.
I was just checking to see if you had any feedback on the way you'd want someone else to submit. Its setup to make it easy for a user to pull in all the examples and reference a fixed lib. Just makes it a little non-linear to submit into it - but that's OK.

SRGDamia1, 2018-12-13:
Travis should be working correctly again for pull requests.  You should merge the changes from develop into your branch and then the CI build test should work again.

closed 2018-12-13 by SRGDamia1

neilh10, 2018-12-20:
@SRGDamia1 created this  https://github.com/EnviroDIY/ModularSensors/wiki/Developer-Setup