EnviroDIY/ModularSensors #211: Discussion on user/developer roles

neilh10, 2018-12-21:
I'm curious if there might be any thoughts or pointers to what is a  "User" role with ModularSensors.   

It seems to me, as a relative newbie to the project, the "User" role is the target user for ModularSensors.    
This is a very typical discussion in the world of software that I've been part of - what set of skills does the "msUser" a)have   b)need.    Essentially what it takes to make the software package accessible to the "msUser". Is terminology useful eg "msUser" or more personally "myUser"?

It seems to me, with ModularSensors the goal appears to be for a "msUser" role capable of  "architecting a logger" , for the  purpose of implementing an outdoor environmental research project.

The msUser roles appear to me to be   
msUser-Mayfly/Comms-FieldResearcher   
msUser-Mayfly/Comms-teacher   
or more generally for a Solar Powered data logger   
msUser-[Mayfy | Autonomo]-[FieldResearcher | teacher]   

All other roles for ModularSensors pkg are defaulted to the developer.   
So just to itemize different skillsets likely in the developer role
developer[historical configurations easily testable]-software   
developer[solar platforms Mayfly | Autonomo]-software   
developer[platform]- document-er/teacher syllabus development    
developer [platform] - new instrument integration
developer-hardware (new hardware with appropriate software instrument integration.    
Unstated is the Quality Testing that is implicit in the developer community.  With large software teams - Scrum provides a template for thinking of roles - though this is just focused on ModualSensors and not the larger world of how to catch the data in the cloud. 

I'm personally in my own ModularSensor fork looking at a  "Field Implementer/technician" role with a subset of "User" skills. Everything of the "User" role except compiling software.

For the "msUser" role I've explicitly referenced outdoor solar  platforms, as I believe this is the target - a very hard set of skills. The "msUser" will need to have and learn specific hardware and able to manage the software deployment for a  realworld geography  that includes the wireless radio horizons and the types of communicating devices that work in it. These are fast evolving technologies - and typically the learning starts with the most current "easy" communicating  technologies - eg cellular 3G.

The developer on the other hand is going to be working with scale-able platforms (Mayfly/Autonoma/Cortex M0?) and next generation communication technologies (LTE) and newer opportunities of MQTT and? other opportunities ( MESH networking?).   
There is some nuts-and-bolts requirement to support historical platforms while they are in use by specific groups.

So just happened to think of this in the middle of the night, as it related to other software teams that I've been part of.

Any comments welcome - including that ModularSensors just needs to grow as needed.:)

SRGDamia1, 2018-12-22:
I think you might be imagining this project to be much bigger and more complex than it is.

@aufdenkampe, @darscott, and @dbressler75 are all much better at outreach and all kinds of use interaction than me.

neilh10, 2018-12-26:
its sort of more a vision-thing - I'm just sharing the work i've done for a meta-analysis - sort of thing I use for myself - and then sharing it here in case its useful. 

I'm ask myself how would  a Modular Sensors working unit be used by  the fishery biologists, fresh water biologists, hydromorphologists & hydrologists that I know at local chapters of tnc.org/ tu.org that I've worked with. Sort of thing I did in 2010 with my own instruments, and then in 2014 with an Onset U30 and blogged in 2016 https://www.envirodiy.org/accurate-low-stream-depth-measurements-viewable-on-internet/   
My experience with land managers (deploying in regional streams on the west coast), is two people are needed for safety, and one of the biggest cost is budgeting for "two people days" for a site visit. The site visit is either deploying a number of gauges, maintenance on gauges/stream flow measurements. 

so just sharing my analysis.

neilh10, 2019-11-10:
I'm closing this as it was sharing what I thought was needed for a role that just programs a hex file into a device - https://github.com/neilh10/ModularSensors/wiki

closed 2019-11-10 by neilh10