EnviroDIY/ModularSensors #215: Y511 giving 0.00, but real numbers during `DEEP_DEBUGGING`

aufdenkampe, 2019-01-02:
@SRGDamia1, do you have any idea why the Y511 isn't giving us real data in a normal sketch, but does when we do extended testing. I think it has to do with not waiting enough stabilization time. See:

```cpp
// Sensor Specific Defines
#define Y511_NUM_VARIABLES 2
#define Y511_WARM_UP_TIME_MS 500
#define Y511_STABILIZATION_TIME_MS 22000
#define Y511_MEASUREMENT_TIME_MS 1700
```

Where would I find this bug?

fisherba, 2019-01-02:
Adding more observations. I changed the warm up time to 5000 and turned off debugging. The first two iterations in sensor testing mode gave 0 NTU, then the rest of sensor testing gave real numbers. When testing mode times out and the sensor starts logging, it reports 0 NTU again.

fisherba, 2019-01-02:
Settings are: 
```
[env:mayfly]
board = mayfly
platform = atmelavr
framework = arduino
lib_ldf_mode = deep+
lib_ignore = RTCZero
build_flags =
    -DSDI12_EXTERNAL_PCINT
lib_deps =
    EnviroDIY_ModularSensors@=0.17.2
    https://github.com/PaulStoffregen/AltSoftSerial.git
    https://github.com/EnviroDIY/SoftwareSerial_ExternalInts.git
```

aufdenkampe, 2019-01-03:
I spent a bunch of time playing around with timing, with `#define DEBUGGING_SERIAL_OUTPUT Serial` turned on in the `YosemitechParent.h`.

To get 5 different, non-zero in a row required that I set the timing to:

```cpp
// Sensor Specific Defines
#define Y511_NUM_VARIABLES 2
#define Y511_WARM_UP_TIME_MS 8000
#define Y511_STABILIZATION_TIME_MS 40000
#define Y511_MEASUREMENT_TIME_MS 4000
```

This may not be the minimums for these, but it's the best I can do for now.

By the way, my Y511 has software version 2.6 installed (ordered in Nov. 2018).

NEXT STEP is to commit these changes to the code.

SRGDamia1, 2019-03-06:
Did changing the timing work for you? @aufdenkampe

aufdenkampe, 2019-03-06:
Yes!

closed 2019-03-06 by aufdenkampe