EnviroDIY/ModularSensors #219: mayfly boot off SD card

neilh10, 2019-01-22:
Some great comments on closed https://github.com/EnviroDIY/ModularSensors/issues/198 - so creating a "new feature" to capture it.

Incorporate a "boot code" reader that is activated at reset and looks for a specific filename on the SD card with a version of software that is greater than the current  ModularSensors in flash. 
 
@SRGDamia1, found https://github.com/zevero/avr_boot

Architecturally this is pretty standard model for an easy local site upgrade using only an SD card. 

This "boot" program is a totally seperate program loaded into a portion of flash memory that is never erased. The only action of the program is to look for a specific file on the SD card that represents an upgrade to any version that may already be resident in the flash. 
If the version on the SD card is earlier or the same as the "ModularSensors" version already in flash, the boot takes no action and jumps into the "ModularSensors" in flash.
If the ModularSensor on the SD card is newer, then the boot erases just the ModularSensors in the flash (taking care not to erase the boot/itself) and then flashes the ModularSensors.  When complete it typically performs a reset and starts again.

Practically speaking a ModularSensors boot 
a) also needs to determine if there is adequate power available to do the upgrade before it starts. 
b) that the new program on SD card is valid - typically done with a simple CRC32 performed across the sd firmware.bin program and the answer should be known value.   A valid program is a lower threshold than a "signed" program.

Typically a boot feature is a precursor to doing remote OTA upgrade. For a remote OTA upgrade, the new version is downloaded to SD, and then when the new file is verified as accurate

The other way for a local site upgrade is through "avrdude" - which would require a computer to be taken to the monitoring  site, but would only require the operator to understand how to connect the USB and operate the avrdude - https://www.nongnu.org/avrdude/

SRGDamia1, 2019-01-22:
I think writing a new boot loader is well beyond the scope of this library.  It's really completely different functionality. A program made using the modular sensors code and compiled into firmware could be flashed vs SD card or over the air, but the actually flashing functionality is a different beast.

That's not to say it would not be useful, but I don't want it to be an intrinsic part of this library. It belongs in another bootloader focused repository.

Changing the bootloader makes the board not really an Arduino any more. I definitely do not want to make this library in any way *dependent* on a non standard bootloader.

aufdenkampe, 2019-01-23:
@neilh10, I really like these ideas! Thanks for thinking them through. Such functionality would be exceptionally useful for those of us deploying (and maintaining) large networks.

@SRGDamia1, I also fully agree that we don't wan to make this ModularSensors library rely on a non-standard boot loader. The intent for this library is and always has been to make it *easier* to program environmental datalogger functions onto an Arduino-framework board.

So the right solution would be to create another repo for this new bootloader approach, and then potentially build in the optional functionality into ModularSensors to also work with that new boot loader.

For those reasons, I would like to keep this issue open, so we don't loose track of this goal, even if such and effort will largely be done in a new repo.

neilh10, 2019-01-23:
From my experience and understanding of bootloaders   they are integral to the final software production version, but aren't necessarily used during development. They are to ease the upgrade capability in the field, and that is a productivity boost to the product - ie a developer doesn't have to be shipped with each release.
 I'm working  on a product (a 10yr old commercial product)  that has this via a USB stick  -it uses a 32bit PIC32MX. If I bring it up in the ICE (emulator) i blow away the production bootloader that is programmed at the factory, however most of my development I build the customers code (like ModularSensors) and then upgrade the target via the USB bootloader.   For the customers to upgrade they are sent a USB stick

So not sure what is hardware 'arduino' - is it something approved in Italy or by Massimo or gets pushed into the development Arduino IDE.  In that case Mayfly is a fantastic addition to the Arduino hardware repository, but I'm so fed up with the Arduino 'baby steps' IDE, that its only the PlatformIO makes it possible to really code.

so looks to me like @SRGDamia1 find of https://github.com/zevero/avr_boot  was pretty interesting.

To make it work and keep it simple, it would be to follow its instructions - build avr_boot and load it in (via IDE)  and then load in the ModularSensors from SD card.  How that that works depends on how relocatable AVR code is.

So I think it would be a fork of the avr_boot  github/EnviroDIY image, but ModularSensors would need to be committed to being loadable by the avr_boot - which may be simple if its relocatable, or might require a tweak to the linker file if its not relocatable.

SRGDamia1, 2019-01-23:
I don't think there's anything about the library that would make it not relocatable. I'm not trying to interact directly with any memory addresses or anything of that sort. I imagine a compiled program built with ModularSensors would work identically regardless of the bootloader.

SRGDamia1, 2019-01-24:
In this case, by being "Arduino compatible" I mean easy programmed with only a USB cable and either PlatformIO or the Arduino IDE.

The Arduino IDE is terrible, but to my knowledge it is the only easy option for programming the bootloader. PlatformIO doesn't support writing the bootloader.

And you still need a dedicated programmer, or at least another board that has been programmed as a programmer, to write the bootloader. You can't do it with just a USB cable.  I'm not even sure the "Arduino as ISP" option would even work for the 1284p.  Most things I've read online say it's good for a 328 but not much bigger.

neilh10, 2019-01-25:
Collecting references here is 2009 description. https://www.avrfreaks.net/sites/default/files/bootloader_faq.pdf

Yes arduino and wiring has made it so much easier to get going. We used to spend ages getting the development environment going - and then came Arduino - plug in a USB cable and bingo, can do a blinkey.  The basis of it is the boot ROM in the Atmel processors and so nice tutorials, and the IDE that loads in all the gcc tools!!!. Then PRJC.com figured out something similar with an small external chip using JTAG  https://www.pjrc.com/teensy/ and Siemens/Nxp has also put in a boot rom in the Kinetis KL27P .

Others are following with the next generation devices - eg one of my favourite bloggers - a guy in Switzerland https://mcuoneclipse.com/2019/01/22/tutorial-booting-the-nxp-i-mx-rt-from-micro-sd-card/#more-25311