EnviroDIY/ModularSensors #224: Split up the begin for the logger

SRGDamia1, 2019-02-08:
Want the RTC sync to come before the file start

neilh10, 2019-02-09:
@SRGDamia1 It would be nice to know why? What are you trying to fix? Mayfly with battery or ?

For smart power analysis reasons, I've implemented a version where RTC sync is a lower priority than power analysis and first file time recorded.  
That is  on a power up (reboot?) if there is not enough power to make an internet connection there is little point in attempting an internet connection.  
assumption 1) if there is valid RTC time - it can largely be trusted. Once an internet connection is initiated then the time can be synchronized. 
Assumption 2) if there isn't valid time, this is probably being set up for the first time and is in a safe environment with enough power so probably will get the time setup when it connects. A file with the initial static date is OK, as it will be superseded next reboot. In the project Software Architectural language (my take), there is a requirement to handle the slow process of harvesting solar energy, and there is no requirement to have an accurate file date from first boot.
Assumption 3) Prioritize errors, and only deal with a single highest priority error. Having no valid power is an architectural possibility. For a running Mayfly, having no valid time is an error condition, either because hardware has gone bad or RTC battery has runout. Both are basket cases and require a field visit.

The primary driver for smart RTC update,  is  KISS principle of  available battery power. 
Putting the RTC sync update as part of the first internet update keeps the power analysis simple .

Just my 2c and how I've cone to prioritize error conditions.

neilh10, 2020-07-08:
Can this be closed. I make it a debug dependency.

closed 2020-07-08 by SRGDamia1