EnviroDIY/ModularSensors #225: Test Atlas sensors

SRGDamia1, 2019-02-08:
@acgold - I've re-written your code to give it more class inheritance and be more in line the rest of the library.  Could you please make sure it still works?

aufdenkampe, 2019-02-08:
@acgold, thanks for contributions to this via https://github.com/EnviroDIY/ModularSensors/pull/221!

Having you test Sara's new code, and add documentation (#226) would complete that contribution! Thanks in advance.

@SRGDamia1, should @acgold test from the Atlas branch at https://github.com/EnviroDIY/ModularSensors/tree/atlas, or is your re-written code in develop?

BTW, thanks @SRGDamia1 for accepting the PR and working it in!

SRGDamia1, 2019-02-09:
It's all in the atlas branch. I made enough changes that I want someone to make sure it still works before I pull it any further.

Adam, so you understand what I did:

I made a parent class that has all the functions and the unique sensors can inherit the functions - they don't reach have a cpp any more. It looks much more confusing because you can't see everything the sensor did in one file, but it is much easier to maintain since you no longer have several different chucks of code that are almost the same but only one copy. So now if you find a mistake, you only have to fix it on one place. 

I changed all of the Serial.print() functions to MS_DBG preprocessor macros. This makes it easy to turn all of the print outs on and off.

I broke up your add measurement result into the three functions it had : first the function to start the measurement (write the r), then the function to read the result, and then the function to put it to sleep. This is the way all of the sensors work. Breaking the functionality into all the little bits means that we can tell one sensor to start a reading and then talk to a different sensor before coming back to the first to collect the value. It saves a lot of waiting around.

acgold, 2019-02-10:
@SRGDamia1 and @aufdenkampe, I'm happy to test it out and add some documentation.  I'll hopefully be able to test it this Friday (2/15) or over that weekend. Thank you so much for cleaning up the code and adding it in! 

@SRGDamia1, is there a built-in way to stagger measurements using the Modular Sensors format? I'm hoping to eventually add-in temperature compensation for the DO and EC probes, so the order or data collection ideally would be temp (sending the probe the command "R"), DO (sending the probe "RT,[temp reading]"), then EC (sending the probe "RT, [temp reading]").  Basically, I'm wondering if I should make a new cpp and h file to do this or try to edit your new code. Thanks!

SRGDamia1, 2019-02-10:
Right now there is no built in way to do that.  All of the sensors are completely independent of each other. You could easily do the temperature correction yourself with a calculated variable, but you're have to set things up differently to send the value to another sensor.

SRGDamia1, 2019-02-10:
I think if I really wanted the correction to be done on-board I'd do something like set up an overload constructor for the atlas sensors that accepts as input another sensor and variable position. I'd have to think a bit more about the timing implications, though. Your also have to create a new set of variables for the temperature corrected result; specific conductance vs conductivity, etc. Since you'd use the same command for multiple sensors, I'd keep the code in the parent cpp.

acgold, 2019-02-11:
The sensor just does the math for you to cut down on data processing later. It seems like doing temperature compensations after downloading the data from the Mayfly will be simplest method rather than doing the compensations during data collection. Thanks for your help!

SRGDamia1, 2019-02-11:
You can still do the compensation on the Mayfly as a calculated variable so you can send the data live. That part the library is already set up for. It's having the Atlas do it that would be tricky.

acgold, 2019-02-15:
@SRGDamia1 I'll look into using the calculated variables, that would be helpful to have temperature compensated data live..

I tested the code and it worked great! Thanks for incorporating it into ModularSensors!

SRGDamia1, 2019-02-15:
The https://github.com/EnviroDIY/ModularSensors/blob/master/examples/baro_rho_correction/baro_rho_correction.ino example is the best for seeing a calculated variable in action.

Onset has some nice info on calculating specific conductance here:  https://www.onsetcomp.com/files/manual_pdfs/15260-B-Conductivity-Assistant-Calculations-Sea-Water.pdf

SRGDamia1, 2019-02-20:
@acgold Adam - please retest.  Make sure you're getting 4 results from the EC.

acgold, 2019-02-20:
@SRGDamia1 I plan on retesting the EC this coming Friday (2/22). Thanks!

SRGDamia1, 2019-02-20:
@acgold - while you're testing, please run this on each of your sensors and give me the results:  https://github.com/EnviroDIY/ModularSensors/blob/atlas/sensor_tests/i2c_warmUp/i2c_warmUp.ino

acgold, 2019-02-22:
Responses from sensors running the I2C warmup test;

Temperature:
```C++
15:20:56.668 -> I2C device replied at address 0x66 after 0 ms, code: 0
15:20:56.668 -> i sucessfully written
15:20:56.668 -> Result available after 4 ms: ?I,RTD,2.10
15:20:56.668 -> Moving to next character - Wait.....................
15:21:01.720 -> I2C device replied at address 0x66 after 0 ms, code: 0
15:21:01.720 -> i sucessfully written
15:21:01.720 -> Result available after 4 ms: ?I,RTD,2.10
15:21:01.720 -> Moving to next character - Wait......................
15:21:06.957 -> I2C device replied at address 0x66 after 2 ms, code: 0
15:21:06.957 -> r sucessfully written
15:21:07.365 -> Result available after 393 ms: 22.362
15:21:07.365 -> Moving to next character - Wait.....................
15:21:12.405 -> I2C device replied at address 0x66 after 0 ms, code: 0
15:21:12.405 -> i sucessfully written
15:21:12.405 -> Result available after 4 ms: ?I,RTD,2.10
15:21:12.405 -> Moving to next character - Wait.....................
15:21:17.422 -> I2C device replied at address 0x66 after 0 ms, code: 0
15:21:17.422 -> i sucessfully written
15:21:17.422 -> Result available after 4 ms: ?I,RTD,2.10
15:21:17.422 -> Moving to next character - Wait......................
15:21:22.680 -> I2C device replied at address 0x66 after 2 ms, code: 0
15:21:22.680 -> r sucessfully written
15:21:23.089 -> Result available after 393 ms: 22.373
15:21:23.089 -> Moving to next character - Wait...
```
EC:
```C++
15:23:56.478 -> I2C Warm Up Timing Test
15:23:56.478 -> Wait.....................
15:24:01.525 -> I2C device replied at address 0x64 after 0 ms, code: 0
15:24:01.525 -> i sucessfully written
15:24:01.525 -> Result available after 4 ms: ?I,EC,2.13
15:24:01.525 -> Moving to next character - Wait......................
15:24:06.780 -> I2C device replied at address 0x64 after 2 ms, code: 0
15:24:06.780 -> r sucessfully written
15:24:07.324 -> Result available after 557 ms: 4893,2643,2.62,1.002
15:24:07.324 -> Moving to next character - Wait.....................
15:24:12.373 -> I2C device replied at address 0x64 after 0 ms, code: 0
15:24:12.373 -> i sucessfully written
15:24:12.373 -> Result available after 4 ms: ?I,EC,2.13
15:24:12.373 -> Moving to next character - Wait.....................
15:24:17.389 -> I2C device replied at address 0x64 after 0 ms, code: 0
15:24:17.389 -> i sucessfully written
15:24:17.389 -> Result available after 4 ms: ?I,EC,2.13
15:24:17.389 -> Moving to next character - Wait......................
15:24:22.638 -> I2C device replied at address 0x64 after 2 ms, code: 0
15:24:22.638 -> r sucessfully written
15:24:23.187 -> Result available after 557 ms: 4897,2644,2.62,1.002
15:24:23.187 -> Moving to next character - Wait.....................
15:24:28.248 -> I2C device replied at address 0x64 after 0 ms, code: 0
15:24:28.248 -> i sucessfully written
15:24:28.248 -> Result available after 4 ms: ?I,EC,2.13
15:24:28.248 -> Moving to next character - Wait.....................
15:24:33.251 -> I2C device replied at address 0x64 after 0 ms, code: 0
```
DO:

```C++
15:26:13.606 -> I2C Warm Up Timing Test
15:26:13.606 -> Wait.....................
15:26:18.632 -> I2C device replied at address 0x61 after 0 ms, code: 0
15:26:18.632 -> i sucessfully written
15:26:18.632 -> Result available after 4 ms: ?I,DO,2.13
15:26:18.632 -> Moving to next character - Wait......................
15:26:23.923 -> I2C device replied at address 0x61 after 2 ms, code: 0
15:26:23.923 -> r sucessfully written
15:26:24.470 -> Result available after 557 ms: 8.14,89.5
15:26:24.470 -> Moving to next character - Wait......................
15:26:29.726 -> I2C device replied at address 0x61 after 0 ms, code: 0
15:26:29.726 -> i sucessfully written
15:26:29.726 -> Result available after 2 ms: ?I,DO,2.13
15:26:29.726 -> Moving to next character - Wait.....................
15:26:34.750 -> I2C device replied at address 0x61 after 2 ms, code: 0
15:26:34.750 -> i sucessfully written
15:26:34.750 -> Result available after 4 ms: ?I,DO,2.13
15:26:34.750 -> Moving to next character - Wait.....................
15:26:39.778 -> I2C device replied at address 0x61 after 2 ms, code: 0
15:26:39.778 -> r sucessfully written
15:26:40.324 -> Result available after 555 ms: 8.23,90.5
15:26:40.324 -> Moving to next character - Wait.....................
15:26:45.369 -> I2C device replied at address 0x61 after 2 ms, code: 0
15:26:45.369 -> i sucessfully written
15:26:45.369 -> Result available after 4 ms: ?I,DO,2.13
15:26:45.369 -> Moving to next character - Wait...........
```

acgold, 2019-02-22:
It all worked great with modular sensors, but TDS on the EC probe shows up with the same metadata as Conductivity in the CSV

SRGDamia1, 2019-02-22:
I fixed the TDS.  When you did the test, did you make sure the I2C was running on **switched** power - check the jumper near the I2C grove jack.  The sensors definitely lost power and had lights go out each time it said "wait" right?  

I stole some sensors from a co-worker and mine responded much, much more slowly.

SRGDamia1, 2019-02-22:
This is what I was getting for the RTD:
```
I2C device replied at address 0x66 after 733 ms, code: 0
r sucessfully written
Result available after 619 ms: -1023.000
Moving to next character - Wait......................
I2C device replied at address 0x66 after 733 ms, code: 0
i sucessfully written
Result available after 4 ms: ?I,RTD,2.02
Moving to next character - Wait......................
I2C device replied at address 0x66 after 731 ms, code: 0
i sucessfully written
Result available after 4 ms: ?I,RTD,2.02
Moving to next character - Wait.....................
I2C device replied at address 0x66 after 733 ms, code: 0
```

acgold, 2019-02-24:
Whoops, my I2C grove port was on 3.3V and not 3.3V switched.  That's probably the reason for our different sensor timings.  Switched 3.3V means that the power gets shut off when the Mayfly is between readings, right?

SRGDamia1, 2019-02-24:
The power to switched 3.3V is controlled by setting pin 22 high or low.  If you have the I2C jumper not on switched the sensors will always get power,  If you have it on switched, the sensors will loose power when pin 22 is set low (like in that test sketch or at sensor power off in the ModularSensors code).

If you'd had the jumper set to not-switched, then even though you might have set the Atlas/I2C sensors to have a power power pin in your ModularSensors code, the power pin wouldn't actually have been turning them off.

SRGDamia1, 2019-02-24:
Are you using anything else on I2C except the Atlas sensors?  Some I2C things can cause the whole program to hang forever if they power down - anything that has a pull-up installed connecting the Vcc to the data lines.  The Atlas sensors don't have this unless you put it in yourself, so you can cut their power without fear.

See https://github.com/EnviroDIY/ModularSensors/issues/176.

acgold, 2019-02-24:
Just using the Atlas sensors on I2C, but good to know if I have to add any other I2C sensors! Thanks!

SRGDamia1, 2019-02-24:
I'd really appreciate if you could re-run the test with the power turning off in between to compare the numbers.

acgold, 2019-02-27:
Sorry for the delay.  Here are the responses after using switched 3.3V.
Temperature:
```C++
15:11:47.235 -> I2C Warm Up Timing Test
15:11:47.235 -> Wait.....................
15:11:52.245 -> I2C device replied at address 0x66 after 0 ms, code: 2
15:11:52.245 -> Received NACK on transmit of address
15:11:52.962 -> I2C device replied at address 0x66 after 735 ms, code: 0
15:11:52.996 -> i sucessfully written
15:11:52.996 -> Result available after 4 ms: ?I,RTD,2.10
15:11:52.996 -> Moving to next character - Wait......................
15:11:58.984 -> I2C device replied at address 0x66 after 735 ms, code: 0
15:11:58.984 -> r sucessfully written
15:11:59.396 -> Result available after 394 ms: 22.105
15:11:59.396 -> Moving to next character - Wait.....................
```
EC:
```C++
15:13:33.166 -> I2C Warm Up Timing Test
15:13:33.166 -> Wait.....................
15:13:38.158 -> I2C device replied at address 0x64 after 0 ms, code: 2
15:13:38.158 -> Received NACK on transmit of address
15:13:38.876 -> I2C device replied at address 0x64 after 739 ms, code: 0
15:13:38.911 -> i sucessfully written
15:13:38.911 -> Result available after 4 ms: ?I,EC,2.13
15:13:38.911 -> Moving to next character - Wait.....................
15:13:44.648 -> I2C device replied at address 0x64 after 740 ms, code: 0
15:13:44.648 -> r sucessfully written
15:13:45.231 -> Result available after 555 ms: 2.92,2,0.00,1.000
15:13:45.231 -> Moving to next character - Wait.....................
```

DO:
```C++
15:08:28.523 -> I2C Warm Up Timing Test
15:08:28.523 -> Wait.....................
15:08:33.541 -> I2C device replied at address 0x61 after 0 ms, code: 2
15:08:33.541 -> Received NACK on transmit of address
15:08:34.262 -> I2C device replied at address 0x61 after 737 ms, code: 0
15:08:34.295 -> i sucessfully written
15:08:34.295 -> Result available after 4 ms: ?I,DO,2.13
15:08:34.295 -> Moving to next character - Wait.....................
15:08:40.033 -> I2C device replied at address 0x61 after 738 ms, code: 0
15:08:40.033 -> r sucessfully written
15:08:40.580 -> Result available after 555 ms: 9.97,109.7
15:08:40.580 -> Moving to next character - Wait.....................
15:08:46.344 -> I2C device replied at address 0x61 after 739 ms, code: 0
15:08:46.344 -> i sucessfully written
15:08:46.344 -> Result available after 4 ms: ?I,DO,2.13
15:08:46.344 -> Moving to next character - Wait.....................
15:08:52.111 -> I2C device replied at address 0x61 after 739 ms, code: 0
15:08:52.111 -> i sucessfully written
15:08:52.111 -> Result available after 2 ms: ?I,DO,2.13
15:08:52.111 -> Moving to next character - Wait.....................
```

SRGDamia1, 2019-03-06:
Put this timing into the library

closed 2019-03-06 by SRGDamia1