EnviroDIY/ModularSensors #228: Using Sodaq SARA R410M modem with modular sensors

acgold, 2019-02-17:
I purchased a Sodaq SARA R410M modem to use with the menu_a_la_carte.ino example in Modular Sensors after I was having issues with the Digi 4G LTE-M modem, and I am wondering if this modem is able to be used with Modular Sensors.  I tried using it with the built-in Ublox settings, but when I run the sketch, it says that it is an "unspecified modem" and returns -9999 as the signal strength and percentage.  It looks like the R410M has a power toggle pin (Mayfly pin 20, BEE_RTS) that has to be set LOW to turn on, but I can't find a mention of this in the Modular Sensors code and I don't know how to edit the code to include it. The default baud rate for the modem is 115200, so I tried changing that in the code but there was no change in function. I'm wondering if incorporating this power toggle pin into the existing modem functions will get it to communicate with the Mayfly. Also, I tried the modem with and without the solder jumper to the LiPo battery, and I have it working without the solder jumper using the sodaq sketch below. 

I edited some code from Sodaq (found here: https://support.sodaq.com/sodaq-one/ubee/) to try and send it AT commands, and I am able to get activity on Hologram and light up the LED on the modem. It will not respond to any AT commands in the serial monitor, however. The code I used is shown below (I added the "else" section that defines the pin numbers without mentioning a board name):

```C++
//*
// * SODAQ UBEE - Serial Passthrough
// * 
// * This sketch is tested with the UBEE with the N2 and the R4 module in combination with a SODAQ Autonomo and SODAQ Mbili
// * The UBEE will work on every board with a (x)bee pinout.
// * The Mbili does not wok with the R4 module, it can't handle the fast baudrate.
// * 
// * If you have questions about this sketch you can ask them on forum.sodaq.com
// */
#include <Arduino.h>

#define R4XX // Uncomment when you use the ublox R4XX module

#if defined(ARDUINO_SODAQ_AUTONOMO)
/* SODAQ AUTONOMO + SODAQ (uBlox SARA) UBee */
#define DEBUG_STREAM SerialUSB
#define MODEM_STREAM Serial1
#define powerPin BEE_VCC
#define enablePin BEEDTR
#define powerTogglePin BEERTS

#elif defined(ARDUINO_AVR_SODAQ_MBILI)
/* SODAQ MBILI + SODAQ (uBlox SARA) UBee */
#define DEBUG_STREAM Serial
#define MODEM_STREAM Serial1
#define powerPin 20
#define enablePin BEEDTR
#define powerTogglePin BEERTS

#else // Section added to work with Mayfly
#define DEBUG_STREAM Serial
#define MODEM_STREAM Serial1
#define powerPin -2
#define enablePin 23
#define powerTogglePin 20

#endif

#ifdef R4XX
unsigned long baud = 115200;  //start at 115200 allow the USB port to change the Baudrate
#else 
unsigned long baud = 9600;  //start at 9600 allow the USB port to change the Baudrate
#endif

void setup() 
{
#ifdef powerPin
  // Turn the nb-iot module on
  pinMode(powerPin, OUTPUT); 
  digitalWrite(powerPin, HIGH);
#endif

#ifdef enablePin
  // Set state to active
  pinMode(enablePin, OUTPUT);
  digitalWrite(enablePin, HIGH);
#endif // enablePin

#ifdef R4XX
    // The R4XX module has an on/off pin. 
    // You can toggle this pin or keep it low to switch on the module.
    pinMode(powerTogglePin, OUTPUT);
    digitalWrite(powerTogglePin, LOW);
#endif
  
  delay(100);

  // Start communication
  DEBUG_STREAM.begin(baud);
  MODEM_STREAM.begin(baud);

  delay(1000);
}

// Forward every message to the other serial
void loop() 
{
  while (DEBUG_STREAM.available())
  {
    MODEM_STREAM.write(DEBUG_STREAM.read());
  }

  while (MODEM_STREAM.available())
  {     
    DEBUG_STREAM.write(MODEM_STREAM.read());
  }
}
```
Thanks in advance for any help or advice!

SRGDamia1, 2019-02-17:
The code for the ublox should work for the uBee.  That wake and sleep function is what you need to wake out using the sleep request on pin 20.  Status is still on pin 19 and power on 23.  I have a heavily annotated uBee scheme with some more notes.

The processor on the Mayfly is identical to the one on the mbili, though, so pay attention to that note about the baud rate problem. You should slow the SARA down to get clean communication with the Mayfly.

SRGDamia1, 2019-02-17:
[image]

SRGDamia1, 2019-02-17:
Ironically, someone else was just asking about the Mayfly at 115200 here: https://www.envirodiy.org/topic/esp8266/

SRGDamia1, 2019-02-17:
Sorry, I'll leave this open until we are sure if it works. I don't have one to test.

acgold, 2019-02-18:
@SRGDamia1 - thanks for the help! I'll start trying to modify the code to those pins for the sleep/wake functions.
 
I couldn't get the Mayfly to talk with the modem. I have an xbee-explorer type board (https://www.waveshare.com/xbee-usb-adapter.htm) that I was able to get working with U-blox's m-center program using 115200 baud rate, but apparently the SARA R410M baud rate can't be changed via USB? I tried multiple times changing the baud rate using `AT+IPR=9600` (which was also referenced in the commands manual), but I could not connect to it using the 9600 baud rate with the xbee-explorer board or the Mayfly.  It was still at the 115200 baud rate.  
[image]

Do you know of a way to get the `AT+IPR=9600` to the modem via the Mayfly, or is there another board that can support the 115200 baud rate that would work (I'm wondering about a Leonardo or Uno + Xbee shield: https://www.sparkfun.com/products/12847)? Thanks!

SRGDamia1, 2019-02-18:
Ok.  I'm reading more on the issue and here are my current thoughts:

 - You're not trying to multiplex (send AT commands to the u-blox from two different ports at the same time) so you don't need to worry about that table.

 - When you're using the USB XBee adapter, you're not actually talking to the u-blox via USB.  You're talking to an adapter chip via USB which is talking to the u-blox via UART.  The Mayfly also communicates with the u-blox vis UART.  You *can* talk to the the u-blox directly via USB, but you'd need to connect your computer's USB D+/D- to the direct USB pins (Sodaq breaks them out at J1, see the schema).

 - The baud rate of the SARA R410M **can** be changed to 9600, 19200, 38400, 57600, or 115200, but __the setting change doesn't persist after the module looses power__.  See appendix B.1 of the AT commands manual.  Also this question on the ublox forum: https://portal.u-blox.com/s/question/0D52p00008HKDqACAX/sara-r4-storing-of-at-command-settings-in-profiles-not-possible.  Every time you take a reading, you're cutting the power to the module.  This means your wake function must have a begin Serial1 at 115200, send the AT+IPR=9600 command, and then issue another begin for Serial1 at 9600.  A bit annoying, but should work.  The outgoing communication from the Mayfly at the fast baud rate is much less likely to be a problem than the reception, so the SARA is very likely to understand the command, even if the Mayfly can't get its response.

 - Both the UNO and the Leonardo are 16MHz boards (2x the speed of the Mayfly) and should do a much better job at communication at 115200 baud __but__ neither the UNO nor the Leonardo have enough memory to support the ModularSensors library.

 - If you're looking to buy another board to replace your Mayfly, I'd look into something like the Sodaq autonomo or one of their other SAMD21 based boards.  You could go straight to the SARA boards which have the SAMD21 main processor and the u-blox chip build into the same board.  The "AFF" SARA boards even have built in solar charging.  You could also go with an Arduino Zero, Adafruit Feather M0 or one of their clones.  The SAMD21 processor is 100% supported by ModularSensors and tested on the Feather M0.

SRGDamia1, 2019-02-18:
Oh, and I've got my Digi LTE-M board working, if you'd like to troubleshoot that.

aufdenkampe, 2019-02-18:
@acgold, thanks for testing the Sodaq SARA R410M modem and starting this Github issue to report your progress. I was just thinking of getting one, so this is all great.

@SRGDamia1, thanks for your detailed responses! Any new developments on the Digi XBee3 LTE-M board, beyond what we've recorded 17 days ago in https://github.com/EnviroDIY/ModularSensors/issues/162#issuecomment-459879699?

SRGDamia1, 2019-02-18:
No, no new developments.  It works, so I wasn't planning on doing anything more unless someone puts one in the wild and has problems.

SRGDamia1, 2019-02-18:
@acgold you can try something like this:

```cpp
bool modemWakeFxn(void)
{
    if (modemVccPin >= 0)  // Turns on when power is applied
    {
        modemSerial.begin(115200);
        delay(4500);  // Ugh.. Don't like this wait, but takes ~4.5sec before UART becomes active
        // would prefer to use at least a tinyModem->testAT(5000);, but worried this would not return any earlier because the Mayfly might not understand the OK's coming back.
        tinyModem->setBaud(9600);  // Don't check for success, just hope
        modemSerial.begin(9600);
        return true;
    }
    else if(modemSleepRqPin >= 0)
    {
        digitalWrite(modemSleepRqPin, LOW);
        digitalWrite(redLED, HIGH);
        delay(200); // 0.15-3.2s pulse for wake on SARA R4/N4
        digitalWrite(modemSleepRqPin, HIGH);
        digitalWrite(redLED, LOW);
        return true;
    }
    else
    {
        return true;
    }
}
```

acgold, 2019-02-18:
@SRGDamia1 thanks, that explains the modem behavior that I was seeing.  My goal is to get this modem up and running smoothly on the Mayfly rather than switching boards - I was thinking that I might have to buy another board to switch the baud rate and make it permanent.

I'll definitely give this code a try.  Are the pin designations in this code using the recommendations that you gave above: modemVccPin = 23, modemSleepRqPin = 20, Status = 19?

acgold, 2019-02-18:
@aufdenkampe - happy to help!

@SRGDamia1 I don't have the Digi Lte-M modem with me today, but the last time I tested them, I was having issues with power.  It would work once and then would stop responding to the Mayfly.

acgold, 2019-02-18:
Sorry for accidentally closing this issue - still learning how to use GitHub!  So far, I've not been able to get the Mayfly to communicate with the modem.  I've tried implementing the edited Wake function with different pin assignments for power and sleep, and I've also tried a simpler script from Sodaq that I edited to include sending the `AT+IPR=9600` at the 115200 baud rate. It doesn't seem like the modem is able to understand the Mayfly.

```C++
#include <Arduino.h>

#define R4XX // Uncomment when you use the ublox R4XX module

#if defined(ARDUINO_SODAQ_AUTONOMO)
/* SODAQ AUTONOMO + SODAQ (uBlox SARA) UBee */
#define DEBUG_STREAM SerialUSB
#define MODEM_STREAM Serial1
#define powerPin BEE_VCC
#define enablePin BEEDTR
#define powerTogglePin BEERTS

#elif defined(ARDUINO_AVR_SODAQ_MBILI)
/* SODAQ MBILI + SODAQ (uBlox SARA) UBee */
#define DEBUG_STREAM Serial
#define MODEM_STREAM Serial1
#define powerPin 20
#define enablePin BEEDTR
#define powerTogglePin BEERTS

#else // Added to work with Mayfly
#define DEBUG_STREAM Serial
#define MODEM_STREAM Serial1
#define powerPin 23
#define enablePin 19
#define powerTogglePin 20

#endif

void setup() 
{
#ifdef powerPin
  // Turn the nb-iot module on
  pinMode(powerPin, OUTPUT); 
  digitalWrite(powerPin, HIGH);
#endif

#ifdef enablePin
  // Set state to active
  pinMode(enablePin, OUTPUT);
  digitalWrite(enablePin, HIGH);
#endif // enablePin

#ifdef R4XX
    // The R4XX module has an on/off pin. 
    // You can toggle this pin or keep it low to switch on the module.
    pinMode(powerTogglePin, OUTPUT);
    digitalWrite(powerTogglePin, LOW);
#endif
  
  delay(5000);

  // Start communication
  MODEM_STREAM.begin(115200);
  delay(5000);
  MODEM_STREAM.write("AT+IPR=9600"); //Also tried "AT+IPR=9600\r"
  MODEM_STREAM.begin(9600);
  DEBUG_STREAM.begin(9600);
}

// Forward every message to the other serial
void loop() 
{
  while (DEBUG_STREAM.available())
  {
    MODEM_STREAM.write(DEBUG_STREAM.read());
  }

  while (MODEM_STREAM.available())
  {     
    DEBUG_STREAM.write(MODEM_STREAM.read());
  }
}
```

SRGDamia1, 2019-02-18:
Yes, there are problems with the Digi, but it does work!  You absolutely need to change solder jumper 13 and use a battery.  And, even then, it just stops responding sometimes.  But I've found that I have pretty good luck in keeping it responsive if I put it into bypass mode and talk to it that way.  The a la carte example has the code in it to do this.

Those are the right pins for the code I posted above.  On the uBee, there are essentially 3 pins that can control power and sleep levels of the board:
    - **Bee pin 1/VCC_3.3** supplies the power to *all* of the IC's on the board, *except* in cases where there is a separate battery plugged in to the board, in which case bee pin 1 only controls the level shifters.  The Mayfly cannot control this pin, but the Sodaq Autonomo (and I think some versions of the mbili) can, so you'll see references in the Sodaq example code to this as the power pin.  If you were able to control it, you'd have to hold this pin high to keep things powered.
    - **Bee pin 9/DTR** (and Mayfly pin 23) is the "ON/OFF" pin.  This is connected to the switch (IC5) which controls whether the on board u-blox (and only that) has power.  This also has to be held high to power the u-blox.  In the Sodaq examples, they call this the "enablePin."  In the modular sensors code example I gave you, I treat this as the power pin.  If you were to run ModularSensors on an Autonomo with a uBee you'd put the VCC_BEE/Pin 1 control pin as the power pin and you'd have to control this pin as part of the wake function.  But, again, since the Mayfly can't control bee pin 1, we just call this the power pin.
    - Bee pin 16/RTS (and Mayfly pin 20) is the "PWR_ON" pin for the SARA R410 and U201.  You can pulse it low to turn the chip on and off or simply hold it low to keep it on.  In the Sodaq examples they call this the "voltagePin."  In my examples, I'm calling it the "modemSleepRqPin."  The Sodaq examples for the uBee sometimes exclude this pin because the u-blox N211, which is one of the options for the uBee doesn't have a pin to control deep sleep like this.

SRGDamia1, 2019-02-18:
```cpp
#include <Arduino.h>

const int8_t modemVccPin = 23;  // power pin
const int8_t modemSleepRqPin = 20;  // power toggle pin

void setup() 
{
  // Start the main serial port
  Serial.begin(9600);

  // Set pin modes
  pinMode(modemVccPin, OUTPUT); 
  pinMode(modemSleepRqPin, OUTPUT); 

  // power the u-blox (not-powered to power-off mode)
  digitalWrite(modemSleepRqPin, HIGH);
  digitalWrite(modemVccPin, HIGH);
  // brief wait to ensure rising edge of VCC reached
  // NB: This is probably way too long, but the timing isn't specified
  // See SARA-R4/N4 series - System Integration Manual 1.6.1
  delay(250);

  // Bring the u-blox from power-off mode to active
  digitalWrite(modemSleepRqPin, LOW);
  delay(250) Low time to switch on the module from power off mode = 0.15-3.2sec
  digitalWrite(modemSleepRqPin, HIGH);

  // Wait for serial ports on u-blox to become active  
  delay(5000); (~4.5s)

  // Start communication
  Serial1.begin(115200);
  // try to get the u-blox to drop speed
  // use PRINT not WRITE to get ascii out
  // println will add the necessary CR and LF (\r\n)
  Serial1.println("AT+IPR=9600");  
  Serial1.begin(9600);
}

// Forward every message to the other serial
void loop() 
{
  while (Serial.available())
  {
    Serial1.write(Serial.read());
  }

  while (Serial1.available())
  {     
    Serial.write(Serial1.read());
  }
}
```

SRGDamia1, 2019-02-18:
Can you verify that when plugged into your XBee explorer, sending the command AT+IPR=9600 does change the baud rate?

That is, put your bee on the bee explorer and connect it to your computer.  Open a terminal at 115200 and send the command "AT+IPR=9600" and listen for "OK." Wait half a second and then just send "AT" and see what happens.  It should _not_ respond to that AT because it's at the wrong baud rate.  Reopen a new terminal at 9600 and send "AT" and make sure you get an "OK." back at that baud rate.

acgold, 2019-02-18:
I'll definitely take another look at the Digi modem.  I was using it with the jumper and battery, and think I was trying to use it in bypass mode, but I'm not entirely sure.

Thanks for the test code. I'm still not getting any response using any of the test code.  The modem responds `OK` at 115200 after the `AT+IPR=9600`, and there is no response to further `AT` commands. I think the m-center software turns off the modem when I change the baud rate (modem LED turns off). When I'm communicating at 115200 and after I've sent the `AT+IPR=9600` command, I have to click "Disconnect", "Set port", change the baud rate, and click "Connect".  It seems like it does change the baud rate, but after it loses power, it goes back to 115200.

SRGDamia1, 2019-02-18:
Ok,  If you have to disconnect and change the baud rate, the baud rate sounds like it was really changing.  Can you try testing with just a basic serial port monitor/terminal emulator instead of m-center?  (tera term, mini-term, arduino serial port monitor, platformio serial port monitor, etc)

Do you have another board you can use to spy the communication between the uBee and your Mayfly?  Something like a mega would be ideal.  With the uBee plugged into the Mayfly, plug the Rx of the mega's Serial1 into the Tx of Mayfly's Serial1 (on the bottom left of the Mayfly) and the Rx of the mega's Serial2 into the Rx of the Mayfly's Serial1.  Then the mega can run a simple program like that just prints everything it hears on Serial1 or Serial2 back to Serial.  What does this show you is happening between the Mayfly and the uBee?

SRGDamia1, 2019-02-18:
Is the LED coming on?  The LED is attached to GPIO1 - by default it's not configured to do anything.  You can set it to either indicate network status or to indicate the module status and I think once set it will maintain that setting after power off.

Try setting it to indicate module status:
GPIO1 has gpio_id = 16
gpio_mode = 2 for network status indication, 10 for module operating status indication
AT+UGPIOC=16,10

acgold, 2019-02-19:
I changed the LED to indicate the module operating status, and it will turn on when connected to m-center via the xbee explorer and responding to AT commands. When I try to connect with any of the other serial port monitors/emulators, the light will not turn on and I don't get a response to any AT commands. Not sure if it's a power problem like the RTS pin is pulled HIGH by default and m-center adjusts that somehow...

Unfortunately, I don't have another board to spy on the communication b/t the Mayfly and the modem.  I tried using another Mayfly just to see if it would work, but it was a bunch of random characters because of the 115200 baud rate, as would be expected. 

@SRGDamia1  - Thank you so much for your help! At least right now, it looks like the modem communicates at too high of a baud rate for the Mayfly (but we're not positive b/c I can't spy on their communication right now). I wonder if storing the `AT+IPR=9600` command is something U-blox could modify with future firmware upgrades?

SRGDamia1, 2019-02-19:
So when you put it on the xbee xpolrer, it can talk to M-center but nothing else? That's.. Unexpected. Very unexpected. You can use the same xbee explorer to talk to other bees, right? The SARA R410 doesn't have or support hardware flow control (DTR/CTS/RTS) so M-center would be letting those float and they'd probably drift high.  Sodaq is only using those pins on the Bee for power control because they're the ones that most sodaq boards (and the mbili based Mayfly) have connected. 

OK. So more brainstorming.

If you have the board connected to M center and the light is on, then close the m center program, does the light immediately turn off? If you open another terminal to that port, is that when the light goes off?

Do you have a schema for the exact xbee explorer you have? Are the xbee DTR/CTS/RTS pins of the explorer connected to the same of the USB to UART chip?

Are you using a Mayfly with the Bee vcc connected to the battery (SJ13) or connected to the 3.3V regulator (unmodified)? Have you tried both configurations? Were the results the same with both?

Change the sketch to simply send AT and print the response instead of the pass through. (Serial1.println("AT"); Serial.println(Serial1.readString()); several times.) Same results?

SRGDamia1, 2019-02-19:
Try running a sketch like this that fiddles with the pin settings to see if /when the light comes on. Uncomment and run it all.  https://github.com/EnviroDIY/ModularSensors/blob/8a04bacd82b958b15adaa9facbbe9765fe36851f/sensor_tests/uBeeTest/uBeeTest.ino.

While you're running that sketch, plug the rx pin of your xbee explorer into the Serial1 rx of the Mayfly and use a port emulator to look for responses from the uBee that the Mayfly can't understand. You can also do this on the Serial1 TX. This will only give half on the conversation at all time, but it is good to see.  All the xbee explorer is doing in this test is providing a uart to usb snoop.

acgold, 2019-02-19:
I'm using an clone xbee explorer-type board (https://www.waveshare.com/xbee-usb-adapter.htm), and it worked with the Digi 4G Lte-M modem.  The green LED on the modem indicating that the module is on turns off when I close m-center. There is a red LED that indicates PWR on the explorer board, and it turns off if I try to connect with another program.

I think that the CTS/RTS/DTR pins are separate, but not positive. 

I've tried both Mayfly SJ13 configurations with the same results.  It's definitely showing activity on hologram and powering on the module while on the Mayfly, but not sending any data or talking to the Mayfly. 

Same results with the printing `AT` and reading response.  I get one `?` and then nothing for any of the other times.  I added a 2 second delay after the print and read:
```C++
void loop() 
{
  Serial1.println("AT"); 
  Serial.println(Serial1.readString()); 
  delay(2000);
}
```

I'll try to do the spy test at some point this week.  I have messed around with the pin designations previously, and the only time I would get activity of hologram was when RTS was set LOW.

acgold, 2019-02-20:
I set the modem baud rate as 115200 for the uBeeTest sketch, and I uncommented all of the code sections in the setup loop except for ` tinyModem->poweroff();` which was commented out throughout each section already. The modem LED came on indicating that the module was on a few times.

The LED came on and the printed characters changed from `AT`  to `AU?`, `j5`, and `AUXHO?HAT` after:

19 and 20 already HIGH, and setting 23 HIGH.  It turned off after setting 23 low again:
```C++
18:58:06.777 -> Pin 20 HIGH: 0
18:58:06.777 -> setting 23 high
```
20 and 23 HIGH w 19 as input and 1 sec low pulse on 20:
```C++
18:59:13.568 -> Pin 20 & 23 HIGH: 0
18:59:13.568 -> Pin 19 state: 0
18:59:13.568 -> outputting a 1 sec low pulse on 20
```
After a 60 µs low pulse on 20:
```C++
18:59:36.899 -> After pin 20 low pulse: 0
18:59:36.899 -> Pin 19 state: 1
18:59:36.899 -> outputting a 60µs low pulse on 20
```
After this next code, it turned off but had numbers respond instead of AT or other characters:
```C++
8:59:59.149 -> After pin 20 low pulse: 0
18:59:59.149 -> Pin 19 state: 1
18:59:59.149 -> outputting a 1 sec low pulse on 20
19:00:00.253 -> 150003
19:00:00.461 -> 150224
```
Turned on after this:
```C++
19:00:23.937 -> Pin 20 & 23 HIGH: 0
19:00:23.937 -> Pin 19 state: 1
19:00:23.937 -> outputting a 1 sec low pulse on 20
```
Stayed on after this:
```C++
19:00:47.316 -> After pin 20 low pulse: 0
19:00:47.316 -> Pin 19 state: 1
19:00:47.316 -> outputting a 60µs low pulse on 20
```
Final command and it turned off but responded with numbers:
```C++
19:01:09.548 -> After pin 20 low pulse: 0
19:01:09.548 -> Pin 19 state: 1
19:01:09.548 -> outputting a 1 sec low pulse on 20
19:01:10.646 -> 220307
```

I tried to spy on the communication using the xbee explorer board, but I was never able to get it to work...

I think I'm going to buy a Sodaq Autonomo and try to run Modular Sensors on it with this modem.  I verified that the Maxbotix ultrasonic sensor that I have will work with the Autonomo, and all I need in addition to that is I2C to run my three Atlas sensors. Running Modular Sensors on the Autonomo would basically just be changing pin numbers, right?

SRGDamia1, 2019-02-20:
Yes, switching to an autonomo should just be pin number changes.

I'm still going to harass you to get this uBee working though. We should be able to get it to work!  I don't like giving up. 

It is responding to the low RTS pulse, which is what I expected. I don't quite get why it's not working in the modular sensors program. I'll send you a new test sketch tomorrow morning.

acgold, 2019-02-20:
@SRGDamia1 - **I just got it to work!!!** Here's the code.  I just added `serial.end()` after sending `IP+IPR=9600` and before starting the other `serial.begin(9600)`. It's now responding to `AT` with `OK` at 9600 baud.  I tested with `AT+CSQ` and its responding with `+CSQ: 8,99`!

I'll test using modular sensors!

```C++
#include <Arduino.h>

const int8_t modemVccPin = 23;  // power pin
const int8_t modemSleepRqPin = 20;  // power toggle pin

void setup() 
{
  // Start the main serial port
  Serial.begin(9600);

  // Set pin modes
  pinMode(modemVccPin, OUTPUT); 
  pinMode(modemSleepRqPin, OUTPUT); 

  // power the u-blox (not-powered to power-off mode)
  digitalWrite(modemSleepRqPin, HIGH);
  digitalWrite(modemVccPin, HIGH);
  // brief wait to ensure rising edge of VCC reached
  // NB: This is probably way too long, but the timing isn't specified
  // See SARA-R4/N4 series - System Integration Manual 1.6.1
  delay(250);

  // Bring the u-blox from power-off mode to active
  digitalWrite(modemSleepRqPin, LOW);
  delay(250);  // Low time to switch on the module from power off mode = 0.15-3.2sec
  digitalWrite(modemSleepRqPin, HIGH);

  // Wait for serial ports on u-blox to become active  
  delay(5000); // ~4.5s

  // Start communication
  Serial1.begin(115200);
  // try to get the u-blox to drop speed
  // use PRINT not WRITE to get ascii out
  // println will add the necessary CR and LF (\r\n)
  Serial1.println("AT+IPR=9600");  
  Serial1.end();
  Serial1.begin(9600);
}

// Forward every message to the other serial
void loop() 
{
  Serial1.println("AT+CSQ"); 
  Serial.println(Serial1.readString()); 
  delay(2000);
}
```

acgold, 2019-02-20:
I'm not able to get it working on Modular Sensors yet.  Since the Power pin and Sleep pin need to be set LOW to be on (opposite of other modems?), would functions like modem.PowerUp() need to be reversed?

SRGDamia1, 2019-02-20:
Modem power up only sets the power pin high, and it does need to be high.  The sleep pin is set in the sleep fxn.

SRGDamia1, 2019-02-20:
And I was too slow, but the new test I made for you is here:  https://github.com/EnviroDIY/ModularSensors/blob/atlas/sensor_tests/uBeeTest/uBeeTest.ino  I'd still be interested in your output from it if you have a chance.

acgold, 2019-02-20:
I'll try to run that new uBeeTest soon and let you know the results.  

Good news, though, I got the modem up and running with Modular Sensors! I edited the Wake function for the Ublox section and added a `modem.wake()` function below in the Setup loop before the RTC sync to NIST. Here's the Wake function that I edited below.  I tried putting the wake up pulse on 20 in only the first code chunk `if modemVccpin >= 0`, but it didn't work.  I tried only the bottom (else if chunk), and it worked, but spit out a bunch of "operation not allowed" messages and would not sync when syncing the RTC with NIST.  I put it in both the "if" and "if else" chunks and it works great.  Probably not the most streamlined way to do it, though...
```C++
bool modemWakeFxn(void)
{
    if (modemVccPin >= 0)  // Turns on when power is applied
    {
        delay(250);
        digitalWrite(modemSleepRqPin, LOW);
        digitalWrite(redLED, HIGH);
        delay(250); // 0.15-3.2s pulse for wake on SARA R4/N4
        digitalWrite(modemSleepRqPin, HIGH);
        digitalWrite(redLED, LOW);
        delay(5000);  // Ugh.. Don't like this wait, but takes ~4.5sec before UART becomes active
        // would prefer to use at least a tinyModem->testAT(5000);, but worried this would not return any earlier because the Mayfly might not understand the OK's coming back.
        modemSerial.begin(115200);
        tinyModem->setBaud(9600);  // Don't check for success, just hope
        modemSerial.end();
        modemSerial.begin(9600);
        return true;
    }
    else if(modemSleepRqPin >= 0)
    {
        delay(5000);  // Ugh.. Don't like this wait, but takes ~4.5sec before UART becomes active
        // would prefer to use at least a tinyModem->testAT(5000);, but worried this would not return any earlier because the Mayfly might not understand the OK's coming back.
        modemSerial.begin(115200);
        tinyModem->setBaud(9600);  // Don't check for success, just hope
        modemSerial.end();
        modemSerial.begin(9600);
        
        return true;
    }
    else
    {
        return true;
    }
}
```
Here is the code around the `modem.wake()` function. I added this because I don't that this code that was above in the sketch (`modemSerial.begin(ModemBaud);`) would work for the RTC sync.  It looks like the Xbees and another modem required some different setup as well, so maybe the R4 could have its own `if()` loop for this?
```C++
// Begin the logger
    // Note:  Please change these battery voltages to match your battery
    // Check that the battery is OK before powering the modem
    if (getBatteryVoltage()> 3.7)
    {
        modem.modemPowerUp();
        modem.wake(); // Added
    }
    // At lowest battery level, skip sensor set-up
    if (getBatteryVoltage()< 3.4)
    {
        dataLogger.begin(true);
    }
    else  // set up file and sensors
    {
        dataLogger.begin();
    }

```
It does about 2 minutes to send 4 variables worth of data to data.envirodiy.org because it's waiting for a response from the website - is that normal? I tried sending it to a Thingspeak channel as well, and it worked once and then responded with "#operation not allowed" the subsequent times. Sending to EnviroDIY has worked every time.

SRGDamia1, 2019-02-20:
It's not running both the "if" and the "else if", only one of them.  It was something else you changed that made it start working, not adding code in both places.

The note "turns on when power applied" was, unfortunately, not true for the R4 series.  It is for essentially every other u-blox.  Go figure.

I think, looking at what worked, that the reason it hadn't been working before was that very first "delay(250)" before the pulse.  That's that time that I'd noted earlier is probably too long but is unclear in the docs.  It's the "warm up" wait - that is, the time between when the power is turned on before the modem (or any other sensor) is ready to receive a wake up call.  In every other sensor, after the powerUp() function is called there's a wait before the wake() function is called (either explicity with waitForWarmup() or implicitly in the timing loops for the variableArray).  I forgot about that wait in the modem.  And the R410 is apparently the first modem anyone's tested where it mattered.  I fixed that in the loggerModem cpp so you can now remove that delay.

With the other things fixed, you should not need that extra modem.wake(); in the setup.  The modem setup function and the NIST sync functions will already call if on their own if needed.

I added a new macro variable for the R410M in the menu with your cleaned wake function.  Can you please test it?

acgold, 2019-02-20:
I'm not able to compile the new menu_a_la_carte.ino.  It's saying that `waitForStability()` in the Wake function was not declared in this scope.

SRGDamia1, 2019-02-20:
Oh dear.  Yes, you're right.  Replace it with a delay(4500);  (or up to 5000 if that doesn't work).

acgold, 2019-02-20:
I tried a delay of 4500, 4700, and 5000. The modem seems to be turning on at the beginning, but it turns off when it tries to sync with the NIST. It will record the cell strength and percentage to the SD card, but it's not sending data to the data portal 
```
15:12:49.657 -> Now running menu_a_la_carte.ino on Logger XXXXX
15:12:49.657 -> 
15:12:49.657 -> Using ModularSensors Library version 0.20.1
15:12:50.267 -> Logger timezone is set to UTC -5
15:12:50.267 -> RTC timezone is set to UTC
15:12:50.267 -> A modem has been tied to this logger!
15:12:50.267 -> Setting I2C Pins to INPUT_PULLUP
15:12:50.302 -> Beginning wire (I2C)
15:12:50.302 -> Beginning real time clock
15:12:50.405 -> Current RTC time is: 2019-02-20T15:12:50-05:00
15:12:50.439 -> Setting up logger XXXXX to record at 2 minute intervals.
15:12:50.439 -> This logger has a variable array with 4 variables, of which 4 come from  3 sensors and 0 are calculated.
15:12:50.439 -> Setting up sensors...
15:12:55.869 -> [6197] ### Unhandled: ⸮H⸮
15:12:56.110 -> [6438] ### Modem: u-blox SARA-R410M-02B
15:12:56.315 -> Successfully connected to SD Card with card/slave select on pin 12
15:12:56.383 -> Data will be saved as XXXXX_2019-02-20.csv
15:12:56.383 -> Push button on pin 21 at any time to enter sensor testing mode.
15:12:56.383 -> Logger setup finished!
15:12:56.383 -> ------------------------------------------
15:12:56.383 -> 
15:12:56.383 -> Attempting to synchronize RTC with NIST
15:12:56.383 -> This may take up to two minutes!
15:12:57.443 -> [7761] ### Unhandled: 
```

and 
```
15:16:00.191 -> ------------------------------------------
15:16:05.841 -> [134086] ### Unhandled: ⸮⸮
15:16:20.985 -> Successfully connected to SD Card with card/slave select on pin 12
15:16:21.020 -> 
15:16:21.020 ->  \/---- Line Saved to SD Card ----\/
15:16:21.020 -> 2019-02-20 15:16:00,4.912,20.50,-61,84
15:16:21.020 -> 
15:16:21.020 -> 
15:17:11.380 -> ------------------------------------------
```

SRGDamia1, 2019-02-20:
Hmm... same results at all times?  Then 4500 is enough.  I'm hunting the turn off at the sync.

SRGDamia1, 2019-02-20:
Can you turn on the streamdebugger code?

SRGDamia1, 2019-02-20:
Oh, and open LoggerModem.h in the library and uncomment the line ```#define MODEM_DEBUGGING_SERIAL_OUTPUT Serial```

acgold, 2019-02-20:
When I uncomment the streamdebugger code, 

```C++
// Use this to create a modem if you want to spy on modem communication through
// a secondary Arduino stream.  Make sure you install the StreamDebugger library!
// https://github.com/vshymanskyy/StreamDebugger
#include <StreamDebugger.h>
StreamDebugger modemDebugger(modemSerial, Serial);
TinyGsm *tinyModem = new TinyGsm(modemDebugger);
```
I get this error:
`error: redefinition of 'TinyGsm* tinyModem'`

acgold, 2019-02-20:
I just made it `TinyGSM *tinyModem2`. Here's what it returns with StreamDebugger and the LoggerModem debugging:

```C++
15:55:18.536 -> Now running menu_a_la_carte.ino on Logger XXXXX
15:55:18.536 -> 
15:55:18.536 -> Using ModularSensors Library version 0.20.1
15:55:19.156 -> Logger timezone is set to UTC -5
15:55:19.156 -> RTC timezone is set to UTC
15:55:19.156 -> A modem has been tied to this logger!
15:55:19.156 -> Powering unspecified modem with pin 23
15:55:19.156 -> Setting I2C Pins to INPUT_PULLUP
15:55:19.156 -> Beginning wire (I2C)
15:55:19.156 -> Beginning real time clock
15:55:19.260 -> Current RTC time is: 2019-02-20T15:55:19-05:00
15:55:19.260 -> Setting up logger XXXXX to record at 2 minute intervals.
15:55:19.294 -> This logger has a variable array with 4 variables, of which 4 come from  3 sensors and 0 are calculated.
15:55:19.294 -> Setting up sensors...
15:55:19.328 -> Setting up the modem ...
15:55:19.328 -> Running given modem wake function ...
15:55:24.516 -> Running modem's begin function ...
15:55:24.755 -> [6201] ### Unhandled: #⸮⸮
15:55:24.961 -> [6443] ### Modem: u-blox SARA-R410M-02B
15:55:25.169 -> ... Complete!  It's a u-blox SARA-R410M-02B
15:55:25.169 -> Resetting warm-up and disconnect timing for a u-blox SARA-R4
15:55:25.169 -> u-blox SARA-R410M-02B warms up in 250 ms, indicates status in 0 ms, is responsive to AT commands in less than 4500 ms, and takes up to 15000 ms to close connections and shut down.
15:55:25.169 -> Running given modem sleep function ...
15:55:25.237 -> Successfully connected to SD Card with card/slave select on pin 12
15:55:25.271 -> Data will be saved as XXXXX_2019-02-20.csv
15:55:25.271 -> Push button on pin 21 at any time to enter sensor testing mode.
15:55:25.271 -> Skipping u-blox SARA-R410M-02B in sensor power down!
15:55:25.271 -> Logger setup finished!
15:55:25.271 -> ------------------------------------------
15:55:25.271 -> 
15:55:25.271 -> Attempting to synchronize RTC with NIST
15:55:25.271 -> This may take up to two minutes!
15:55:25.271 -> Waking u-blox SARA-R410M-02B
15:55:25.271 -> u-blox SARA-R410M-02B was already on!  Will not run wake function.
15:55:25.271 -> u-blox SARA-R410M-02B should be awake.
15:55:25.271 -> 
15:55:25.271 -> Waiting for u-blox SARA-R410M-02B to respond to AT commands...
15:55:25.271 -> ... AT OK after 23 milliseconds!
15:55:25.271 ->                                                           //Looks like LED turns off here
15:55:25.305 -> Waiting up to 120 seconds for cellular network registration...
15:55:27.631 -> [9086] ### Unhandled: 
15:57:27.200 -> ...GPRS connection failed.
15:57:27.200 -> Turning u-blox SARA-R410M-02B off.
15:57:27.200 -> u-blox SARA-R410M-02B appears to have already been off.  Will not run sleep function.
15:57:27.200 -> Waiting up to 15000 milliseconds for graceful shutdown...
15:57:27.200 -> ... shutdown complete after 12 ms.
15:57:27.200 -> Turning off power to u-blox SARA-R410M-02B with pin 23
15:57:59.901 -> ------------------------------------------
15:57:59.901 -> Powering u-blox SARA-R410M-02B with pin 23
15:57:59.901 -> Skipping u-blox SARA-R410M-02B in sensor power up!
15:58:00.141 -> Waking u-blox SARA-R410M-02B
15:58:00.141 -> Running wake function for u-blox SARA-R410M-02B
15:58:05.342 -> u-blox SARA-R410M-02B should be awake.
15:58:05.547 -> [134178] ### Unhandled: #
15:58:05.685 -> Starting measurement on u-blox SARA-R410M-02B
15:58:20.617 -> Getting signal quality:
15:58:20.691 -> Raw signal quality: 24
15:58:20.691 -> RSSI: -65
15:58:20.691 -> Percent signal strength: 78
15:58:20.691 -> Skipping u-blox SARA-R410M-02B in sensor power down!
15:58:20.691 -> Successfully connected to SD Card with card/slave select on pin 12
15:58:20.744 -> 
15:58:20.744 ->  \/---- Line Saved to SD Card ----\/
15:58:20.744 -> 2019-02-20 15:58:00,4.897,19.50,-65,78
15:58:20.744 -> 
15:58:20.744 -> 
15:58:20.744 -> 
15:58:20.744 -> Waiting for u-blox SARA-R410M-02B to respond to AT commands...
15:58:20.794 -> ... AT OK after 23 milliseconds!
15:58:20.794 ->                                                   
15:58:20.794 -> Waiting up to 50 seconds for cellular network registration...
15:59:11.031 -> ...GPRS connection failed.
15:59:11.031 -> Turning u-blox SARA-R410M-02B off.
15:59:11.031 -> Running given sleep function for u-blox SARA-R410M-02B
15:59:11.102 -> Waiting up to 15000 milliseconds for graceful shutdown...
15:59:11.155 -> ... shutdown complete after 92 ms.
15:59:11.155 -> Turning off power to u-blox SARA-R410M-02B with pin 23
15:59:11.155 -> ------------------------------------------
```

SRGDamia1, 2019-02-20:
Sorry, to turn on the debugger you have to have the line ```TinyGsm *tinyModem = new TinyGsm(modemDebugger);```  Instead of making it modem2, comment out the line ```TinyGsm *tinyModem = new TinyGsm(modemSerial);```

SRGDamia1, 2019-02-20:
It's not getting the NIST sync at the beginning because the modem takes a long time to go to sleep (~15s).  So the setup function woke it up and then put it back to sleep, but then the NIST function checks if it's awake and sees that it still is, not realizing that it's on it's way to sleep.  I'll have to think about the best way to fix it.

The not sending data puzzles me.  It looks like it's getting great signal from the csq response, but the gprs is failing.  Did you, by chance, forget to put your APN into the program?

acgold, 2019-02-21:
I did the debugging and a summary of the output is below.  I did almost forget to put in the APN, but I do have "hologram" in the APN spot

```C++
19:28:23.577 -> Now running menu_a_la_carte.ino on Logger XXXXX
19:28:23.577 -> 
19:28:23.577 -> Using ModularSensors Library version 0.20.1
19:28:24.200 -> Logger timezone is set to UTC -5
19:28:24.200 -> RTC timezone is set to UTC
19:28:24.200 -> A modem has been tied to this logger!
19:28:24.200 -> Powering unspecified modem with pin 23
19:28:24.200 -> Setting I2C Pins to INPUT_PULLUP
19:28:24.200 -> Beginning wire (I2C)
19:28:24.200 -> Beginning real time clock
19:28:24.304 -> Current RTC time is: 2019-02-20T19:28:24-05:00
19:28:24.304 -> Setting up logger XXXXX to record at 2 minute intervals.
19:28:24.337 -> This logger has a variable array with 4 variables, of which 4 come from  3 sensors and 0 are calculated.
19:28:24.337 -> Setting up sensors...
19:28:24.372 -> Setting up the modem ...
19:28:24.372 -> Running given modem wake function ...
19:28:29.575 -> AT+IPR=9600
19:28:29.575 -> Running modem's begin function ...
19:28:29.575 -> AT
19:28:29.575 -> 8⸮⸮[6201] ### Unhandled: 8⸮⸮
19:28:29.886 -> AT
19:28:29.886 -> AT

19:28:29.886 -> OK
19:28:29.886 -> ATE0
19:28:29.920 -> ATE0

19:28:29.920 -> OK
19:28:29.920 -> AT+CMEE=2
19:28:29.954 -> 
19:28:29.954 -> OK
19:28:29.954 -> AT+CGMI
19:28:29.954 -> 
19:28:29.988 -> u-blox
19:28:29.988 -> 
19:28:29.988 -> OK
19:28:29.988 -> AT+GMM
19:28:29.988 -> 
19:28:30.023 -> SARA-R410M-02B
19:28:30.023 -> 
19:28:30.023 -> OK
19:28:30.023 -> [6445] ### Modem: u-blox SARA-R410M-02B
19:28:30.023 -> AT+CPIN?
19:28:30.023 -> 
19:28:30.023 -> +CPIN: READY
19:28:30.058 -> 
19:28:30.058 -> OK
19:28:30.058 -> AT+CPIN?
19:28:30.091 -> 
19:28:30.091 -> +CPIN: READY
19:28:30.091 -> 
19:28:30.091 -> OK
19:28:30.091 -> AT+CGMI
19:28:30.125 -> 
19:28:30.125 -> u-blox
19:28:30.125 -> 
19:28:30.125 -> OK
19:28:30.125 -> AT+GMM
19:28:30.160 -> 
19:28:30.160 -> SARA-R410M-02B
19:28:30.160 -> 
19:28:30.227 -> OK
19:28:30.227 -> ... Complete!  It's a u-blox SARA-R410M-02B
19:28:30.227 -> Resetting warm-up and disconnect timing for a u-blox SARA-R4
19:28:30.227 -> u-blox SARA-R410M-02B warms up in 250 ms, indicates status in 0 ms, is responsive to AT commands in less than 4500 ms, and takes up to 15000 ms to close connections and shut down.
19:28:30.227 -> Running given modem sleep function ...
19:28:30.227 -> AT+CPWROFF
[... 170 lines cut ...]
19:32:21.549 -> 
19:32:21.549 -> OK
19:32:21.549 -> AT+CGATT?
19:32:21.549 -> AT+CGATT?
19:32:21.584 -> +CGATT: 0
19:32:21.584 -> 
19:32:21.584 -> OK
19:32:21.861 -> AT+CGREG?
19:32:21.861 -> AT+CGREG?
19:32:21.896 -> +CGREG: 0,4
19:32:21.896 -> 
19:32:21.896 -> OK
// repeats for a long time

19:33:11.054 -> ...GPRS connection failed.
19:33:11.054 -> Turning u-blox SARA-R410M-02B off.
19:33:11.054 -> Running given sleep function for u-blox SARA-R410M-02B
19:33:11.054 -> AT+CPWROFF
19:33:11.054 -> AT+CPWROFF
19:33:11.054 -> OK
19:33:11.054 -> Waiting up to 15000 milliseconds for graceful shutdown...
19:33:11.159 -> ... shutdown complete after 92 ms.
19:33:11.159 -> Turning off power to u-blox SARA-R410M-02B with pin 23
19:33:11.159 -> ------------------------------------------
```

SRGDamia1, 2019-02-21:
Huh.  Well, actually... the code looks great, except for that NIST sync problem.  WAHOO!!!  YAY!!  (I'll try to get to that NIST issue later.)

It typically takes a bit to register a valid signal strength; that's all of the AT+CSQ.  

Then after saving data to the SD card, it's asking the modem over and over if it's registered or attached to the network (AT+CGATT/AT+CGREG) and the modem is saying not attached and unknown reason (4) for network registration failure.  Um.  Are you using a board without SJ13 or with?  Maybe it's not getting enough power?  Otherwise.. I'm stumped.

acgold, 2019-02-22:
Awesome!

I tried with the SJ13 jumper and got the same results. There's no activity on my hologram dashboard. I'm wondering if it just takes a while to register with the network? https://help.hologram.io/hologram-nova/troubleshooting/r410-cat-m1nb-iot-nova-troubleshooting. Or maybe the SIM card is bad for some reason?

I'll run this code for an hour or two and see if it ever registers.  
```C++
#include <Arduino.h>

const int8_t modemVccPin = 23;  // power pin
const int8_t modemSleepRqPin = 20;  // power toggle pin

void setup() 
{
  // Start the main serial port
  Serial.begin(115200);

  // Set pin modes
  pinMode(modemVccPin, OUTPUT); 
  pinMode(modemSleepRqPin, OUTPUT); 

  // power the u-blox (not-powered to power-off mode)
  digitalWrite(modemSleepRqPin, HIGH);
  digitalWrite(modemVccPin, HIGH);
  // brief wait to ensure rising edge of VCC reached
  // NB: This is probably way too long, but the timing isn't specified
  // See SARA-R4/N4 series - System Integration Manual 1.6.1
  delay(250);

  // Bring the u-blox from power-off mode to active
  digitalWrite(modemSleepRqPin, LOW);
  delay(250);  // Low time to switch on the module from power off mode = 0.15-3.2sec
  digitalWrite(modemSleepRqPin, HIGH);

  // Wait for serial ports on u-blox to become active  
  delay(5000); // ~4.5s

  // Start communication
  Serial1.begin(115200);
  // try to get the u-blox to drop speed
  // use PRINT not WRITE to get ascii out
  // println will add the necessary CR and LF (\r\n)
  Serial1.println("AT+IPR=9600");  
  Serial1.end();
  Serial1.begin(9600);
}

// Forward every message to the other serial
void loop() 
{
  Serial1.println("AT+CSQ"); 
  Serial.println(Serial1.readString()); 
  delay(5000);
  Serial1.println("AT+URAT?");
  Serial.println(Serial1.readString()); 
  delay(5000);
  Serial1.println("AT+CGREG?");
  Serial.println(Serial1.readString()); 
}
```

SRGDamia1, 2019-02-22:
In discussions of the LTE XBee with the same ublox SARA chip at heart, others were saying that it was taking on the order of 20 minutes to connect, especially if it was for the first time in a while.  I haven't had it take that long, but I think I may be lucky.

SRGDamia1, 2019-02-22:
Forcing the module to be in LTE-M only mode may speed up the connection (maybe a lot).  You set that with AT+URAT=7

acgold, 2019-02-24:
It was a network issue.  I changed it to just LTE-M, and let it sit connected to m-center, and it's back up and running.  It seems like it can take a really long time to connect the first time in a new location.

The code works really well! I'm using the tinyGsmUBLOX.h code before your latest commit because I'm getting a few errors with the latest version, and I'm running the latest menu_a_la_carte.ino (20.2) with the develop Modular Sensors branch.  RTC sync is working great now. Only issue I'm having is that sometimes it will not publish to ThingSpeak (status = -1).

SRGDamia1, 2019-02-24:
Are you publishing both to enviroDIY and ThingSpeak?  If you're going to both, are you using multiple clients? (client2/client3)  One of the quirks of the MQQT library that talks to ThingSpeak is that it always checks for an open connection on the client and attempts to use that instead of re-connecting.  Usually, that would be great and save time, but it's a problem if you're opening and closing websockets to more than one place because the library thinks there's a connection to ThingSpeaks open when the connection that's open is really the connection to EnviroDIY.  I've tried to add extra stops on the websockets to fix it, but it still gets hung up sometimes.

acgold, 2019-02-28:
Yes, I tried publishing to both as well as just ThingSpeak. When logging to just ThingSpeak, I tried using both the tinyclient3 publishing for the R410M and the other option that uses the doesn't specify a tinyclient object. Typically, it will post a few times (max 4) if I'm logging both and then stop.

SRGDamia1, 2019-03-15:
Are you still having issues with ThingSpeak?  Could you turn on the dubugging in LoggerModem and in ThinkSpeakPublisher and also turn on the StreamDebugger as the modem stream and post the log?

jtbthe3, 2019-03-18:
[thingspeakDEBUG.txt](https://github.com/EnviroDIY/ModularSensors/files/2975937/thingspeakDEBUG.txt)

(Using old version because new master does not log to SD right now)
I have the same problem.  It publishes to ThingSpeak on the first connection after setup and then every other log interval after it hangs for about the same time (232 or 282 secs) right at the spot where my MQTT write key is and then it gives status -4.

jtbthe3, 2019-03-18:
sorry that output file is garbage and not helpful. having technical difficulties copying the serial monitor from arduino ide

jtbthe3, 2019-03-18:
i just set up a test wiki watershed and I'm getting the same behavior. The first post worked and every post after that i get a 504 code.

jtbthe3, 2019-03-20:
the newest update has resolved the SD card issue and data is posting to thingspeak!  In datapublisherbase.h I set the send_buffer_size to 1500 and in PubSubClient.h I have the max packet size set to 1000.  It seems to be working pretty consistently for right now. I am seeing a hangup when it connects to NIST during initial setup but if i close the serial monitor and open again it gets going.

SRGDamia1, 2019-03-20:
So, I can't reproduce this.  It works every time for me.  

Were you using a platformio.ini with a build flag for MQTT_MAX_PACKET_SIZE?  You do need that if your sending more than 1 or two variables.  I think I have it set to 240 in all of the examples, which should be enough for 8 variables (the max number of ThingSpeak fields).

jtbthe3, 2019-03-21:
I'm just using Arduino IDE.

SRGDamia1, 2019-03-21:
@jtbthe3 ahh.. yes, you can't set build flags using the Arduino IDE, so you do have to change PubSubClient.h.  I'm sorry; I didn't think of that.  I think those buffer sizes are bigger than you need, but if your board has enough memory there's no real downside to making them oversized.  I would suggest you look into PlatformIO if you're expecting to do much programming.

SRGDamia1, 2019-06-24:
I think this should be resolved and simplified with #256

acgold, 2019-06-27:
I'm having some issues getting this modem to turn on after recent updates.  I'm using the menu_a_la_carte example from the 0.22.6 version, and here are the relevant pieces of code that I changed:

```C++
// Modem Pins - Describe the physical pin connection of your modem to your board
const int8_t modemVccPin = 23;      // MCU pin controlling modem power (-1 if not applicable)
const int8_t modemStatusPin = 19;   // MCU pin used to read modem status (-1 if not applicable)
const int8_t modemResetPin = -1;    // MCU pin connected to modem reset pin (-1 if unconnected)
const int8_t modemSleepRqPin = 20;  // MCU pin used for modem sleep/wake request (-1 if not applicable)
const int8_t modemLEDPin = redLED;  // MCU pin connected an LED to show modem status (-1 if unconnected)

// Network connection information
const char *apn = "hologram";  // The APN for the gprs connection, unnecessary for WiFi
```

Here is the output when I turn on the debugging within SodaqUbeeR410M.h:

```
Now running menu_a_la_carte.ino on Logger 180314

Using ModularSensors Library version 0.22.6
Logger timezone is set to UTC -4
RTC timezone is set to UTC
This logger has a variable array with 6 variables, of which 6 come from 6 sensors and 0 are calculated.
Current RTC time is: 2019-06-27T11:38:03-04:00
Logger setup finished!
Sending a wake-up pulse on pin 20 for Sodaq UBee R410M <--SodaqUBeeR410M
AT+IPR=9600
Sending a wake-up pulse on pin 20 for Sodaq UBee R410M <--SodaqUBeeR410M
AT+IPR=9600
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT
AT+CGMI
AT+URAT=7
Asking u-blox R410M to power down <--SodaqUBeeR410M
AT+CPWROFF
```

Any thoughts @SRGDamia1 on parameters I could change to try to get the pulse to work?

SRGDamia1, 2019-06-27:
Can you turn on the debugging in loggerModem, too?

[11 more replies]

closed 2019-07-11 by acgold